linux-nvme/libnvme #17: Missing AENs

martin-belanger, 2021-07-08:
Hi @keithbusch 
CC: @hreinecke 

I use libnvme to connect to a discovery controller. I create a persistent nvme device because I need to monitor AENs. I also register for uevents to know when changes (i.e. AENs) are detected on the persistent device. I do get a uevent the first time I apply changes to the target system. I use this uevent to trigger a call to the `nvmf_get_discovery_log()` method. But after that, I never get uevents (AENs) for subsequent changes to the target. It's as if libnvme only sends the AER command when the connection is first established, but it does not send another AER after each call to `nvmf_get_discovery_log()`.  

My questions are:  How can I keep receiving AENs on a persistent device that is managed through libnvme? Is calling `nvmf_get_discovery_log()` the right way to re-enable AENs or is there another function I should call?

keithbusch, 2021-07-08:
Thanks for the notice, I'll take a look. The library should not be sending AER requests on its own, so we should be relying completely on the kernel to be sending the command and notifying of the completion when such an event arises. The only thing that user space should be responsible for is re-arming the AER via a log page read, but it sounds like that's happening.

hreinecke, 2021-07-09:
Have you checked whether the 'RAE' (retain asynchronous event) bit is set when calling nvmf_get_discovery_log()?
Without it there's a chance that the state machine in the kernel will be messed up.

Also, it should be pretty easy to figure out if the kernel forgets to send an AER; a tcpdump should tell you ...

keithbusch, 2021-07-09:
Oh, it looks like RAE is always set to true. The last discovery log page read needs to have that cleared.

martin-belanger, 2021-07-12:
Thanks for the fix @keithbusch. I tested it and the AENs are now working properly.

closed 2021-07-12 by martin-belanger