linux-nvme/libnvme #32: broken scan

animeshtrivedi, 2021-08-30:
hi, i might be missing something with the latest updates in libnvme, but the nvme_scan seems to be broken (from this commit: https://github.com/linux-nvme/libnvme/commit/c1f6a24d0e8781ad0fe8e18215957436a504d2de) 

all examples with scans (in the test directory) work before this commit, and then no nvme devices are listed after this commit. See the example below (first with a previous commit that shows 2 NVMe devices in the system, and then with the commit, no updates). May i am missing something obvious...  

```shell
atr@qemuss20:~/src/nvme-cli/libnvme$ git checkout 10daac024f8d14930f33775eac7765e306923d2e
Previous HEAD position was fb7fdf7 Merge pull request #31 from martin-belanger/fix-compilation-warning
HEAD is now at 10daac0 Merge pull request #29 from martin-belanger/fix-compilation-warning
atr@qemuss20:~/src/nvme-cli/libnvme$ make clean ; make -j  
make[1]: Entering directory '/home/atr/src/nvme-cli/libnvme/src'
[...] 
make[1]: Leaving directory '/home/atr/src/nvme-cli/libnvme/examples'
atr@qemuss20:~/src/nvme-cli/libnvme$ sudo ./test/cpp 
nvme-subsys1 - NQN=nqn.2019-08.org.qemu:nvme-dev
 `- nvme1 pcie 0000:00:05.0 live
   `- nvme1n1lba size:4096 lba max:262144
nvme-subsys0 - NQN=nqn.2019-08.org.qemu:zns-dev
 `- nvme0 pcie 0000:00:04.0 live
   `- nvme0n1lba size:4096 lba max:262144

atr@qemuss20:~/src/nvme-cli/libnvme$ git checkout c1f6a24d0e8781ad0fe8e18215957436a504d2de 
Previous HEAD position was 10daac0 Merge pull request #29 from martin-belanger/fix-compilation-warning
HEAD is now at c1f6a24 Merge pull request #28 from hreinecke/topology
atr@qemuss20:~/src/nvme-cli/libnvme$ make clean; make -j 
make[1]: Entering directory '/home/atr/src/nvme-cli/libnvme/src'
[...]
make[1]: Leaving directory '/home/atr/src/nvme-cli/libnvme/examples'
atr@qemuss20:~/src/nvme-cli/libnvme$ sudo ./test/cpp 

atr@qemuss20:~/src/nvme-cli/libnvme$ 
```

martin-belanger, 2021-08-30:
Hi Animesh,

I tried the latest "master" branch and it's working for me. Did you run "sudo make install" before running the tests?

Regards,
Martin

Internal Use - Confidential

animeshtrivedi, 2021-08-30:
Hi @martin-belanger thanks for checking. 

I did a local installation (with --prefix in my `~/tmp/local`) and it did not make any difference. Actually the problem started from `nvme-cli` tool when I updated it to use the master libnvme code. The current build of that is on libnvme @ 667334f (git submodule dependency). 

I have now tested it on two machines (QEMU, and one server with NVMe) both with Ubuntu20.04 toolchain, gcc 9.3 (if it helps). I had a clean fresh checkout of nvme-cli and built it once with (i) 667334f libnvme; and then (ii) with the libnvme master. Here are the examples of the two `nvme`: 

```shell
atr@node3:/home/atr/tmp/local/usr/sbin$ sudo ./nvme list
Node                  SN                   Model                                    Namespace Usage                      Format           FW Rev  
--------------------- -------------------- ---------------------------------------- --------- -------------------------- ---------------- --------
atr@node3:/home/atr/tmp/local/usr/sbin$ sudo ./nvme-667334f list
Node                  SN                   Model                                    Namespace Usage                      Format           FW Rev  
--------------------- -------------------- ---------------------------------------- --------- -------------------------- ---------------- --------
nvme0n1               PHM20341005S280AGN   INTEL SSDPE21D280GA                      1         280.07  GB / 280.07  GB    512   B +  0 B   E2010480
atr@node3:/home/atr/tmp/local/usr/sbin$
```

not sure what might be missing here...

keithbusch, 2021-08-30:
What kernel are you using?

animeshtrivedi, 2021-08-30:
I have tested it with 5.4 (default Ubuntu 20.04 kernel) and 5.12 kernels. Same result.

keithbusch, 2021-08-31:
Huh, those are all recent enough and using the same sysfs hierarchy that the "list" command uses that there's nothing I can think of off the top of my head why that wouldn't be successful, and it's working for me using those kernel versions (generic, not. the Ubuntu ones, but that shouldn't matter).

If you think this is a libnvme problem, can you verify if running `examples/display-columnar` from the libnvme source tree also fails?

animeshtrivedi, 2021-08-31:
hi @keithbusch , `examples/display-columnar` also fails. See the command sequences below. I first ran the `libnvme` checkout from `4c3e5ed` (the last working commit for me), and it showed all the NVMe drives properly. Then a master checkout, and rerun the command with no output. 

```shell
atr@node1:/home/atr/tmp/libnvme$ git status  
HEAD detached at 4c3e5ed
nothing to commit, working tree clean
atr@node1:/home/atr/tmp/libnvme$ make -j && sudo ./examples/display-columnar 
[...]
make[1]: Leaving directory '/home/atr/tmp/libnvme/examples'
Subsystem        Subsystem-NQN                                                                                    Controllers
---------------- ------------------------------------------------------------------------------------------------ ----------------
nvme-subsys1     nqn.2014.08.org.nvmexpress:80868086PHM20341005T280AGN  INTEL SSDPE21D280GA                       
nvme-subsys0     nqn.2014.08.org.nvmexpress:80868086PHM20341007S280AGN  INTEL SSDPE21D280GA                       nvme0

Device   SN                   MN                                       FR       TxPort Address        Subsystem    Namespaces      
-------- -------------------- ---------------------------------------- -------- ------ -------------- ------------ ----------------
nvme0    PHM20341007S280AGN   INTEL SSDPE21D280GA                      E2010480 pcie   0000:af:00.0   nvme-subsys0 nvme0n1

Device       NSID     Sectors          Format   Controllers     
------------ -------- ---------------- -------- ----------------
nvme0n1      1        547002288        512      nvme0
atr@node1:/home/atr/tmp/libnvme$ make clean && git checkout master && make -j 
make[1]: Entering directory '/home/atr/tmp/libnvme/src'
[...]
make[1]: Leaving directory '/home/atr/tmp/libnvme/examples'
atr@node1:/home/atr/tmp/libnvme$ sudo ./examples/display-columnar 
Subsystem        Subsystem-NQN                                                                                    Controllers
---------------- ------------------------------------------------------------------------------------------------ ----------------

Device   SN                   MN                                       FR       TxPort Address        Subsystem    Namespaces      
-------- -------------------- ---------------------------------------- -------- ------ -------------- ------------ ----------------

Device       NSID     Sectors          Format   Controllers     
------------ -------- ---------------- -------- ----------------
atr@node1:/home/atr/tmp/libnvme$
```

keithbusch, 2021-09-02:
I've tried various kernels and unable to reproduce this, so it's slow progress on my side. If I write a debug patch, could you test with that?

animeshtrivedi, 2021-09-02:
hi @keithbusch absolutely, happy to run patches. Meanwhile I have reverted on the old working commit. Many thanks for your help.

keithbusch, 2021-09-02:
never mind, i can see that the makefile rule isn't actually rebuilding my source tree correctly!!! Started fresh, yeah, it's broken because it requires it finds hosts, which it won't find here.... I'll fix it.

keithbusch, 2021-09-02:
@hreinecke , just fyi, it looks like the host scanning portion recently added doesn't work for pci attached.

keithbusch, 2021-09-02:
It should be successful now.

animeshtrivedi, 2021-09-02:
just tried - the current master works (with the reverted patch).

keithbusch, 2021-09-02:
thank you for confirming, and for the report.

I'll try to fix the build rules so I don't get these false positives again. Maybe switch to 'meson' from 'make' since some people think that's easier to maintain.

closed 2021-09-02 by keithbusch