linux-nvme/libnvme #90: Python tests fail if library is embedded

igaw, 2021-11-02:
When the library is embedded the testsuite fails in the Python tests:

https://github.com/linux-nvme/nvme-cli/runs/4079968220?check_suite_focus=true

I suspect the paths are not correct.

igaw, 2021-11-02:
@martin-belanger do happen to have time to look into this problem?

martin-belanger, 2021-11-02:
Yes I can have a look today.

igaw, 2021-11-02:
I've went ahead and disabled the tests in nvme-cli. I had the tests for nmve-cli disabled anyway, so there is little point in enabling them for the embedded version of libnvme. So there is no rush. Just in case you see easily spot the problem.

martin-belanger, 2021-11-02:
Hi @igaw - I just submitted a pull request (https://github.com/linux-nvme/libnvme/pull/91) that attempts to fix this issue. Since we were pretty much blind to the cause of the failure, I added a new step to the workflow that collects the meson test log file in case of failure. This should help us diagnose the problem further.

igaw, 2021-11-03:
Okay merged the commit from PR #91. It's probably a good idea to have some more verbose logs when things go wrong anyway.

igaw, 2021-11-17:
The tests are disabled as there were too many problems at the moment. Let's close this is for the time being.

closed 2021-11-17 by igaw