linux-nvme/libnvme #95: Add versioning to exported symbols

igaw, 2021-11-08:
Add versioning to all exported symbols,  see also discussion in #51

igaw, 2021-11-10:
I've been looking at https://github.com/GNOME/glib is doing the symbol versioning. 

1) There is no the linker map file.  Instead every symbol which needs to be exported is annotated directly in the source code
```
GLIB_AVAILABLE_IN_ALL
GArray* g_array_new               (gboolean          zero_terminated,
				   gboolean          clear_,
				   guint             element_size);
```
That means the symbols are either 'hidden', 'internal' or 'default'. The versioning is not done by moving the symbols into a different linker section. It's probably safe to assume doing anything else then 'hidden', 'internal' or 'default' is asking for troubles.

2) There are macro definitions for every release which allow to attach versions symbols or if it's getting deprecated. But this is only done on source code level. This means this kind of versioning is useful during compile time. IIRC, the library issues depreciated warnings. See https://github.com/GNOME/glib/blob/main/glib/gversionmacros.h for more details.

3) The library versioning is done via
```
glib_version = meson.project_version()
glib_api_version = '2.0'
version_arr = glib_version.split('.')
major_version = version_arr[0].to_int()
minor_version = version_arr[1].to_int()
micro_version = version_arr[2].to_int()

interface_age = minor_version.is_odd() ? 0 : micro_version
binary_age = 100 * minor_version + micro_version

soversion = 0
# Maintain compatibility with previous libtool versioning
# current = minor * 100 + micro
current = binary_age - interface_age
library_version = '@0@.@1@.@2@'.format(soversion, current, interface_age)
darwin_versions = [current + 1, '@0@.@1@'.format(current + 1, interface_age)]
```
which is almost what we currently have. The soversion part is currently hard coded to '1'. This should be made a bit more flexible.

Given this, I think we should think add something like the version macros as glib has and replace the libnvme.map with it. The rest looks good IMO.

igaw, 2021-11-11:
I did some more research, as glib seems not the best example. 

Anyway, @hreinecke's proposal seems to be used also by other projects, e.g. dpdk or systemd,
```
LIBSYSTEMD_248 {
global:
        sd_bus_open_user_machine;
        sd_bus_message_send;

        sd_event_source_set_ratelimit;
        sd_event_source_get_ratelimit;
        sd_event_source_is_ratelimited;

        sd_device_get_action;
        sd_device_get_seqnum;
        sd_device_new_from_stat_rdev;
        sd_device_trigger;
} LIBSYSTEMD_247;

LIBSYSTEMD_249 {
global:
        sd_device_monitor_filter_add_match_sysattr;
        sd_device_monitor_filter_add_match_parent;
        sd_device_get_usec_initialized;
        sd_device_trigger_with_uuid;
        sd_device_get_trigger_uuid;
        sd_device_new_from_ifname;
        sd_device_new_from_ifindex;
} LIBSYSTEMD_248;

LIBSYSTEMD_250 {
global:
        sd_device_get_diskseq;
} LIBSYSTEMD_249;
```
So I'd say we should do it the same way. 

Though there is  one thing I noticed is that the standard semantic versioning is usually used with libraries. For example systemd uses a single number as version but for the shipped library the project defaults to the triple:

```
project('systemd', 'c',
        version : '249',
       ... 
       )

libsystemd_version = '0.32.0'

libsystemd = shared_library(
        'systemd',
        version : libsystemd_version,
        link_args : ['-shared',
                     '-Wl,--version-script=' + libsystemd_sym_path]
         ....
)
```
which seems to do the whole trick
```
ls /usr/lib64/libsystemd.*   
/usr/lib64/libsystemd.so  /usr/lib64/libsystemd.so.0  /usr/lib64/libsystemd.so.0.32.0
```
So this would mean we should also switch to the commonly used semantic versioning to avoid problems in the long run. Any opinions?

hreinecke, 2021-11-11:
Indeed, that would be fine.
(Not surprisingly, seeing that it was my idea to begin with.)

However, the real _challenging_ part here is that you cannot _remove_ symbols that way.
IE if one version marked a symbol as global, it'll _stay_ global during the entire lifetime.
And you cannot have a future library _not_ having that symbol as global.
(Like making a former global symbol local or completely removing it.)
Not sure if that's just me being too stupid or indeed a feature; we should probably ask the gcc / glibc heroes for that.

igaw, 2021-11-11:
If I understood the triple versioning scheme correctly, in case you want to remove symbols you increase the major number. That means we would do the following

- 0.y+1.0: for backwards compatible releases
- x+1.0.0: when breaking API

igaw, 2021-11-11:
https://semver.org/

hreinecke, 2021-11-11:
Oh, _that_ was not my concern.
My concern was more: _how_ do we remove symbols from the symbol map file?
We can't remove the symbol from the original version (as it was present there), and I haven't found instructions on how to deprecate symbols once added.

igaw, 2021-11-11:
Not sure what's the problem you see. When doing a major number increase, symbols can be removed from the symbol map and from the API. In this case I would move all symbols under a new prefix.

igaw, 2021-11-17:
The libnvme.map is udpated as discussed above. Also the libtool versioning is done like systemd does it.

closed 2021-11-17 by igaw