linux-nvme/libnvme #154: nvme_ctrl_alloc() can't handle duplicate connections

tbzatek, 2021-12-21:
Okay, I admit my use case is a little weird and might not reflect real world scenario: loop transport, two (duplicate) connections, resulting in a single subsystem and two controllers in sysfs. The libnvme scanning logic would however only find one controller.

https://github.com/linux-nvme/libnvme/blob/ddd725e780625d25dd199dc367dda5503e13c685/src/nvme/tree.c#L1270-L1271

Looking at how `nvme_lookup_ctrl()` is called and how are the keys matched inside I see no way to distinguish seemingly duplicate controller entries, just having different name and sysfs path.

This is a broader issue, posting now in light of ongoing tree scanning logic rewrite in #108 to get some consideration.

----

```
# old/nvme list -v
NVM Express Subsystems

Subsystem        Subsystem-NQN                                                                                    Controllers
---------------- ------------------------------------------------------------------------------------------------ ----------------
nvme-subsys0     test_subnqn                                                                                      nvme0, nvme1

NVM Express Controllers

Device   SN                   MN                                       FR       TxPort Address        Subsystem    Namespaces      
-------- -------------------- ---------------------------------------- -------- ------ -------------- ------------ ----------------
nvme0    cce1b445357e708d9d62 Linux                                    5.14.17- loop                  nvme-subsys0 nvme0n1, nvme0n2, nvme0n3
nvme1    cce1b445357e708d9d62 Linux                                    5.14.17- loop                  nvme-subsys0 nvme0n1, nvme0n2, nvme0n3

NVM Express Namespaces

Device       NSID     Usage                      Format           Controllers     
------------ -------- -------------------------- ---------------- ----------------
nvme0n1      1          1.07  GB /   1.07  GB    512   B +  0 B   nvme0, nvme1
nvme0n2      2          1.07  GB /   1.07  GB    512   B +  0 B   nvme0, nvme1
nvme0n3      3          1.07  GB /   1.07  GB    512   B +  0 B   nvme0, nvme1
```

```
# git-master/nvme list -v
Subsystem        Subsystem-NQN                                                                                    Controllers
---------------- ------------------------------------------------------------------------------------------------ ----------------
nvme-subsys0     test_subnqn                                                                                      nvme1

Device   SN                   MN                                       FR       TxPort Address        Subsystem    Namespaces      
-------- -------------------- ---------------------------------------- -------- ------ -------------- ------------ ----------------
nvme1    cce1b445357e708d9d62 Linux                                    5.14.17- loop   (null)         nvme-subsys0 nvme0n3, nvme0n2, nvme0n1, nvme0n3, nvme0n2, nvme0n1

Device       Generic      NSID     Usage                      Format           Controllers     
------------ ------------ -------- -------------------------- ---------------- ----------------
nvme0n3      ng0n3        3          1.07  GB /   1.07  GB    512   B +  0 B   nvme1
nvme0n2      ng0n2        2          1.07  GB /   1.07  GB    512   B +  0 B   nvme1
nvme0n1      ng0n1        1          1.07  GB /   1.07  GB    512   B +  0 B   nvme1
```

keithbusch, 2021-12-21:
Thanks for the report! I don't think your setup is at all unusual for a fabrics target (even if it is "loop"). I'll take a look.

hreinecke, 2021-12-22:
Hmm. That is true, we don't have a way of distinguish between them.
Typically we wouldn't notice it as we don't allow for duplicate connections.

Guess the only way to fix this is to return a list of controllers ...

igaw, 2022-02-02:
Ah, in this case we should at least change the current API so it is possible to return several controllers before we do our 1.0 release.

hreinecke, 2022-02-03:
I've now prepared PR #219, which should address this issue. Can you test with that?

hreinecke, 2022-02-03:
With PR #219 I now get:

# /tmp/nvme list -v
Subsystem        Subsystem-NQN                                                                                    Controllers
---------------- ------------------------------------------------------------------------------------------------ ----------------
nvme-subsys0     nqn.2014-08.org.nvmexpress:NVMf:uuid:95096971-6a11-46a0-8fbc-c301194c5857                        nvme2, nvme1, nvme0

Device   SN                   MN                                       FR       TxPort Address        Subsystem    Namespaces      
-------- -------------------- ---------------------------------------- -------- ------ -------------- ------------ ----------------
nvme2    f51d6e34723a74bea185 Linux                                    5.16.0-r loop   traddr=2       nvme-subsys0 
nvme1    f51d6e34723a74bea185 Linux                                    5.16.0-r loop   traddr=2       nvme-subsys0 
nvme0    f51d6e34723a74bea185 Linux                                    5.16.0-r loop   traddr=1       nvme-subsys0

hreinecke, 2022-02-03:
PR has been merged.

closed 2022-02-03 by hreinecke

tbzatek, 2022-02-04:
Works fine now in my original scenario, thanks.