linux-nvme/libnvme #165: Build documentation error

alexmchou, 2022-01-09:
Hi,
I used "meson .build -Dman=true" then "cd .build, ninja", but got following error messages:
[1/1] Generating man with a custom command.
FAILED: doc/libnvme.1 
/usr/bin/sphinx-build -b man /home/zxm/libnvme/doc /home/zxm/libnvme/.build/doc
Running Sphinx v1.8.5
building [mo]: targets for 0 po files that are out of date
building [man]: all manpages
updating environment: 2 added, 0 changed, 0 removed
reading sources... [ 50%] index
reading sources... [100%] libnvme

/home/zxm/libnvme/doc/libnvme.rst:3906: WARNING: Unexpected indentation.
/home/zxm/libnvme/doc/libnvme.rst:6585: WARNING: Unexpected indentation.
/home/zxm/libnvme/doc/libnvme.rst:6583: WARNING: Inline interpreted text or phrase reference start-string without end-string.
/home/zxm/libnvme/doc/libnvme.rst:6586: WARNING: Block quote ends without a blank line; unexpected unindent.
/home/zxm/libnvme/doc/libnvme.rst:10192: WARNING: Inline strong start-string without end-string.
/home/zxm/libnvme/doc/libnvme.rst:10652: WARNING: Unexpected indentation.
/home/zxm/libnvme/doc/libnvme.rst:11132: WARNING: Inline interpreted text or phrase reference start-string without end-string.
/home/zxm/libnvme/doc/libnvme.rst:11153: WARNING: Inline interpreted text or phrase reference start-string without end-string.
/home/zxm/libnvme/doc/libnvme.rst:11153: WARNING: Inline interpreted text or phrase reference start-string without end-string.
/home/zxm/libnvme/doc/libnvme.rst:11180: WARNING: Inline interpreted text or phrase reference start-string without end-string.

Sphinx error:
master file /home/zxm/libnvme/doc/contents.rst not found
ninja: build stopped: subcommand failed.

igaw, 2022-01-10:
Does this also happen when you do 

```
meson .build -Dman=true
ninja -C .build
```

We had the same problem for 'test'.  Let me try to figure out what's going on.

BTW, if you want to update the in tree sources check the update-docs.sh script.

martin-belanger, 2022-01-10:
Hi @igaw - I get the exact same result. It looks like sphinx expects a file called `contents.rst`. Instead we have a file called `libnvme.rst`. I'm not familiar with sphinx. I need to Google it. I'll let you know what I find.

martin-belanger, 2022-01-10:
Submitted https://github.com/linux-nvme/libnvme/pull/167 to fix the error.

closed 2022-01-10 by igaw