linux-nvme/libnvme #168: Visibility for library functions __nvme_msg, __nvme_get_log_page

tasleson, 2022-01-11:
```
$ nm --defined-only -g ./src/libnvme.so.1.0.1 | grep __
0000000000007d10 T __nvme_get_log_page
0000000000010860 T __nvme_msg
```

From the naming, it would seem like these shouldn't be exposed?

igaw, 2022-01-12:
`__nvme_get_log_page` seems to be part of linux.h API. But the naming is not great. We should find a better name for it, IMO.

About `__mvne_msg `, we have a discussion on this in #160 and how to expose the logging API.

igaw, 2022-01-14:
#173 get's rid of `__nvme_msg`

igaw, 2022-01-18:
As it turns out both symbols need to be exported. The naming is just not good.

tasleson, 2022-01-20:
@igaw
> As it turns out both symbols need to be exported. The naming is just not good.

Can you elaborate on this?  This works, but I'm likely missing something: https://github.com/linux-nvme/libnvme/pull/189/commits/3ffb178cab669217bb7953accdd2e6193b76a786

igaw, 2022-01-21:
Do you compile nvme-cli with libnvme together? See me comments in your POC PR on this.

igaw, 2022-02-01:
Both functions are now gone from the export map.

closed 2022-02-01 by igaw