linux-nvme/libnvme #174: Python tests fail in debug build

igaw, 2022-01-14:
When the library is configured as debug build the Python test fail due to missing str_strstr() function from the ccan library.

martin-belanger, 2022-01-14:
I'll take a look

martin-belanger, 2022-01-14:
@igaw - I figured it out.

When in debug mode, `CCAN_STR_DEBUG` is defined. This causes `#ifdef CCAN_STR_DEBUG` to evaluate to true and the standard POSIX APIs such as `strstr()` to be replaced by `str_strstr()` (see file `ccan/str/str.h`). I'm not sure why we go through all this trouble. Maybe this comment in `ccan/str/str.h` gives us a clue:
```
/* With GNU magic, we can make const-respecting standard string functions. */
```
In any case, when `CCAN_STR_DEBUG` is defined, any application that includes `ccan/str/str.h` will now be instructed to call `str_strstr()` instead of `strstr()`. And it will expect to find `str_strstr()` in `libnvme.so`. However, `str_strstr()` is not exposed in `libnvme.so`'s global symbols. It needs to be added to `libnvme.map`. The question is: _should debug wrapper functions such as `str_strstr()` be added to `libnvme.map`_?

The other thing we could do is to keep the standard POSIX APIs (i.e. `strstr()`, `strchr()`, `strrchr()`, etc.) and never replace them by wrapper functions.

martin-belanger, 2022-01-14:
BTW, while working on this I found some `meson.build` clean-ups that I would like to submit. If that's ok?

igaw, 2022-01-16:
Ah that's the reason. The str debug feature of ccan was already helpful in one of my debug session, so I'd like to keep it somehow. I wonder if it would be possible to add the str_strstr during debug builds to the export list.

This brings up the topic if we want to introduce a EXPORT() macro which manages the export symbols and drop the libnvme.map approach. Another point is that we need to annotate the header files anyway so that we can 
deprecated function etc. But that is something for another issue I suppose :)

Sure, send the clean-ups.

eli-schwartz, 2022-01-16:
If it's in a public header, you must export it. If it's in a private header, you should not export it. :p 

Maybe for libnvme-internal useful utility functions, you'd prefer to create an e.g. `libcommon = static_library('common', sources, install: false)` and have both the libnvme library and any executables link to it -- then that static library would contain functions which aren't exported, but are used across both shared libraries and executables.

igaw, 2022-01-19:
Fixed with #185. Thanks a lot.

closed 2022-01-19 by igaw