linux-nvme/libnvme #187: Ability to have error messages go to different file descriptor or user supplied callback

tasleson, 2022-01-19:
I think it would be good to have a library user provide a file descriptor for logging or a call back as suggested here: https://github.com/linux-nvme/libnvme/pull/160#issuecomment-1010314146

Having all these messages to written to `stderr` isn't ideal for library users.

tbzatek, 2022-01-24:
Well, the messages currently logged on `stderr` are more customized than the `nvme_errno_to_string()` translation but I find the error codes useful enough at the current state.

Perhaps we should set `nvme_log_level` by default to `-1` and let the clients (e.g. `nvme-cli`) set their own console logging level. Not every (consumer) library has the luxury of providing initialisation functions that can set up things in the nested (`libnvme`) libraries.

hreinecke, 2022-01-28:
Please check PR #202. I've added a 'nvme_root_t' argument to nvme_msg, which now holds the filedescriptor where the messages should be sent to.
Applications can then set the filedescriptor to whatever they want.
Would that work for you?

tasleson, 2022-01-31:
> Would that work for you?

Yes, this works thanks

igaw, 2022-02-01:
PR #202 has been merged. Closing this issue.

closed 2022-02-01 by igaw