linux-nvme/libnvme #206: API cleanup: nvme_$NAME_get() vs nvme_get_$NAME()

igaw, 2022-01-31:
The current API isn't really consistent in the naming pattern. We should use one an stick to it.

ikegami-t, 2023-06-15:
The PR #660 is closed as correct implementation so seems the issue can be closed now.

igaw, 2023-06-15:
I've opened this issue for the various variants of our getters/setters for the non MI API. When we do the 2.0 we should go over the existing function names an make them consistent.

ikegami-t, 2023-06-15:
Seems there is no other remaining function name to fix as below.
```sh
tokunori@tokunori-desktop:~/nvme-cli/subprojects/libnvme$ git grep "_get("
examples/mi-conf.c:     bus = dbus_bus_get(DBUS_BUS_SYSTEM, &berr);
examples/mi-mctp.c:int do_config_get(nvme_mi_ep_t ep, int argc, char **argv)
examples/mi-mctp.c:             rc = do_config_get(ep, argc, argv);
libnvme/nvme.i: const char *nvme_host_dhchap_key_get(struct nvme_host *h) {
libnvme/nvme.i: const char *nvme_subsystem_name_get(struct nvme_subsystem *s) {
libnvme/nvme.i: struct nvme_host *nvme_subsystem_host_get(struct nvme_subsystem *s) {
libnvme/nvme.i: const char *nvme_ctrl_name_get(struct nvme_ctrl *c) {
libnvme/nvme.i: struct nvme_subsystem *nvme_ctrl_subsystem_get(struct nvme_ctrl *c) {
libnvme/nvme.i: const char *nvme_ctrl_state_get(struct nvme_ctrl *c) {
libnvme/nvme.i: const char *nvme_ctrl_dhchap_key_get(struct nvme_ctrl *c) {
libnvme/nvme.i: const char *nvme_ctrl_dhchap_host_key_get(struct nvme_ctrl *c) {
libnvme/nvme.i: const char *nvme_ns_name_get(struct nvme_ns *n) {
libnvme/nvme.i: PyObject *nbft_get(const char * filename)
libnvme/nvme.i:PyObject *nbft_get(const char * filename);
libnvme/tests/test-nbft.py:        actual_nbft = nvme.nbft_get(args.filename)
src/nvme/json.c:        attr_obj = json_object_object_get(port_obj, "transport");
src/nvme/json.c:        attr_obj = json_object_object_get(port_obj, "traddr");
src/nvme/json.c:        attr_obj = json_object_object_get(port_obj, "host_traddr");
src/nvme/json.c:        attr_obj = json_object_object_get(port_obj, "host_iface");
src/nvme/json.c:        attr_obj = json_object_object_get(port_obj, "trsvcid");
src/nvme/json.c:        attr_obj = json_object_object_get(port_obj, "dhchap_key");
src/nvme/json.c:        attr_obj = json_object_object_get(port_obj, "dhchap_ctrl_key");
src/nvme/json.c:        nqn_obj = json_object_object_get(subsys_obj, "nqn");
src/nvme/json.c:        app_obj = json_object_object_get(subsys_obj, "application");
src/nvme/json.c:        port_array = json_object_object_get(subsys_obj, "ports");
src/nvme/json.c:        attr_obj = json_object_object_get(host_obj, "hostnqn");
src/nvme/json.c:        attr_obj = json_object_object_get(host_obj, "hostid");
src/nvme/json.c:        attr_obj = json_object_object_get(host_obj, "dhchap_key");
src/nvme/json.c:        attr_obj = json_object_object_get(host_obj, "hostsymname");
src/nvme/json.c:        attr_obj = json_object_object_get(host_obj, "persistent_discovery_ctrl");
src/nvme/json.c:        subsys_array = json_object_object_get(host_obj, "subsystems");
src/nvme/mi-mctp.c:     bus = dbus_bus_get(DBUS_BUS_SYSTEM, &berr);
src/nvme/mi.c:int nvme_mi_mi_config_get(nvme_mi_ep_t ep, __u32 dw0, __u32 dw1,
src/nvme/mi.h: * See &nvme_mi_mi_config_get() and &nvme_mi_config_set().
src/nvme/mi.h:int nvme_mi_mi_config_get(nvme_mi_ep_t ep, __u32 dw0, __u32 dw1,
src/nvme/mi.h:  rc = nvme_mi_mi_config_get(ep, dw0, 0, &tmp);
src/nvme/mi.h:  rc = nvme_mi_mi_config_get(ep, dw0, 0, &tmp);
```

igaw, 2023-06-16:
This is issue is about these function here:

```
		[...]
		nvme_ctrl_disconnect;
		nvme_ctrl_first_ns;
		nvme_ctrl_first_path;
		nvme_ctrl_get_address;
		nvme_ctrl_get_config;
		nvme_ctrl_get_dhchap_key;
		nvme_ctrl_get_discovery_ctrl;
		[...]
```

vs

```
		[...]
		nvme_get_feature_length;
		nvme_get_features;
		nvme_get_features_arbitration;
		nvme_get_features_async_event;
		nvme_get_features_auto_pst;
		nvme_get_features_endurance_event_cfg;
		nvme_get_features_err_recovery;
		nvme_get_features_hctm;
		nvme_get_features_host_behavior;
		[...]
```

ikegami-t, 2023-06-16:
Thanks for your explanation. But seems both function lists follow the naming rules as below.
1. nvme_$NAME_$VERB()
nvme_ctrl_disconnect; ($NAME: ctrl, $VERB: disconnect)
2. nvme_$NAME_$VERB_$NOUN()
nvme_ctrl_get_address; ($NAME: ctrl, $VERB: get, $NOUN: address)
nvme_ctrl_get_config; ($NAME: ctrl, $VERB: get, $NOUN: config)
nvme_ctrl_get_dhchap_key; ($NAME: ctrl, $VERB: get, $NOUN: dhchap_key)
nvme_ctrl_get_discovery_ctrl; ($NAME: ctrl, $VERB: get, $NOUN: discovery_ctrl)
3. nvme_$NAME_$NOUN()
nvme_ctrl_first_ns; ($NAME: ctrl, $NOUN: first_ns)
nvme_ctrl_first_path; ($NAME: ctrl, $NOUN: first_path)
nvme_get_features_arbitration; ($NAME: get_features, $NOUN: arbitration)
nvme_get_features_async_event; ($NAME: get_features, $NOUN: async_event)
nvme_get_features_auto_pst; ($NAME: get_features, $NOUN: auto_pst)
nvme_get_features_endurance_event_cfg; ($NAME: get_features, $NOUN: endurance_event_cfg)
nvme_get_features_err_recovery; ($NAME: get_features, $NOUN: err_recovery)
nvme_get_features_hctm; ($NAME: get_features, $NOUN: hctm)
nvme_get_features_host_behavior; ($NAME: get_features, $NOUN: host_behavior)
4. nvme_$VERB_$NOUN()
nvme_get_feature_length; ($VERB: get, $NOUN: feature_length)
5. nvme_$NAME()
nvme_get_features; ($NAME: get_features)