linux-nvme/libnvme #228: meson.build:39:0: ERROR: Dependency "uuid" not found, tried pkgconfig and cmake

Bobarshad, 2022-02-10:
The Meson build system
Version: 0.61.1
Source dir: /home/ngd/hossein/libnvme
Build dir: /home/ngd/hossein/libnvme/.build
Build type: native build
Project name: libnvme
Project version: 1.0
C compiler for the host machine: cc (gcc 7.5.0 "cc (Ubuntu 7.5.0-3ubuntu1~18.04) 7.5.0")
C linker for the host machine: cc ld.bfd 2.30
C++ compiler for the host machine: c++ (gcc 7.5.0 "c++ (Ubuntu 7.5.0-3ubuntu1~18.04) 7.5.0")
C++ linker for the host machine: c++ ld.bfd 2.30
Host machine cpu family: x86_64
Host machine cpu: x86_64
Found pkg-config: /usr/local/bin/pkg-config (0.29.2)
Found CMake: /usr/local/bin/cmake (3.23.0)
Run-time dependency uuid found: NO (tried pkgconfig and cmake)

meson.build:39:0: ERROR: Dependency "uuid" not found, tried pkgconfig and cmake

A full log can be found at /home/ngd/hossein/libnvme/.build/meson-logs/meson-log.txt
-----------------------------------------------------------------------------------------------------------------

cat /home/ngd/hossein/libnvme/.build/meson-logs/meson-log.txt

Build started at 2022-02-09T17:43:34.106390
Main binary: /usr/bin/python3
Build Options: 
Python system: Linux
The Meson build system
Version: 0.61.1
Source dir: /home/ngd/hossein/libnvme
Build dir: /home/ngd/hossein/libnvme/.build
Build type: native build
Project name: libnvme
Project version: 1.0
Sanity testing C compiler: cc
Is cross compiler: False.
Sanity check compiler command line: cc sanitycheckc.c -o sanitycheckc.exe -D_FILE_OFFSET_BITS=64
Sanity check compile stdout:

-----
Sanity check compile stderr:

-----
Running test binary command: /home/ngd/hossein/libnvme/.build/meson-private/sanitycheckc.exe
C compiler for the host machine: cc (gcc 7.5.0 "cc (Ubuntu 7.5.0-3ubuntu1~18.04) 7.5.0")
C linker for the host machine: cc ld.bfd 2.30
Sanity testing C++ compiler: c++
Is cross compiler: False.
Sanity check compiler command line: c++ sanitycheckcpp.cc -o sanitycheckcpp.exe -D_FILE_OFFSET_BITS=64
Sanity check compile stdout:

-----
Sanity check compile stderr:

-----
Running test binary command: /home/ngd/hossein/libnvme/.build/meson-private/sanitycheckcpp.exe
C++ compiler for the host machine: c++ (gcc 7.5.0 "c++ (Ubuntu 7.5.0-3ubuntu1~18.04) 7.5.0")
C++ linker for the host machine: c++ ld.bfd 2.30
Sanity testing C compiler: cc
Is cross compiler: False.
Sanity check compiler command line: cc sanitycheckc.c -o sanitycheckc.exe -D_FILE_OFFSET_BITS=64
Sanity check compile stdout:

-----
Sanity check compile stderr:

-----
Running test binary command: /home/ngd/hossein/libnvme/.build/meson-private/sanitycheckc.exe
C compiler for the build machine: cc (gcc 7.5.0 "cc (Ubuntu 7.5.0-3ubuntu1~18.04) 7.5.0")
C linker for the build machine: cc ld.bfd 2.30
Sanity testing C++ compiler: c++
Is cross compiler: False.
Sanity check compiler command line: c++ sanitycheckcpp.cc -o sanitycheckcpp.exe -D_FILE_OFFSET_BITS=64
Sanity check compile stdout:

-----
Sanity check compile stderr:

-----
Running test binary command: /home/ngd/hossein/libnvme/.build/meson-private/sanitycheckcpp.exe
C++ compiler for the build machine: c++ (gcc 7.5.0 "c++ (Ubuntu 7.5.0-3ubuntu1~18.04) 7.5.0")
C++ linker for the build machine: c++ ld.bfd 2.30
Build machine cpu family: x86_64
Build machine cpu: x86_64
Host machine cpu family: x86_64
Host machine cpu: x86_64
Target machine cpu family: x86_64
Target machine cpu: x86_64
Pkg-config binary for 1 is not cached.
Pkg-config binary missing from cross or native file, or env var undefined.
Trying a default Pkg-config fallback at pkg-config
Found pkg-config: /usr/local/bin/pkg-config (0.29.2)
Determining dependency 'uuid' with pkg-config executable '/usr/local/bin/pkg-config'
env[PKG_CONFIG_PATH]: 
Called `/usr/local/bin/pkg-config --modversion uuid` -> 1

CMake binary for 1 is not cached
CMake binary missing from cross or native file, or env var undefined.
Trying a default CMake fallback at cmake
Found CMake: /usr/local/bin/cmake (3.23.0)
Extracting basic cmake information
CMake Toolchain: Calling CMake once to generate the compiler state
Calling CMake (['/usr/local/bin/cmake']) in /home/ngd/hossein/libnvme/.build/meson-private/__CMake_compiler_info__ with:
  - "--trace-expand"
  - "--trace-format=json-v1"
  - "--no-warn-unused-cli"
  - "--trace-redirect=cmake_trace.txt"
  - "-G"
  - "Ninja"
  - "-DCMAKE_TOOLCHAIN_FILE=/home/ngd/hossein/libnvme/.build/meson-private/__CMake_compiler_info__/CMakeMesonTempToolchainFile.cmake"
  - "."
Try CMake generator: auto
Calling CMake (['/usr/local/bin/cmake']) in /home/ngd/hossein/libnvme/.build/meson-private/cmake_uuid with:
  - "--trace-expand"
  - "--trace-format=json-v1"
  - "--no-warn-unused-cli"
  - "--trace-redirect=cmake_trace.txt"
  - "-DCMAKE_TOOLCHAIN_FILE=/home/ngd/hossein/libnvme/.build/meson-private/cmake_uuid/CMakeMesonToolchainFile.cmake"
  - "."
  -- Module search paths:    ['/', '/home/ngd/cmake-3.23.0-rc1-linux-x86_64', '/opt', '/usr', '/usr/local']
  -- CMake root:             /home/ngd/cmake-3.23.0-rc1-linux-x86_64/share/cmake-3.23
  -- CMake architectures:    ['x86_64-linux-gnu']
  -- CMake lib search paths: ['lib', 'lib32', 'lib64', 'libx32', 'share', 'lib/x86_64-linux-gnu']
Preliminary CMake check failed. Aborting.
Run-time dependency uuid found: NO (tried pkgconfig and cmake)

meson.build:39:0: ERROR: Dependency "uuid" not found, tried pkgconfig and cmake

igaw, 2022-02-10:
```
Found pkg-config: /usr/local/bin/pkg-config (0.29.2)
Found CMake: /usr/local/bin/cmake (3.23.0)
Run-time dependency uuid found: NO (tried pkgconfig and cmake)
```

Does this version of pkg-config and cmake lookup the system paths as well? 

We have a CI build using Ubuntu as distro. So I am pretty sure there is no bug in libnvme.

closed 2022-02-10 by igaw

martin-belanger, 2022-02-10:
I tried on Ubuntu 18.04 and everything works fine for me. 

After "`apt-get install uuid-dev`", I have the following files installed:
```
/usr/share/doc/libuuid1
/usr/lib/x86_64-linux-gnu/libuuid.so
/usr/lib/x86_64-linux-gnu/libuuid.so.1
/usr/lib/x86_64-linux-gnu/libuuid.so.1.3.0
/usr/lib/x86_64-linux-gnu/libuuid.a
/usr/lib/i386-linux-gnu/libuuid.so.1
/usr/lib/i386-linux-gnu/libuuid.so.1.3.0
/usr/include/uuid/uuid.h
/usr/share/doc/uuid-dev
/usr/lib/x86_64-linux-gnu/pkgconfig/uuid.pc   <--- That's the pkgconfig file that should provide meson with all that it needs.
```
Maybe something went wrong with your installation. Have you tried running "sudo ldconfig" to reconfigure the linker run-time bindings. https://man7.org/linux/man-pages/man8/ldconfig.8.html

martin-belanger, 2022-02-10:
Scratch that. That was on Ubuntu 20.04

eli-schwartz, 2022-02-10:
> Have you tried running "sudo ldconfig" to reconfigure the linker run-time bindings.

This would have absolutely no effect on whether the pkg-config program can find the text files describing "uuid"s compile flags.

All it could ever do is tell `/lib/ld-linux-x86-64.so.2` the newly cached location of `libuuid.so.1` when trying to execute a program which depends on `libuuid.so.1`.

The non-standard pkg-config program located in /usr/local is probably a much better bet. @Bobarshad, what is the output of:
```
pkg-config --debug --modversion uuid
```

Bobarshad, 2022-02-10:
Thanks all, 
I used another machine with Ubuntu 18.04, with the following commands. It works now.

$ sudo apt-get install uuid-dev
$ sudo apt-get install meson
$ meson --version
0.45.1

$ meson .build
	The Meson build system
	Version: 0.45.1
	Source dir: /home/ngd/hossein/project/libnvme
	Build dir: /home/ngd/hossein/project/libnvme/.build
	Build type: native build
	meson_options.txt:9:0: ERROR: Unknown type feature.
	A full log can be found at /home/ngd/hossein/project/libnvme/.build/meson-logs/meson-log.txt

$ python3 --version
Python 3.6.9
$ sudo apt-get install ninja-build
$ python3 -m pip --version
$ python3 -m pip install --upgrade pip
$ python3 -m pip --version
$ python3 -m pip install meson
$meson --version
0.45.1
$ sudo update-alternatives --install /usr/bin/meson meson /usr/share/meson/meson 1
$ sudo update-alternatives --install /usr/bin/meson meson /home/ngd/.local/bin/meson 2
$ meson --version
0.61.1

$ meson .build