linux-nvme/libnvme #240: Installation fails

rwk-unil, 2022-02-15:
Hello.

I am trying to install libnvme following the README. I get into this error
```
$ sudo meson install
ninja: Entering directory `.'
ninja: no work to do.
Installing subprojects/json-c-0.15/libjson-c.so to /usr/lib/x86_64-linux-gnu
Installing subprojects/openssl-1.1.1l/libcrypto.so to /usr/lib/x86_64-linux-gnu
Installing subprojects/openssl-1.1.1l/libssl.so to /usr/lib/x86_64-linux-gnu
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/mesonbuild/mesonmain.py", line 140, in run
    return options.run_func(options)
  File "/usr/lib/python3/dist-packages/mesonbuild/minstall.py", line 554, in run
    installer.do_install(datafilename)
  File "/usr/lib/python3/dist-packages/mesonbuild/minstall.py", line 372, in do_install
    self.install_targets(d)
  File "/usr/lib/python3/dist-packages/mesonbuild/minstall.py", line 468, in install_targets
    raise RuntimeError('File {!r} could not be found'.format(t.fname))
RuntimeError: File 'subprojects/openssl-1.1.1l/openssl' could not be found
```

The contents of `subproject/openssl-1.1.1l` are as follow :
```bash
libnvme/.build/subprojects/openssl-1.1.1l$ ls
generated-config  libcrypto.so  libcrypto.so.p  libssl.so  libssl.so.p
```

The build succeeded and unit tests pass
```
libnvme/.build$ meson test
ninja: Entering directory `/home/reds/libnvme/.build'
ninja: no work to do.
1/5 main     OK             1.32s
2/5 main     OK             1.30s
3/5 cpp      OK             1.32s
4/5 register OK             1.29s
5/5 zns      OK             0.05s

Ok:                 5   
Expected Fail:      0   
Fail:               0   
Unexpected Pass:    0   
Skipped:            0   
Timeout:            0  
```

System information : Running on Ubuntu 21.04 with Linux kernel 5.13.0

Best Regards.
Rick

igaw, 2022-02-15:
Thanks for the report. There is something wrong with the subprojects. These are handy for development but I wouldn't recommend for normal installation (unless you know what you are doing). For such cases it's better to depend on the libraries provided by the distro. In your case you should install developer packages for json-c and OpenSSL.

We should update the README on this topic.

rwk-unil, 2022-02-15:
Thanks for the fast reply,
I installed the missing dependencies through my package manager
(`sudo apt install libjson-c-dev libssl-dev`)

The installation worked as intended.

Thanks

closed 2022-02-15 by rwk-unil

igaw, 2022-02-15:
Glad I could help. I'll see if I can reproduce your problem. But this has no high prio :)

glimchb, 2022-03-13:
fixed via https://github.com/linux-nvme/libnvme/pull/277