linux-nvme/libnvme #260: endianity macros: ccan pulled in public nvme header files

tbzatek, 2022-02-28:
Commit fc8073a735da04944cdf5f976183f36bc6928244 made use of `le16_to_cpu()` from `<ccan/endian/endian.h>`:

https://github.com/linux-nvme/libnvme/blob/16abb4449f79ab842aa2b015ec22ee3375fac1dc/src/nvme/util.h#L551-L555

The ccan headers are obviously not installed as part of libnvme and this is the only use of ccan macros in public headers. The question which way to go is rather philosophical. Do other libc's provide `endian.h` like glibc?

martin-belanger, 2022-02-28:
That is a good question. In my initial commit I was using the standard `le16toh()` found in `/usr/include/endian.h`, but I was asked to use the definitions from ccan instead (e.g. `le16_to_cpu()`).

BTW, in an older version of libnvme, there were macros defining `le16_to_cpu()` et al. in `types.h`. Maybe we should go back to that approach instead of using ccan.

igaw, 2022-02-28:
IMO the simplest solution here would be not to provide the implementation in the header file instead move into the util.c file.

closed 2022-02-28 by igaw