linux-nvme/libnvme #263: wrong hostnqn uuid

igaw, 2022-03-01:
The dmi hostnqn uuid code seems broken. I got following downstream bug report:

```
# Wrong hostnqn UUID:
dhcp-10-231-46-32:~ # nvme gen-hostnqn
nqn.2014-08.org.nvmexpress:uuid:38363739-3539-4d58-5138-33333030585a

# Correct hostnqn UUID:
dhcp-10-231-46-32:~ # dmidecode -s system-uuid
39373638-3935-584d-5138-33333030585a

It looks like nvme-cli is not byteswapping the first 4 bytes.
```

igaw, 2022-03-01:
@martin-belanger

igaw, 2022-03-01:
Interesting, dmidecode might or might not swap the first bytes depending on the version of bios:

https://github.com/mirror/dmidecode/blob/024b0fb43ada238659c61f8d2e1f9ddeae603708/dmidecode.c#L440

igaw, 2022-03-01:
and it looks like the kernel already takes care of this. I think we don't have to do any swapping.

https://github.com/torvalds/linux/blob/719fce7539cd3e186598e2aed36325fe892150cf/drivers/firmware/dmi_scan.c#L174

igaw, 2022-03-01:
The Linux code is since 2012-12-20 in place f1d8e614d74b ("drivers/firmware/dmi_scan.c: check dmi version when get system uuid"). So I don't think we can safely rely on it that we get the right byte order.

martin-belanger, 2022-03-01:
The important thing is to get a unique, consistent, host NQN. Whether the bytes are swapped or not should not matter as long as we always get the same NQN for a given host. In fact, there is another way to do exactly that and it is to use the machine-id (`/etc/machine-id`).

Ref: 
https://www.freedesktop.org/software/systemd/man/machine-id.html
http://0pointer.de/blog/projects/ids.html

igaw, 2022-03-01:
Some vendors provision their machine via the this way their setup. Thus it really matters that the UUID is the one which was provisioned in the firmware. We had the systemd's machine-id interface just recently removed because we could fall back using just the kernel. gen-hostnqn is a bit of a misnomer in this regard. 'get-hostnqn' would have been a better name. Well, it's probably too late.

PR #264 (2022-03-01): fabrics: Do not swap bytes for system uuid
The kernel is taking care of swapping the system uuid bytes since
f1d8e614d74b ("drivers/firmware/dmi_scan.c: check dmi version when get
system uuid"). Hence we should not swap them again. Instead just read
/sys/class/dmi/id/product_uuid and store the bytes directly into the
system_uuid buffer.

Fixes: #263
closed 2022-03-01 by igaw