linux-nvme/libnvme #268: libnvme openssl meson subproject compilation fails

glimchb, 2022-03-01:
**This works fine:** ( using pre-installed packages and skips meson subprojects) 
```
docker build https://github.com/linux-nvme/libnvme.git#master -f-<<EOF
FROM fedora:33
RUN dnf install -y gcc g++ cmake libuuid-devel swig python-devel meson
RUN dnf install -y openssl-devel json-c-devel
WORKDIR /root
COPY . .
RUN meson .build && ninja -C .build && meson install -C .build
EOF
```
**This fails:**  ( uses meson subprojects instead of pre-installed packages) 
```
docker build https://github.com/linux-nvme/libnvme.git#master -f-<<EOF
FROM fedora:33
RUN dnf install -y gcc g++ cmake libuuid-devel swig python-devel meson
WORKDIR /root
COPY . .
RUN meson .build && ninja -C .build && meson install -C .build
EOF
```
**With this error:** (looks like installation stage)
```
[753/754] Compiling C object libnvme/_nvme.cpython-39-x86_64-linux-gnu.so.p/meson-generated_.._nvme_wrap.c.o
[754/754] Linking target libnvme/_nvme.cpython-39-x86_64-linux-gnu.so
ninja: Entering directory `/root/.build'
ninja: no work to do.
Traceback (most recent call last):
  File "/usr/lib/python3.9/site-packages/mesonbuild/mesonmain.py", line 228, in run
    return options.run_func(options)
  File "/usr/lib/python3.9/site-packages/mesonbuild/minstall.py", line 720, in run
    installer.do_install(datafilename)
  File "/usr/lib/python3.9/site-packages/mesonbuild/minstall.py", line 512, in do_install
    self.install_targets(d, dm, destdir, fullprefix)
  File "/usr/lib/python3.9/site-packages/mesonbuild/minstall.py", line 615, in install_targets
    raise RuntimeError(f'File {t.fname!r} could not be found')
RuntimeError: File 'subprojects/openssl-1.1.1l/openssl' could not be found
Installing subprojects/json-c-0.15/libjson-c.so to /usr/lib64
Installing subprojects/openssl-1.1.1l/libcrypto.so to /usr/lib64
Installing subprojects/openssl-1.1.1l/libssl.so to /usr/lib64
The command '/bin/sh -c meson .build && ninja -C .build && meson install -C .build' returned a non-zero code: 2
```

I can't figure out what is missing during compile stage that fails during installation...

igaw, 2022-03-02:
Setting `install: false` seems to do the trick for me. The question here what should happen if `build_by_default: false` and `install: true` is set. I haven't found anything related to this in https://mesonbuild.com/Reference-manual_functions.html#executable I think we need to ask the meson project what's the right to do here.

glimchb, 2022-03-02:
I found this small hint https://github.com/mesonbuild/meson/blob/master/docs/yaml/functions/custom_target.yaml#L73
```
      *(since 0.50.0)* If `build_by_default` is explicitly set to false, `install`
      will no longer override it. If `build_by_default` is not set, `install` will
      still determine its default.
```

igaw, 2022-03-02:
https://github.com/mesonbuild/meson/issues/4370

igaw, 2022-03-03:
There are two workarounds until we figure out how to fix it properly:
- either patch meson.build
- build the openssl binary and install it alongside
I think the second approach is better.

glimchb, 2022-03-03:
@igaw I also experimented with this commit to test this use case in CI https://github.com/glimchb/libnvme/commit/0cd38cb6b93be6b992b299b905e3684aa9aa7323

glimchb, 2022-03-03:
@igaw the fix is not working see
```
$ docker build https://github.com/linux-nvme/libnvme.git#master -f-<<EOF
FROM fedora:33
RUN dnf install -y gcc g++ cmake libuuid-devel swig python-devel meson
#RUN dnf install -y openssl-devel json-c-devel
RUN update-ca-trust
WORKDIR /root
COPY . .
RUN meson .build && ninja -C .build && meson install -C .build
```
error
```
[753/754] Compiling C object libnvme/_nvme.cpython-39-x86_64-linux-gnu.so.p/meson-generated_.._nvme_wrap.c.o
[754/754] Linking target libnvme/_nvme.cpython-39-x86_64-linux-gnu.so
ninja: Entering directory `/root/.build'
ninja: no work to do.
Traceback (most recent call last):
  File "/usr/lib/python3.9/site-packages/mesonbuild/mesonmain.py", line 228, in run
    return options.run_func(options)
  File "/usr/lib/python3.9/site-packages/mesonbuild/minstall.py", line 720, in run
    installer.do_install(datafilename)
  File "/usr/lib/python3.9/site-packages/mesonbuild/minstall.py", line 512, in do_install
    self.install_targets(d, dm, destdir, fullprefix)
  File "/usr/lib/python3.9/site-packages/mesonbuild/minstall.py", line 615, in install_targets
    raise RuntimeError(f'File {t.fname!r} could not be found')
RuntimeError: File 'subprojects/openssl-1.1.1l/openssl' could not be found
Installing subprojects/json-c-0.15/libjson-c.so to /usr/lib64
Installing subprojects/openssl-1.1.1l/libcrypto.so to /usr/lib64
Installing subprojects/openssl-1.1.1l/libssl.so to /usr/lib64
The command '/bin/sh -c meson .build && ninja -C .build && meson install -C .build' returned a non-zero code: 2
```

igaw, 2022-03-03:
I've tested it locally and it worked fine. Are you sure you have an updated version of libnvme?

glimchb, 2022-03-03:
sure, I built directly from github to be extra sure `docker build https://github.com/linux-nvme/libnvme.git#master`

igaw, 2022-03-03:
workaround: `meson .build && ninja -C .build && ninja -C .build openssl && meson install -C .build`

glimchb, 2022-03-03:
another workaround: `meson .build && ninja -C .build && meson install -C .build --skip-subprojects=openssl`

glimchb, 2022-03-06:
Should be fixed by https://github.com/mesonbuild/wrapdb/pull/297

PR #273 (2022-03-03): build: Depend on libssl when using the fallback dependency
libnvme doesn't depend on the openssl application. Update the fallback
dependency to libssl.

Fixes: #268
PR #277 (2022-03-08): openssl: update to patch-3 to fix installation issue
Fixes #268

Patch taken from https://mesonbuild.com/Wrapdb-projects.html
Wrap file used https://wrapdb.mesonbuild.com/v2/openssl_1.1.1l-3/openssl.wrap
Fixed by https://github.com/mesonbuild/wrapdb/pull/322
commit 9ea2de7d (2022-03-08): openssl: update to patch-3 to fix installation issue

Fixes #268

Patch taken from https://mesonbuild.com/Wrapdb-projects.html
Wrap file used https://wrapdb.mesonbuild.com/v2/openssl_1.1.1l-3/openssl.wrap
Fixed by mesonbuild/wrapdb#322
closed 2022-03-08 by igaw