linux-nvme/libnvme #271: self: compare between `mesonpy` and `mesonpep517` python build backends

glimchb, 2022-03-02:
learn more and compare:
- https://pypi.org/project/meson-python/
- https://pypi.org/project/mesonpep517/

igaw, 2022-04-11:
We can't upload the binaries with mesonpep517, because we don't do the repair step. This leads to the situation that PyPi says `linux_x86_64` is not supported. The repair step would fix this up to `manylinux_`.

The repair step doesn't work because mesonpep517 is not packaging up libnvme.so only the Python c binding library is in the dist package. There is also no way to workaround in tricking meson to install libnvme.so into the python side-config directory because config_settings is not supported in v0.2. And there was no release for over one year. I guess we can say this project is not ready for this taks

mesonpy is not working either, at least not out of the box and I am running out of steam. I think we can't really upload precompiled binaries at this point. Uploading just the source code to PyPi is currently the only working solution I suppose.

(I've spend way too much time today to figure this out.)

glimchb, 2022-04-12:
thanks @igaw  indeed a lot o work you put into this and I put originally about the same amount of time
I don't object to your decision to remove PyPi integration attempt for now
We can re-introduce it once we figure out how to do it properly

igaw, 2022-04-12:
So the only thing I am still pondering around, if uploading just the sdist would actually make sense nonetheless. I am sure we are not the only one in this situation. Are there any projects which just upload the Python binding?

glimchb, 2022-04-12:
yes, I know many project just upload sdist without `wheels`
This means it will be built during `pip install`
`wheels` is just an optimization

igaw, 2022-04-12:
FWIW, I've uploaded the binding to PyPi manually (yeah, I know, sneaky thing...) but I wanted to get the project name claimed on PyPi before someone else is getting it. So the next release should work through the normal CI means.

glimchb, 2022-04-12:
just FYI,
I was struggling last time with wheels as well, that's why I left `CIBW_REPAIR_WHEEL_COMMAND_LINUX` explicitly empty
I was hoping that using `LD_LIBRARY_PATH` to pint to `libnvme.so` location in `CIBW_REPAIR_WHEEL_COMMAND_LINUX` we can make wheels to work

just documenting here for later... once we come back to this problem...

glimchb, 2022-04-12:
> FWIW, I've uploaded the binding to PyPi manually (yeah, I know, sneaky thing...) but I wanted to get the project name claimed on PyPi before someone else is getting it. So the next release should work through the normal CI means.

Yay. looks good ! https://pypi.org/project/libnvme/

glimchb, 2022-04-12:
just for documentation this looks ok
```
ubuntu@localhost:~$ pip install libnvme
Collecting libnvme
  Downloading libnvme-1.0.tar.gz (337 kB)
     |████████████████████████████████| 337 kB 2.5 MB/s
  Installing build dependencies ... done
  Getting requirements to build wheel ... done
    Preparing wheel metadata ... done
Building wheels for collected packages: libnvme
  Building wheel for libnvme (PEP 517) ... done
  Created wheel for libnvme: filename=libnvme-1.0-cp38-cp38-linux_aarch64.whl size=76671 sha256=3e1555e9be79b9fb2fc32f5c0b9bfb0a1316f14c3a235585c511dce84310a3bc
  Stored in directory: /home/ubuntu/.cache/pip/wheels/ea/33/1f/c50122cba5ecefc0b094cac6511caaeceb3d449696d1c2c122
Successfully built libnvme
Installing collected packages: libnvme
Successfully installed libnvme-1.0
```

igaw, 2022-04-12:
Hmm, I see 'wheel'. Do we need to keep the wheel requeriment in pyproject.toml?

glimchb, 2022-04-12:
non-no-no, this is excellent. Exactly as expected. 
`wheel` is always built on the machine during the `pip install` phase. 
If we upload `wheels` to pypi, then during installation they are just downloaded instead of build in place...

eli-schwartz, 2022-04-13:
> First of, we would need to be able to build libnvme completely statically (not possible at this point, we are missing libuuid as wrap)

https://github.com/util-linux/util-linux/pull/1654

igaw, 2022-04-13:
@eli-schwartz wow! Thanks for helping out. It was on my TODO list and even started with it but got nowhere...

eli-schwartz, 2022-04-13:
It is now merged upstream, please give it a try.

glimchb, 2022-04-13:
something like this https://github.com/linux-nvme/libnvme/pull/349 ? not sure...