linux-nvme/libnvme #287: Versioning : git tags, release, meson project version and library version

glimchb, 2022-03-13:
### what is the right thing to do ?

let's say today we are at https://github.com/linux-nvme/libnvme/releases/tag/v1.0-rc6

today there are too many places to update:
- git tag and github release
- meson project version
- library version (taken from above)
- python package version

when uploading to pypi we can't override the same `1.0` again and again. 
See https://test.pypi.org/help/#file-name-reuse 

so we want to add `rc` and treat them as pre-release
see example here https://pypi.org/project/OctoPrint/#history

also see more discussions and `systemd` and `dpdk` versioning here https://github.com/linux-nvme/libnvme/issues/95

igaw, 2022-03-17:
Thanks for bringing this up. I am very reluctant to change anything on the name scheme for this release cycle. Any change now will likely cause more harm then it fixes. But sure, for the next iteration I am open for changes. I need to dive into the topic once more to understand what options are there. 

One thing I had in mind (which is what systemd does/did?) is to increase the version number in the meson.build file right after the current release. This makes sure any new build is showing the not yet released version number. I saw some projects hack around this by adding a script which generates new version numbers (based on the git hash). Though this seems to be discouraged by the meson documents. When I played with it I a found it is indeed tricky and not all workflows will work (creating the tar release file via meson was one of those problem areas).

igaw, 2022-07-07:
BTW, the git version is now added to the 'nvme --version' output.

```
$ .build/nvme --version
nvme version 2.0 (git 2.0-162-g9f344eb)
libnvme version 1.0 (git 1.0-106-g72121c8)
```

- 1f54f1536bbd ("nvme: Add git ref to the binary")
- 1fde82bbd3c0 ("nvme: output also libnvme version")

igaw, 2022-07-07:
Ideally we can reduce the number of 'sources' a bit now. IIRC, it's possible to use the meson.project_version() everywhere inside the tree. That means only one place to set the version to '1.1' for example. The git tag is obviously outside of the meson world. Maybe we could just have a small release script which updates the version number, sets the tag correctly and uploads the result to github. I have done this in other small projects and it worked pretty good and gave consistent results. Also it is a way to document what to do :)

PR #424 (2022-07-12): Add release script
Automate the release with a script. This ensures we always do the same steps for an release. It also documents how to do it.

Fixes: #287
closed 2022-07-12 by igaw