linux-nvme/libnvme #291: [loop] Invalid subsystem records after scanning

tbzatek, 2022-03-14:
This is somewhat similar to #154 except this fails with just a single connection. We've been using the loop transport for simple setups in our CI and found this via segfaulting nvme-cli:

Steps to reproduce:
1. Setup a simple target using the loop transport: [target.json](https://github.com/linux-nvme/libnvme/files/8246505/target.txt)
2. `# nvme connect --transport=loop --hostnqn=test_hostnqn --nqn=test_subnqn`
3. `# nvme list --verbose`
```
Subsystem        Subsystem-NQN                                                                                    Controllers
---------------- ------------------------------------------------------------------------------------------------ ----------------
nvme-subsys0     test_subnqn                                                                               
(null)           test_subnqn                                                                                      nvme0

Device   SN                   MN                                       FR       TxPort Address        Subsystem    Namespaces      
-------- -------------------- ---------------------------------------- -------- ------ -------------- ------------ ----------------
nvme0    d55ce7cdd937df5c7dea Linux                                    5.17.0-0 loop                  (null)       

Device       Generic      NSID     Usage                      Format           Controllers     
------------ ------------ -------- -------------------------- ---------------- ----------------
nvme0n2      ng0n2        2          1.07  GB /   1.07  GB    512   B +  0 B   
nvme0n1      ng0n1        1          1.07  GB /   1.07  GB    512   B +  0 B   
```
4. `# nvme list --output-format=json --verbose`
```
Segmentation fault (core dumped)
```
(this fails on `nvme_subsystem_get_name()` returning `NULL`)
```
#0  0x00007ffff7ed7dbd in __strlen_avx2 () from /lib64/libc.so.6
#1  0x00007ffff7f78e2d in json_object_new_string (s=0x0) at /usr/src/debug/json-c-0.15-3.fc36.x86_64/json_object.c:1290
#2  0x000000000043e8e5 in json_detail_list (r=0x4d4be0) at ../nvme-print.c:7102
```

I haven't looked at where this actually comes from, reporting early.

tbzatek, 2022-03-21:
Alright, managed to find several issues, my unique hostnqn being part of the story, will post details tomorrow.

tbzatek, 2022-03-22:
So the major source of discrepancy and the reason for duplicate records is a default hostnqn (fallback-generated) when scanning subsystems in `nvme_scan_subsystem()` as there's no `/sys/class/nvme-subsystem/nvme-subsys0/hostnqn` file. Previously scanned controllers from a first step in `nvme_scan_topology()` have the right hostnqn and are mismatched for that reason.

First look on the code doesn't ring a bell to me for a convenient fix right away.

hreinecke, 2022-03-30:
Works with latest upstream; this is probably a fallout of the missing upstream commit 0e5e1e316d5c05c7227dafc8e3a0531a666f6ab7.

tbzatek, 2022-03-30:
Nope, still doesn't work with the git master. Let me rephrase the conditions:
1. no `/etc/nvme/hostnqn` file (on purpose)
2. hostqnq supplied to `nvme connect` differs from the default one generated by `nvmf_hostnqn_generate()`

In this case `nvme_scan_ctrl()` picks up the actual `/sys/class/nvme/nvme0/hostnqn` value properly, however `nvme_scan_subsystem()` would fall back to the default as there's no `/sys/class/nvme-subsystem/nvme-subsys0/hostnqn`. This is the point where discrepancy is created - see the output of [`nvme list --output-format=json --verbose`](https://github.com/linux-nvme/libnvme/files/8380041/nvme_list_verbose.txt)

Although a corner case, the kernel will cope with this setup just fine and I believe it should be supported by the library as well.

hreinecke, 2022-03-30:
Gnaa. Of course there is no 'hostnqn' setting under nvme subsystem.

hreinecke, 2022-03-30:
Can you check PR #324 ? That should fix this issue (if I did everything correct, that is ...)

tbzatek, 2022-03-30:
> Can you check PR #324 ? That should fix this issue (if I did everything correct, that is ...)

Confirming, this fixes my issues, thanks!

hreinecke, 2022-03-31:
Fine. I've merged the (updated) PR now.

closed 2022-03-31 by hreinecke