linux-nvme/libnvme #303: python.platlibdir and python.install_env are mutually exclusive

igaw, 2022-03-22:
https://github.com/linux-nvme/libnvme/actions/runs/2021673441

glimchb, 2022-03-23:
let me check

glimchb, 2022-03-23:
I noticed `python3.8` vs `Python/3.10.2` looking into that

glimchb, 2022-03-23:
@igaw i have 2 proposal fixes... testing them now

igaw, 2022-03-23:
Cool thanks. Personally, the first version looks better to my eyes because it depends on less special stuff (e.g. not using pipx (?)).

glimchb, 2022-03-23:
@igaw found it, the problem was with newly (2days ago) release [meson `0.62` version](https://github.com/mesonbuild/meson/releases/tag/0.62.0)

eli-schwartz, 2022-03-23:
But why are they both being specified?

glimchb, 2022-03-23:
@eli-schwartz bug in https://pypi.org/project/mesonpep517/
I opened #271 evaluate switch to https://pypi.org/project/meson-python/

eli-schwartz, 2022-03-23:
If you are so positive it's a bug in that project, maybe you could answer my question by describing the bug? :)

glimchb, 2022-03-23:
@eli-schwartz you are right, I didn't investigate that project and didn't see an error in the code
we can keep this issue open but merge the PR to allow CI to pass

glimchb, 2022-04-12:
fixed by https://github.com/mesonbuild/meson/issues/10181 @igaw you can close

igaw, 2022-04-13:
Okay, so we just have to wait for the next release of meson to drop the version requirement in pyproject.toml.

PR #311 (2022-03-24): CI: fixing dist packaging error
Fixes #303

python.platlibdir and python.install_env are mutually exclusive
commit c4296ed1 (2022-03-24): CI: fixing dist packaging error

Fixes #303

python.platlibdir and python.install_env are mutually exclusive
closed 2022-04-13 by igaw