linux-nvme/libnvme #316: Should kato be set by default when setting persistent?

martin-belanger, 2022-03-28:
The function `nvme_ctrl_set_persistent()` sets the boolean `c->persistent` as follows:
```
void nvme_ctrl_set_persistent(nvme_ctrl_t c, bool persistent)
{
	c->persistent = persistent;
}
```
From what I can tell, the boolean `c->persistent` is not really used for anything (at least from the perspective of configuring the kernel). Should we use `c->persistent` to set a default KATO value for persistent connections to discovery controllers? 

For example:
```
void nvme_ctrl_set_persistent(nvme_ctrl_t c, bool persistent)
{
	c->persistent = persistent;
	if (c->persistent && c->discovery_ctrl && c->cfg.keep_alive_tmo == 0) {
		c->cfg.keep_alive_tmo = DEFAULT_KATO; /* Default could be 120 sec (as in base specs) or any value we agree on */
	}
}
```

igaw, 2022-03-28:
You mean we should move `set_discovery_kato` into libnvme?

martin-belanger, 2022-03-28:
> You mean we should move `set_discovery_kato` into libnvme?

Not exactly. I guess I don't see what `nvme_ctrl_set_persistent()` is used for. All this does is set a boolean, but this boolean is not used to change the way we configure the kernel. It's just an internal boolean that one can set or read. Maybe the `persistent` boolean should not even exist if it's not going to be used for anything useful.

igaw, 2022-03-28:
Yes, the whole decision if a controller is removed is completely in nvme-cli. 

The only place I found some code around c->persistent is in json_update_attributes()
```
		if (!strcmp("persistent", key_str) &&
		    !nvme_ctrl_is_persistent(c))
			nvme_ctrl_set_persistent(c, true);
		if (!strcmp("discovery", key_str) &&
		    !nvme_ctrl_is_discovery_ctrl(c))
			nvme_ctrl_set_discovery_ctrl(c, true);
```
But that's all. So not really used. It looks like we could drop this bits from the library.

martin-belanger, 2022-03-28:
I will submit a patch that removes it then. Thanks for the feedback.

hreinecke, 2022-03-29:
Ehh ... the only thing the 'persistent' flag does is _not_ removing the discovery controller:
~~~
                               if (e->subtype != NVME_NQN_NVME &&
                                    !persistent) {
                                        nvme_disconnect_ctrl(child);
                                        nvme_free_ctrl(child);
                                }
~~~
Which why it's 'persistent'. How would go about this without the persistent flag?

hreinecke, 2022-03-29:
Just for reference: the persistent discovery controller as described in the spec has no meaning to nvme-cli; we only have one controller type. And the KATO value is being set or modified by the 'keep-alive-tmo' setting, irrespective on whether it's a persistent or non-persistent controller.
From the nvme-cli side a persistent controller is a controller which is _not_ removed when the program exits; that really is everything which matters. And hence we _need_ the '--persistent' flag, as otherwise we wouldn't know whether to remove a controller or not.

hreinecke, 2022-03-29:
Which also means we need this information in libnvme, as otherwise we cannot write it out in the json configuration, and that information would be lost.

igaw, 2022-03-29:
Okay, so libnvme doesn't need this information but nvme-cli. nvme-cli doesn't store this state information itself it relies on libnvme. Hmm, it really sounds like we have too much logic in nvme-cli...

martin-belanger, 2022-03-29:
Thanks for the explanation. I will close the pull request and this issue.

closed 2022-03-29 by martin-belanger