linux-nvme/libnvme #328: ZNS user data vs I/O command set specific namespace data structure

hreinecke, 2022-03-31:
The code base has this:

~~~
        NVME_IDENTIFY_CNS_NS_USER_DATA_FORMAT                   = 0x09,
        NVME_IDENTIFY_CNS_CSI_NS_USER_DATA_FORMAT               = 0x0A,
~~~

but the ZNS spec says:

~~~
4.1.5.3 Identify I/O Command Set specific Namespace data structure (CNS 0Ah, CSI 02h)
~~~

So how should we go about this? Avoid using 'userdata' as it's being obsoleted by the spec?
Keep the existing wording for compability? Add a duplicate / cloned value with the new name?

igaw, 2022-03-31:
I tend to say to replace it with the new name. I don't see nvme-cli using it. And we haven't released the library yet. After that I'd say we have to keep such things as around for backwards compatibility until the next major release.

igaw, 2022-04-08:
Well, I am going to do the 1.0 release now, so it's going to be duplicate/cloned value..

igaw, 2022-07-07:
FWIW, in libnvme-me @jk-ozlabs came up with a nice  solution: Fix the enums and use a define to introduce the compatibility indirection. The nice thing about this is, that we clearly document what is new and what is deprecated. When we are going to do a major update we just going to delete the defines.

igaw, 2022-07-12:
I haven't found the corresponding entries my copy of the spec. Is this in some unreleased TPs?

igaw, 2023-01-13:
The current base spec (2.0c) defines the values `09h` to `0Fh` as reserved (Table 273: Identify - CNS Values). Can't find any matching `4.1.5.3` section in the specs. Where did you see this?

igaw, 2023-01-23:
No feedback. Closing.

closed 2023-01-23 by igaw