linux-nvme/libnvme #355: `SCOPE_DELIMITER` not defined in musl

subnut, 2022-04-16:
```
src/nvme/fabrics.c:328:	char *p = strchr(tmp, SCOPE_DELIMITER);
```

glibc just defines it as so -
```c
#ifdef __USE_GNU
/* Scope delimiter for getaddrinfo(), getnameinfo().  */
# define SCOPE_DELIMITER        '%'
#endif
```

But there is no such definition in musl libc.  
Alpine Linux uses this patch -

https://git.alpinelinux.org/aports/tree/community/libnvme/scope-delimiter.patch
```patch
--- a/src/nvme/fabrics.c
+++ b/src/nvme/fabrics.c
@@ -325,7 +325,7 @@
 		nvme_msg(r, LOG_ERR, "cannot copy: %s\n", src);
 
 	const char *scope = NULL;
-	char *p = strchr(tmp, SCOPE_DELIMITER);
+	char *p = strchr(tmp, '%');
 	if (p) {
 		*p = '\0';
 		scope = src + (p - tmp) + 1;
```

igaw, 2022-04-21:
I think we got this covered by https://github.com/linux-nvme/libnvme/commit/10f9bbd60cc4039cab7063465e3c0db0af598640 already. Or did we miss another one?

subnut, 2022-04-21:
Sorry, I'd forgotten that I had `v1.0` checked out instead of `master`

closed 2022-04-21 by subnut

igaw, 2022-04-21:
No worries. Thanks for double checking!