linux-nvme/libnvme #364: Is nvme_subsystem_lookup_namespace doing the right thing?

igaw, 2022-06-02:
`nvme_subsystem_lookup_namespace()`

```c
	ret = asprintf(&name, "%sn%u", s->name, nsid);
	if (ret < 0)
		return NULL;
```

```gdb
(gdb) p name
$1 = 0x715260 "nvme-subsys1n17"
(gdb) p *s
$2 = {entry = {next = 0x70fbe0, prev = 0x70fbe0}, ctrls = {n = {next = 0x7193f0, prev = 0x70ffc0}},
  namespaces = {n = {next = 0x715100, prev = 0x715100}}, h = 0x70fbd0, name = 0x70fee0 "nvme-subsys1",
  sysfs_dir = 0x70fba0 "/sys/class/nvme-subsystem/nvme-subsys1",
  subsysnqn = 0x70fcd0 "nqn.1988-11.com.dell:powerstore:00:f03028e73ef7D032D81E",
  model = 0x70fdf0 "dellemc-powerstore", serial = 0x70fe10 "85JZX23", firmware = 0x70fe30 "2.1.0.0",
```

I wanted to lookup nsid=17 but this obviously `/dev/nvme-subsys1n17` doesn't exit. Is this function really
doing what's supposed to do?

igaw, 2022-06-02:
@hreinecke

hreinecke, 2022-06-02:
Hmm. The name is obviously wrong; it should've been /dev/nvme<subsys-id>n<nsid>.

hreinecke, 2022-06-10:
It's even worse. Looks as if we would never clear up the list of namespaces, only ever adding to it.
We really need to have a look at that.

PR #375 (2022-06-13): tree: simplifiy nvme_subsystem_lookup_namespace()
The list of namespaces needs to be considered static, and should
only be changed on a full rescan. So simplify the function to just
return the corresponding namespace entry from the topology tree.

Fixes: #364
closed 2022-06-13 by igaw