linux-nvme/libnvme #377: libnvme-mi build failures

igaw, 2022-06-13:
```bash
[42/82] Linking target subprojects/libnvme/src/libnvme-mi.so.1.0.0
FAILED: subprojects/libnvme/src/libnvme-mi.so.1.0.0 
cc  -o subprojects/libnvme/src/libnvme-mi.so.1.0.0 'subprojects/libnvme/src/1edc449@@nvme-mi@sha/nvme_cleanup.c.o' 'subprojects/libnvme/src/1edc449@@nvme-mi@sha/nvme_log.c.o' 'subprojects/libnvme/src/1edc449@@nvme-mi@sha/nvme_mi.c.o' 'subprojects/libnvme/src/1edc449@@nvme-mi@sha/nvme_mi-mctp.c.o' -Wl,--as-needed -Wl,--no-undefined -shared -fPIC -Wl,--start-group -Wl,-soname,libnvme-mi.so.1 subprojects/libnvme/ccan/libccan.a -Wl,--version-script=/root/nvme-cli/subprojects/libnvme/src/libnvme-mi.map /usr/lib64/libuuid.so -Wl,--end-group '-Wl,-rpath,$ORIGIN/../ccan' -Wl,-rpath-link,/root/nvme-cli/.build/subprojects/libnvme/ccan
/usr/lib64/gcc/x86_64-suse-linux/7/../../../../x86_64-suse-linux/bin/ld: subprojects/libnvme/src/1edc449@@nvme-mi@sha/nvme_log.c.o: in function `nvme_identify_cns_nsid':
/root/nvme-cli/.build/../subprojects/libnvme/src/nvme/ioctl.h:496: undefined reference to `nvme_identify'
/usr/lib64/gcc/x86_64-suse-linux/7/../../../../x86_64-suse-linux/bin/ld: subprojects/libnvme/src/1edc449@@nvme-mi@sha/nvme_log.c.o: in function `nvme_get_log_ana':
/root/nvme-cli/.build/../subprojects/libnvme/src/nvme/ioctl.h:1712: undefined reference to `nvme_get_log'
/usr/lib64/gcc/x86_64-suse-linux/7/../../../../x86_64-suse-linux/bin/ld: subprojects/libnvme/src/1edc449@@nvme-mi@sha/nvme_mi.c.o: in function `nvme_identify_cns_nsid':
/root/nvme-cli/.build/../subprojects/libnvme/src/nvme/ioctl.h:496: undefined reference to `nvme_identify'
/usr/lib64/gcc/x86_64-suse-linux/7/../../../../x86_64-suse-linux/bin/ld: subprojects/libnvme/src/1edc449@@nvme-mi@sha/nvme_mi.c.o: in function `nvme_get_log_ana':
/root/nvme-cli/.build/../subprojects/libnvme/src/nvme/ioctl.h:1712: undefined reference to `nvme_get_log'
/usr/lib64/gcc/x86_64-suse-linux/7/../../../../x86_64-suse-linux/bin/ld: subprojects/libnvme/src/1edc449@@nvme-mi@sha/nvme_mi-mctp.c.o: in function `nvme_identify_cns_nsid':
/root/nvme-cli/.build/../subprojects/libnvme/src/nvme/ioctl.h:496: undefined reference to `nvme_identify'
/usr/lib64/gcc/x86_64-suse-linux/7/../../../../x86_64-suse-linux/bin/ld: subprojects/libnvme/src/1edc449@@nvme-mi@sha/nvme_mi-mctp.c.o: in function `nvme_get_log_ana':
/root/nvme-cli/.build/../subprojects/libnvme/src/nvme/ioctl.h:1712: undefined reference to `nvme_get_log'
```

igaw, 2022-06-13:
FYI @jk-ozlabs

igaw, 2022-06-13:
The obvious fix is:

```diff
@@ -82,7 +82,7 @@ libnvme_mi = library(
     link_depends: mi_mapfile,
     include_directories: [incdir, internal_incdir],
     install: true,
-    link_with: libccan,
+    link_with: [libccan, libnvme]
 )
```

igaw, 2022-06-13:
I suppose it should be address by adding it to `dependencies`.

closed 2022-06-13 by igaw

jk-ozlabs, 2022-06-13:
> The obvious fix is:

No, we shouldn't need libnvme for libnvme-mi; the only shared components are the headers (hence some of the earlier questions about creating the `root_t`, and whether we have a mi-specific one).

Glad that this has been fixed, but I didn't see those errors when compiling here; is there some build configuration specific to this issue?

igaw, 2022-06-13:
The problem appeared when linking nvme-cli against libnvme using subprojects.

igaw, 2022-06-13:
I did expect some fallouts, so no worries. Usually we would see this build errors as soon we update libnvme.wrap in nvme-cli. But this time I was faster :)

PR #379 (2022-06-13): ioctl: Add inline to nvme_identify_cns_nsid and nvme_get_log_ana
Both function should always be inlined.

Fixes: #377