linux-nvme/libnvme #401: Figure out if readthedocs could run the `.rst.in` to `.rst` step.

igaw, 2022-06-23:
Maybe it is possible to trick readthedocs to run the `.rst.in` to `.rst` step. Then we could drop these files from the source tree.

_Originally posted by @igaw in https://github.com/linux-nvme/libnvme/issues/399#issuecomment-1164105438_

martin-belanger, 2022-11-15:
I did exactly that for nvme-stas. The trick is to provide a `.readthedocs.yaml` ([linux-nvme/nvme-stas/.readthedocs.yaml](https://github.com/linux-nvme/nvme-stas/blob/main/.readthedocs.yaml)). I also created a special meson option (`readthedocs=true`) so that the Read-the-Docs bot can build only the documentation and not the whole code. 

With this I no longer have to copy pre-compiled documentation to the source tree.

igaw, 2022-11-16:
Thanks @martin-belanger for the tip with .readthedocs.yaml file. There wasn't much missing to get it going. There was already the 'rst' doc target. readthedocs is now building the docs from the source. Yay!

PR #532 (2022-11-16): doc: Generate rst documentation on the fly for readthedocs
Generate the rst files from the source files directly and avoid storing pre generated ones in the source tree.

Signed-of-by: Daniel Wagner <dwagner@suse.de>

Fixes: #401