linux-nvme/libnvme #404: Segmentation fault when running from nvme-stas

glimchb, 2022-06-23:
```
Program terminated with signal SIGSEGV, Segmentation fault.
#0 0x00007fe92c53d265 in nvmf_add_ctrl () from /home/mbelanger/work/linux-nvme/nvme-stas/.build/subprojects/libnvme/libnvme/../src/libnvme.so.1
[Current thread is 1 (Thread 0x7fe92b817700 (LWP 38956))]
(gdb) bt
#0 0x00007fe92c53d265 in nvmf_add_ctrl () at /home/mbelanger/work/linux-nvme/nvme-stas/.build/subprojects/libnvme/libnvme/../src/libnvme.so.1
#1 0x00007fe92c561ee6 in _wrap_ctrl_connect () at /home/mbelanger/work/linux-nvme/nvme-stas/.build/subprojects/libnvme/libnvme/_nvme.cpython-38-x86_64-linux-gnu.so
#2 0x00000000005f3a2a in cfunction_call_varargs
(kwargs=<optimized out>, args=<optimized out>, func=<built-in method ctrl_connect of module object at remote 0x7fe92c5c1130>)
at ../Include/internal/pycore_pyerrors.h:13
#3 PyCFunction_Call (func=<built-in method ctrl_connect of module object at remote 0x7fe92c5c1130>, args=<optimized out>, kwargs=<optimized out>)
at ../Objects/call.c:773
#4 0x00000000005f3e1e in _PyObject_MakeTpCall
(callable=<built-in method ctrl_connect of module object at remote 0x7fe92c5c1130>, args=<optimized out>, nargs=<optimized out>, keywords=<optimized out>)
at ../Include/internal/pycore_pyerrors.h:13
#5 0x0000000000570674 in _PyObject_Vectorcall
(kwnames=0x0, nargsf=<optimized out>, args=0x7fe92b819390, callable=<built-in method ctrl_connect of module object at remote 0x7fe92c5c1130>)
at ../Include/cpython/abstract.h:125
#6 _PyObject_Vectorcall (kwnames=0x0, nargsf=<optimized out>, args=0x7fe92b819390, callable=<built-in method ctrl_connect of module object at remote 0x7fe92c5c1130>)
at ../Include/cpython/abstract.h:115
#7 call_function (kwnames=0x0, oparg=<optimized out>, pp_stack=<synthetic pointer>, tstate=0x7fe91c001360) at ../Python/ceval.c:4963
#8 _PyEval_EvalFrameDefault (f=<optimized out>, throwflag=<optimized out>) at ../Python/ceval.c:3469
#9 0x000000000056939a in PyEval_EvalFrameEx
(throwflag=0, f=Frame 0x7fe92b819200, for file /home/mbelanger/work/linux-nvme/nvme-stas/.build/subprojects/libnvme/libnvme/nvme.py, line 310, in connect (self=<ctrl(this=<SwigPyObject at remote 0x7fe92c06b330>) at remote 0x7fe92c06b3d0>, h=<host(this=<SwigPyObject at remote 0x7fe92c153fc0>) at remote 0x7fe92c153df0>, cfg={'hdr_digest': False, 'data_digest': False})) at ../Python/ceval.c:741
#10 _PyEval_EvalCodeWithName
(_co=<optimized out>, globals=<optimized out>, locals=<optimized out>, args=<optimized out>, argcount=<optimized out>, kwnames=<optimized out>, kwargs=0x7fe92b816888, kwcount=<optimized out>, kwstep=1, defs=0x7fe92c5a1358, defcount=1, kwdefs=0x0, closure=0x0, name='connect', qualname='ctrl.connect') at ../Python/ceval.c:4298
#11 0x00000000005f6a13 in _PyFunction_Vectorcall (func=<optimized out>, stack=0x7fe92b816870, nargsf=<optimized out>, kwnames=<optimized out>) at ../Objects/call.c:436
#12 0x000000000050aa2c in _PyObject_Vectorcall (kwnames=<optimized out>, nargsf=<optimized out>, args=<optimized out>, callable=<optimized out>)
at ../Include/cpython/abstract.h:127
#13 method_vectorcall (method=<optimized out>, args=0x7fe92c13d918, nargsf=<optimized out>, kwnames=<optimized out>) at ../Objects/classobject.c:89
#14 0x00000000005f3547 in PyVectorcall_Call (kwargs=<optimized out>, tuple=<optimized out>, callable=<method at remote 0x7fe92c093700>) at ../Objects/dictobject.c:1753
#15 PyObject_Call (callable=<method at remote 0x7fe92c093700>, args=<optimized out>, kwargs=<optimized out>) at ../Objects/call.c:228
#16 0x000000000056c8cd in do_call_core
(kwdict=0x0, callargs=(<host(this=<SwigPyObject at remote 0x7fe92c153fc0>) at remote 0x7fe92c153df0>, {'hdr_digest': False, 'data_digest': False}), func=<method at remote 0x7fe92c093700>, tstate=<optimized out>) at ../Python/ceval.c:5010
#17 _PyEval_EvalFrameDefault (f=<optimized out>, throwflag=<optimized out>) at ../Python/ceval.c:3559
#18 0x000000000056939a in PyEval_EvalFrameEx
(throwflag=0, f=Frame 0x7fe92c091440, for file /home/mbelanger/work/linux-nvme/nvme-stas/.build/staslib/gutil.py, line 197, in in_thread_exec (task=<Task at remote 0x7fe92c08c800>, self=<_AsyncCaller(_user_function=<method at remote 0x7fe92c093700>, _user_args=(<host(this=<SwigPyObject at remote 0x7fe92c153fc0>) at remote 0x7fe92c153df0>, {'hdr_digest': False, 'data_digest': False})) at remote 0x7fe92c093880>, task_data=None, cancellable=<Cancellable at remote 0x7fe92c0934c0>, value=<Object at remote 0x7fe92c26c800>)) at ../Python/ceval.c:741
#19 _PyEval_EvalCodeWithName
(_co=<optimized out>, globals=<optimized out>, locals=<optimized out>, args=<optimized out>, argcount=<optimized out>, kwnames=<optimized out>, kwargs=0x7fe92c094528, kwcount=<optimized out>, kwstep=1, defs=0x0, defcount=0, kwdefs=0x0, closure=0x0, name='in_thread_exec', qualname='_AsyncCaller.communicate.<locals>.in_thread_exec')
at ../Python/ceval.c:4298
#20 0x00000000005f6a13 in _PyFunction_Vectorcall (func=<optimized out>, stack=0x7fe92c094508, nargsf=<optimized out>, kwnames=<optimized out>) at ../Objects/call.c:436
#21 0x00000000005f6517 in PyVectorcall_Call (kwargs=0x0, tuple=<optimized out>, callable=<function at remote 0x7fe92c08e0d0>) at ../Objects/call.c:174
#22 PyObject_Call (kwargs=0x0, args=<optimized out>, callable=<function at remote 0x7fe92c08e0d0>) at ../Objects/call.c:228
#23 PyEval_CallObjectWithKeywords (kwargs=0x0, args=<optimized out>, callable=<function at remote 0x7fe92c08e0d0>) at ../Objects/call.c:810
#24 PyObject_CallObject (callable=<function at remote 0x7fe92c08e0d0>, args=<optimized out>) at ../Objects/call.c:818
#25 0x00007fe92d456e4e in () at /usr/lib/python3/dist-packages/gi/_gi.cpython-38-x86_64-linux-gnu.so
#26 0x00007fe92e787e06 in () at /lib/x86_64-linux-gnu/libffi.so.7
#27 0x00007fe92e788188 in () at /lib/x86_64-linux-gnu/libffi.so.7
#28 0x00007fe92d0b8c62 in () at /lib/x86_64-linux-gnu/libgio-2.0.so.0
#29 0x00007fe92d36b374 in () at /lib/x86_64-linux-gnu/libglib-2.0.so.0
```

igaw, 2022-06-24:
Where in nvmf_add_ctrl() do we crash?

martin-belanger, 2022-06-24:
Hi @igaw - I was experimenting with threads in nvme-stas. I think that's where the issue really is and not in libnvme. Please disregard.

glimchb, 2022-06-24:
I'm not sure - should libnvme be thread safe or not?

igaw, 2022-06-24:
libnvme should be thread safe in the sence of only stack variables or in the context handed in). BUT it operates on finite numbers of resources which is not thread safe.