linux-nvme/libnvme #412: Persistent Event Log - Invalid Field Error

tkenbo, 2022-07-01:
Hello, I'm trying to understand if this is an issue or intended.  Hope anybody could enlighten me.

`[root@xxxxxxxx ~]#nvme persistent-event-log /dev/nvme1 -a 0`
`NVMe status: INVALID_FIELD: A reserved coded value or an unsupported value in a defined field(0x2002)`

I traced the command and here is the trace log for the failed command.
` <...>-552510 [031] .... 172168.299315: nvme_setup_cmd: nvme1: qid=0, cmdid=2, nsid=4294967295, flags=0x0, meta=0x0, cmd=(nvme_admin_get_log_page cdw10=0d 00 ff ff ff ff 00 00 00 90 3e 00 00 00 00 00 00 00 00 00 00 00 00 00)`
`<idle>-0     [061] d.h. 172168.299485: nvme_sq: nvme1: qid=0, head=23, tail=23`
`<idle>-0     [061] ..s. 172168.299492: nvme_complete_rq: nvme1: qid=0, cmdid=2, res=0x0, retries=0, flags=0x2, status=0x2002`

looking at the cdw10 (0xffff000d) and cdw11 (0x0000ffff), NUMD is set to 0xFFFFFFFF.  Per our engineer, this value exceeds MDTS and thus it returns the invalid field error.

Looking at the libnvme code, I can see how numd is calculated.
`int nvme_get_log(struct nvme_get_log_args *args)`
`    _u32 numd = (args->len >> 2) - 1;`

In my case, args->len is set to 2, which causes `numd = 0 - 1 = 0xFFFFFFFF`
if len is anything below 3, it makes numd to be 0xFFFFFFFF.

`int nvme_get_log(int fd, __u32 nsid, __u8 log_id, bool rae, __u8 lsp, __u32 data_len, void *data)`
`{`
`	__u32 offset = 0, xfer_len = data_len;`
`	do {`
`		xfer_len = data_len - offset;`
`		ret = nvme_get_log13(fd, nsid, log_id, lsp, offset, 0, rae, xfer_len, ptr);`

Here 
data_len = 0x3E9002
offset = 0x3E9000

Thanks in advance,

igaw, 2022-07-04:
Good catch, the expression `(args->len >> 2) - 1` is not correct for all values of `len`. 

From the spec:
```
Number of Dwords (NUMD): This field specifies the number of dwords of the Reservation Status
data structure to transfer. This is a 0’s based value.
```

I suppose we need to do 

```diff
 int nvme_get_log(struct nvme_get_log_args *args)
 {
-       __u32 numd = (args->len >> 2) - 1;
+       __u32 numd = ((args->len + 3) >> 2) - 1;
        __u16 numdu = numd >> 16, numdl = numd & 0xffff;
 
        __u32 cdw10 = NVME_SET(args->lid, LOG_CDW10_LID) |
@@ -410,6 +410,9 @@ int nvme_get_log(struct nvme_get_log_args *args)
                        NVME_SET(!!args->ot, LOG_CDW14_OT) |
                        NVME_SET(args->csi, LOG_CDW14_CSI);
 
+       if (!args->len)
+               return -EINVAL;
+
        struct nvme_passthru_cmd cmd = {
                .opcode         = nvme_admin_get_log_page,
                .nsid           = args->nsid,
```

igaw, 2022-07-12:
I think we should just ignore the `Total Log Length` range `[0..3)` (in bytes)

```diff
diff --git a/nvme.c b/nvme.c
index 9296ec8651b2..161748a87ffa 100644
--- a/nvme.c
+++ b/nvme.c
@@ -1156,6 +1156,11 @@ static int get_persistent_event_log(int argc, char **argv,
 
        if (!cfg.log_len && cfg.action != NVME_PEVENT_LOG_EST_CTX_AND_READ) {
                cfg.log_len = le64_to_cpu(pevent->tll);
+               if (cfg.log_len < 4) {
+                       fprintf(stderr, "persistent event log total length is small %d\n",
+                               cfg.log_len);
+                       goto free_pevent;
+               }
        } else if (!cfg.log_len && cfg.action == NVME_PEVENT_LOG_EST_CTX_AND_READ) {
                printf("Establishing Persistent Event Log Context\n");
                goto free_pevent;
```

closed 2022-07-19 by igaw