linux-nvme/libnvme #417: Return codes used in libnvme-mi

igaw, 2022-07-06:
> Another point which is independent to this patch is the error codes and return values: The libnvme API returns 0 on success on -1 on error with errno set to the error code. I think we should do the same with the mi.h API. Having two different styles is not really good IMO.

So this is a good point, and does highlight one difference in the MI interface: we may have failures that are due to the transport itself, rather than the command.

Because of that, the API so far has (as yet unwritten) semantics for the error return values:

- 0 for success
- negative return value (negative errno if possible) for transport-reported failures
- positive return value (always the MI status value) for MI-device reported failures

I'm happy to document that, but I'd like to first confirm whether that's suitable in general (I'd just propose a small tidy up: for the second case to also set `errno`). If not, we can move discussion to a bug for an alternative convention.

_Originally posted by @jk-ozlabs in https://github.com/linux-nvme/libnvme/issues/415#issuecomment-1175692629_

igaw, 2022-07-06:
For libnvme we decided to go with the convention from glibc, e.g.

```
 * Return: The nvme command status if a response was received (see
 * &enum nvme_status_field) or -1 with errno set otherwise.
```

I think this is well understood practice and there aren't any nasty surprises. So I would prefer if libnvme-mi could do the same.

jk-ozlabs, 2022-07-06:
Oh, super - that sounds just like #1, but instead of `enum nvme_status_field` we would use the status values from the MI spec (section 4.1.2, response values table in Figure 27).

I could add an enum for those MI status values, and then document accordingly. Would that work?

igaw, 2022-07-06:
The `enum nvme_status_field` was just there from my copy&paste, so pure accidentally.  But yes it makes sense to add a new enum for this.

jk-ozlabs, 2022-07-07:
Oh, turns out I already added this for More Processing Required support:

```c
enum nvme_mi_resp_status {
	NVME_MI_RESP_SUCCESS = 0x00,
	NVME_MI_RESP_MPR = 0x01,
	NVME_MI_RESP_INTERNAL_ERR = 0x02,
	NVME_MI_RESP_INVALID_OPCODE = 0x03,
	NVME_MI_RESP_INVALID_PARAM = 0x04,
	NVME_MI_RESP_INVALID_CMD_SIZE = 0x05,
	NVME_MI_RESP_INVALID_INPUT_SIZE = 0x06,
	NVME_MI_RESP_ACCESS_DENIED = 0x07,
};
```

So, I will:

 * audit the current functions to ensure the return values are consistent with this approach
 * set `errno` when we have an internal error; and
 * document these conventions in the kernel-doc comments

Feel free to assign the bug to me, I'll reference in any PRs.

closed 2022-07-07 by igaw