linux-nvme/libnvme #426: dhchap_key fails to apply at port level of config JSON file

martin-gpy, 2022-07-12:
Turns out specifying dhchap_key in the config JSON file works only if specified at the host level in the config JSON file, and not at the corresponding port level. 

For e.g. this works:

```
# cat /home/config-success.json
[
  {
    "hostnqn":"nqn.2014-08.org.nvmexpress:uuid:6d6b4ca0-8bed-4bdb-a28e-07ea8ed0acf7",
    "hostid":"17503367-6ae0-4b0b-9f9b-113daf169556",
    "dhchap_key":"DHHC-1:00:QaxYTIQLMUZOE+\/dmIForZfP5Mf9OX0Hiy8+uhG8kdOjrVxN:",
    "subsystems":[
      {
        "nqn":"nqn.1992-08.com.netapp:sn.94d3e4c5f3ed11eca5871c666dbe2b2a:subsystem.ss_tcpnvme_2_1",
        "ports":[
          {
            "transport":"tcp",
            "traddr":"10.228.36.89",
            "host_traddr":"10.228.32.190",
            "trsvcid":"4420"
          }
        ]
      }
    ]
  }
]
```

But this fails:

```
# cat /home/config-fail.json
[
  {
    "hostnqn":"nqn.2014-08.org.nvmexpress:uuid:6d6b4ca0-8bed-4bdb-a28e-07ea8ed0acf7",
    "hostid":"17503367-6ae0-4b0b-9f9b-113daf169556",
    "subsystems":[
      {
        "nqn":"nqn.1992-08.com.netapp:sn.94d3e4c5f3ed11eca5871c666dbe2b2a:subsystem.ss_tcpnvme_2_1",
        "ports":[
          {
            "transport":"tcp",
            "traddr":"10.228.36.89",
            "host_traddr":"10.228.32.190",
            "trsvcid":"4420",
            "dhchap_key":"DHHC-1:00:QaxYTIQLMUZOE+\/dmIForZfP5Mf9OX0Hiy8+uhG8kdOjrVxN:"
          }
        ]
      }
    ]
  }
]
```

martin-gpy, 2022-07-19:
Turns out the dhchap key fails to apply at the subsystem level too. For e.g, this too fails:

```
# cat /etc/nvme/config-subsystem-fail.json
[
  {
    "hostnqn":"nqn.2014-08.org.nvmexpress:uuid:6d6b4ca0-8bed-4bdb-a28e-07ea8ed0acf7",
    "hostid":"17503367-6ae0-4b0b-9f9b-113daf169556",
    "subsystems":[
      {
        "nqn":"nqn.1992-08.com.netapp:sn.94d3e4c5f3ed11eca5871c666dbe2b2a:subsystem.ss_tcpnvme_2_1",
        "dhchap_key":"DHHC-1:00:QaxYTIQLMUZOE+\/dmIForZfP5Mf9OX0Hiy8+uhG8kdOjrVxN:",
        "ports":[
          {
            "transport":"tcp",
            "traddr":"10.228.36.89",
            "host_traddr":"10.228.32.190",
            "trsvcid":"4420"
          }
        ]
      }
    ]
  }
]
```

martin-gpy, 2022-07-19:
The current JSON config schema at https://github.com/linux-nvme/libnvme/blob/master/doc/config-schema.json clearly illustrates that the dhchap keys should be applicable at a port level too, but currently this is not possible as stated above.

Coming to the use case for this requirement - one should have the flexibility to specify different dhchap keys for different subsystems/ports. With the above limitation, this is not currently possible in the config JSON file.

Basically, if we have one host (i.e one hostnqn) connecting to multiple subsystems, the config JSON file should be flexible enough to permit different dhchap keyps for different subsystems.

igaw, 2022-07-26:
> The current JSON config schema at https://github.com/linux-nvme/libnvme/blob/master/doc/config-schema.json clearly illustrates that the dhchap keys should be applicable at a port level too, but currently this is not possible as stated above.

The first example is not correct according the config-schema.json but works and the second one is correct according the schema but fails.

hreinecke, 2022-07-26:
Ah. Well. Yes. Probably a bit too much future design wishing happening here.
The original idea indeed was to have the entries applicable to several levels, with lower levels overwriting values passed in from the upper levels (some sort of poor mans object inheritance).
But of course, I never got around to actually implement that.
So for now I'd vouch for updating the json schema to reflect what's implemented, and reflect upon how we could implement that in the future.

hreinecke, 2022-07-29:
Ah, no, it's even simpler.

The json schema declares:
~~~
		"dhchap_key": {
		    "description": "Host DH-HMAC-CHAP key",
		    "type": "string"
		},
		"dhchap_ctrl_key": {
		    "description": "Controller DH-HMAC-CHAP key",
		    "type": "string"
		},
~~~
but the json parser just checks for 'dhchap_key', and uses _that_ for the controller key.
So the 'dhchap_ctrl_key' setting is never applied.

martin-gpy, 2022-08-01:
It is actually https://github.com/linux-nvme/libnvme/issues/438 that is fixed with the above https://github.com/linux-nvme/libnvme/pull/450/commits/51eac082acecc3d418d081047f4aedbff2762838, and not this current issue. 

So could you please reopen this?

igaw, 2022-08-02:
I do not follow. #438 and #450 are merged.

I haven't merged #451 as it needs a little bit more work. Is this the PR you are referring to?

martin-gpy, 2022-08-02:
> I do not follow. #438 and #450 are merged.
> 
> I haven't merged #451 as it needs a little bit more work. Is this the PR you are referring to?

Let me clarify - https://github.com/linux-nvme/libnvme/issues/438 is addressed with https://github.com/linux-nvme/libnvme/pull/450. So that can be marked as closed (which you already did). So no concerns there.

But this current issue (i.e. https://github.com/linux-nvme/libnvme/issues/426) is still valid. I guess we need #451 to completely address this. So requesting you to reopen this.

igaw, 2022-08-02:
Sure, this issue got closed by github because #450 has the 'Fixes' in the description. Let's reopen this isuse and link it #451.

PR #461 (2022-08-25): Allow to get and set DHCHAP host key on controller level
The json config schema already defines a DHCHAP host key on the controller level, so add the infrastructure for parsing and setting it. And while we're at it fixup the json config schema to define the DHCHAP key on the host level.

Fixes: #426
closed 2022-08-25 by igaw