linux-nvme/libnvme #429: pypi workflow should not upload release candidates

igaw, 2022-07-14:
The current workflow rule

```yaml
  upload_pypi:
    needs: [build_sdist]
    runs-on: ubuntu-latest
    if: startsWith(github.ref, 'refs/tags/v')
```

uploads release candidates and marks them as normal release. E.g. v1.1-rc0 was uploaded as v1.1. 

@glimchb

glimchb, 2022-07-16:
@igaw 
> The `[{a|b|rc}N]` part indicates pre-release versions, with a indicating an alpha release, b a beta release, and rc a “release candidate”. 

https://pythonpackaging.info/07-Package-Release.html

why https://github.com/linux-nvme/libnvme/commit/6c5aedd78690b042250f3dc82b48b7302ea91dbd doesn't have RC in the version name ?

igaw, 2022-07-18:
https://semver.org/#spec-item-9

igaw, 2022-07-18:
So at least we need this here
```diff
diff --git a/meson.build b/meson.build
index f64bd7ad78cb..472172d4de3f 100644
--- a/meson.build
+++ b/meson.build
@@ -18,7 +18,8 @@ project(
     ]
 )
 
-library_version = meson.project_version() + '.0'
+maj_min = meson.project_version().split(
```

glimchb, 2022-07-18:
you can consider updating `library_version` also from the `release.sh` or parse regex in the `meson.build` like suggested in the example above... not sure what is better

PR #433 (2022-08-01): Add support for uploading release candidates to PyPI
This changes slightly how releases are done. It's possible now to have the `-rc[0-9]+` suffix added to the project version. 

Fixes #429
closed 2022-08-01 by igaw