linux-nvme/libnvme #430: Add LibreSSL support

ismaell, 2022-07-15:
As of LibreSSL 3.5.3, it doesn't provide `<openssl/core_names.h>`. Build error:

    samu: job failed: ccache cc -Isrc/libnvme.so.1.0.0.p -Isrc -I../src -I. -I.. -Iccan -I../ccan -Iinternal -I../internal -I/usr/include/uuid -I/usr/include/json-c -fdiagnostics-color=always -D_FILE_OFFSET_BITS=64 -Wall -Winvalid-pch -std=gnu99 -Os -g -fomit-frame-pointer -D_GNU_SOURCE -include internal/config.h -march=native -mtune=native -m64 -pipe -fPIC -ffast-math -funroll-loops -Os -fPIC -MD -MQ src/libnvme.so.1.0.0.p/nvme_linux.c.o -MF src/libnvme.so.1.0.0.p/nvme_linux.c.o.d -o src/libnvme.so.1.0.0.p/nvme_linux.c.o -c ../src/nvme/linux.c
    ../src/nvme/linux.c:26:10: fatal error: openssl/core_names.h: No such file or directory
       26 | #include <openssl/core_names.h>
          |          ^~~~~~~~~~~~~~~~~~~~~~

igaw, 2022-07-18:
LibreSSL announces itself as OpenSSL 3.x. Hence meson will set the 
CONFIG_OPENSSL_3 define.

```shell
Determining dependency 'openssl' with pkg-config executable '/usr/bin/pkg-config'
env[PKG_CONFIG_PATH]: /tmp/libressl/lib64/pkgconfig
Called `/usr/bin/pkg-config --modversion openssl` -> 0
3.6.0
env[PKG_CONFIG_PATH]: /tmp/libressl/lib64/pkgconfig
Called `/usr/bin/pkg-config --cflags openssl` -> 0
-I/tmp/libressl/include
env[PKG_CONFIG_PATH]: /tmp/libressl/lib64/pkgconfig
env[PKG_CONFIG_ALLOW_SYSTEM_LIBS]: 1
Called `/usr/bin/pkg-config --libs openssl` -> 0
-L/tmp/libressl/lib64 -lssl -lcrypto
env[PKG_CONFIG_PATH]: /tmp/libressl/lib64/pkgconfig
Called `/usr/bin/pkg-config --libs openssl` -> 0
-L/tmp/libressl/lib64 -lssl -lcrypto
Run-time dependency openssl found: YES 3.6.0
```

When I define CONFIG_OPENSSL_1 instead of CONFIG_OPENSSL_3, the build is successful. 

Obviously, the meson file is written with OpenSSL in mind. The only way I see is to have
a configure flag added where you can force which version of the API is used. Or is it possible to detect that we are actually using LibreSSL?

ismaell, 2022-07-18:
> Or is it possible to detect that we are actually using LibreSSL?

LibreSSL provides `LIBRESSL_VERSION_NUMBER` in `openssl/opensslv.h`.

igaw, 2022-07-18:
I don't see how we can express this via meson's `dependency()`

I suppose it's possible to use  `cc.compiles` to figure out LIBRESSL_VERSION_NUMBER is present and have this incorporated into the current set logic.

ismaell, 2022-07-18:
I've decided to go with just checking for the header in case libressl implements it at some point.

ismaell, 2022-07-22:
Solved by commit 8a7cf44be48066ca2a52de5ca9562931963ef651.

closed 2022-07-22 by ismaell