linux-nvme/libnvme #437: `nvme_get_nsid_log` ignores `rae` parameter, other log helpers don't.

jk-ozlabs, 2022-07-20:
The `nvme_get_nsid_log()` helper has a `rae` (Retain Asynchronous Events) parameter, but we currently ignore this when constructing the Get Log Page command arguments:

https://github.com/linux-nvme/libnvme/blob/11337761f72958be024a1d13b3d617f9b711b4ef/src/nvme/ioctl.h#L1229-L1249

Although the fix is fairly trivial (`.rae = rae`), there may be some consequences where API users are currently passing `true` (which is a reasonable default) for event logs that do not contain asynchronous events. From the NVMe Base spec:

> Host software should clear this bit to ‘0’ for log pages that are not used with Asynchronous Events.

\- and we do seem to have have `get_log` helpers that take (and pass) a `rae` argument for non-rae log pages (eg `nvme_get_log_fw_slot`).

This might warrant a bit of an audit of those helpers, to see which ones might need to ignore the passed `rae` value.

igaw, 2022-07-20:
Here is a list for all nvme-cli users of the helpers which set `rae=true`.

- nvme_get_log_supported_log_pages

- nvme_get_log_error
  nvme.c: err = nvme_get_log_error(fd, cfg.log_entries, true, err_log);

- nvme_get_log_smart
  plugins/innogrit/innogrit-nvme.c:       nvme_get_log_smart(fd, cfg.namespace_id, true, &smart_log);
  plugins/seagate/seagate-nvme.c: err = nvme_get_log_smart(fd, 0xffffffff, true, &smart_log);
  plugins/transcend/transcend-nvme.c:     result = nvme_get_log_smart(fd, 0xffffffff, true, &smart_log);
  plugins/virtium/virtium-nvme.c: ret = nvme_get_log_smart(fd, NVME_NSID_ALL, true, &smart.raw_smart);
  plugins/wdc/wdc-nvme.c: ret = nvme_get_log_smart(fd, cfg.namespace_id, true, &smart_log);

- nvme_get_log_fw_slot
  nvme.c: err = nvme_get_log_fw_slot(fd, true, &fw_log);
  plugins/micron/micron-nvme.c:    if (nvme_get_log_fw_slot(fd, 1, &fw_log) == 0) {
  plugins/virtium/virtium-nvme.c: ret = nvme_get_log_fw_slot(fd, true, &header.raw_fw);
  plugins/wdc/wdc-nvme.c: ret = nvme_get_log_fw_slot(fd, true, &fw_log);

- nvme_get_log_mi_cmd_supported_effects

- nvme_get_log_reservation
  nvme.c: err = nvme_get_log_reservation(fd, true, &resv);

- nvme_get_log_sanitize
  nvme.c: err = nvme_get_log_sanitize(fd, cfg.rae, &sanitize_log);

Hmpf, it looks a bit of copy & paste programming...

igaw, 2022-07-20:
And yes, I think we should fix this. The good thing is that any binary already build with the `rae=false` in the inline will not change it's behavior, when we change it to `rae=rae`. So any newly build against the fixed header will need to be fixed accordingly. 

So this means we should review the nvme-cli now and figure out if we need to change to `rae=false` in specific cases.

jk-ozlabs, 2022-07-21:
Yep, sounds good. My issue with doing that is that the spec doesn't seem very clear on which log pages the `rae` field actually applies to. Unless I'm missing something, there seem to be various suggestions in the spec, but nothing definitive.

Is there someone who could help with interpreting those parts of the spec?

igaw, 2022-07-21:
The section `5.2 Asynchronous Event Request command`is talking about `rae`.

I think these callers should be changed to `rae=false` if I read the spec correct.
- nvme_get_log_error
- nvme_get_log_smart

Found nothing about rae for these here:
- nvme_get_log_fw_slot
- nvme_get_log_reservation
- nvme_get_log_sanitize

To be on the safe side we could just set them all to false... If no one speaks up we just do this.

PR #440 (2022-07-25): ioctl: Honor rae in nvme_get_nsid_log
The nvme_get_nsid_log() helper has a rae (Retain Asynchronous Events)
parameter, but we currently ignore this when constructing the Get Log
Page command arguments.

This is a behavior change and not an API change. But the good thing is
only for newly build binaries because all helpers are static inline
fuctions.

Reported-by: Reported-by: Jeremy Kerr <jk@codeconstruct.com.au>

Fixes: #437 

BTW, I've decided against updating `test/test.c`, as I am not sure if this
was intended setting or just set without a lot of thinking.
closed 2022-07-25 by igaw