linux-nvme/libnvme #444: ZNS Read BUG

moradicomputer, 2022-07-23:
I tested some ZNS SSD functions on the QEMU emulator.
When I used "nvme_ns_read" for reading a specific address (that data was appended to that), this error was demonstrated
"nvme0n1: I/O Cmd(0x2) @ LBA 2305843009213693951, 0 blocks, I/O Error (sct 0x0 / sc 0x2) DNR".
but with nvme-cli ("nvme read"), I can read my data without any problem. It seems that there is a bug here.

igaw, 2022-07-23:
Which version of nvme-cli do you use?

moradicomputer, 2022-07-24:
I used nvme-cli 1.16

igaw, 2022-10-27:
This is likely to be a duplicate of #485. It's a kernel bug caused by an over eager error reporting code snipped. Should be fixed eventually by https://lore.kernel.org/linux-nvme/20221027150417.1954662-1-kbusch@meta.com/

closed 2022-10-27 by igaw