linux-nvme/libnvme #448: Add endpoint data structure for nvme API

igaw, 2022-07-26:
The MI API uses the endpoint data struct to abstracts from implementation details. Introduce an similiar API for the nvme interface. The goal here is to decouple from the low level API and be able to mock certain APIs for testing.

jk-ozlabs, 2022-08-13:
It would be great if we can use a common struct between the direct and MI interfaces - this means we could avoid duplicating the API for each transport.

I'm happy to propose something for this, if you like.

igaw, 2022-08-16:
Sorry, for the late response. Public holiday :)

Let me have a look at it now. IIRC this work was depending on other work? Anyway, I'l review it now

jk-ozlabs, 2022-08-16:
Yes, this would be a fairly big change - it would alter the arguments for almost every function. Definitely something for a ABI-breaking milestone.

I'll put together some sketches for this.

igaw, 2025-05-26:
I've started to work on libnvme 2. Here are the first bits an early RFC so please have a look and let me know if this is going into the right direct. @jk-ozlabs 

https://github.com/nvme-experiments/libnvme/commits/passthru/

igaw, 2025-05-27:
> Looks good! I would see the next step as unifying the `nvme_passthru_dev`:
> 
> struct nvme_passthru_dev {
> 	int fd;
> };
> 
> into a structure that contains both transport types (ie, ioctl and MI); that would mean we could remove all of the specialisation between transports, and just have one set of functions for all¹ commands. Is that what you're intending here?

Yes, that was my idea. First, I wanted to pass in `nvme_root_t` around but then I figured this wont work, because we have a device for each `ctrl` and `ns` in `tree.c`. So simplest thing is to introduce an abstraction just for the file descriptor. My idea with the `nvme_root_t` was to be able to have a logging context but we can easily also extend the `struct nvme_passthru_dev` to be able to carry  around the logging context. This is for later though.

> Also, any reason it's named 'passthru' ? isn't this primarily the structure that represents our communication facility with the device?

No particular reason other than I wanted to name it clearly what it supposed to do. Though `struct nvme_dev` is nicer after all :)

> 1: We'd still need spearate MI functions for the MI-specific commands, of course

Yes, next step. 

Adding @keithbusch @hreinecke so they are aware what's going on.

jk-ozlabs, 2025-05-28:
> I've updated libnvme-mi to use nvme_dev_t instead of nvme_mi_ctrl_t. There is already a nvme_ctrl_t so we can't use it for the transport abstraction.

OK, sounds good - the single handle here would be great.

However, I think the `nvme_dev_t` is more analogous to a `nvme_mi_ep_t` (rather than a `nvme_mi_ctrl_t`, as that represents the "channel" to the device, like the `fd` for ioctl.

> because libnvme and libnvme-mi are currently not depending on each other.

If we end up with a common `nvme_dev_t`, then maybe it would make sense to merge it all into libnvme, and get rid of libnvme-mi?

igaw, 2025-05-28:
> > I've updated libnvme-mi to use nvme_dev_t instead of nvme_mi_ctrl_t. There is already a nvme_ctrl_t so we can't use it for the transport abstraction.
> 
> OK, sounds good - the single handle here would be great.

Yes, I think that's what we should aim for.

> However, I think the `nvme_dev_t` is more analogous to a `nvme_mi_ep_t` (rather than a `nvme_mi_ctrl_t`, as that represents the "channel" to the device, like the `fd` for ioctl.

Hmm, okay, but then I got it wrong. `nvme_mi_admin_admin_passthru`, `nvme_mi_admin_identify_partial` etc.  are expecting a `nvme_mi_ctrl_t` and not a `nvme_mi_ep_t` argument. If `nvme_dev_t` is equivalent to `nvme_mi_ep_t` than we need first to change those functions signature.  

> > because libnvme and libnvme-mi are currently not depending on each other.
> 
> If we end up with a common `nvme_dev_t`, then maybe it would make sense to merge it all into libnvme, and get rid of libnvme-mi?

If this is okay with you, let's merge these libraries and if someone complains we can still think later on how to split code if necessary (or adding a different 'transport', e.g. xnvme)

igaw, 2025-05-28:
Currently we have this in nvme-cli:

```
struct nvme_dev {
	enum nvme_dev_type type;
	union {
		struct {
			int fd;
			struct stat stat;
		} direct;
		struct {
			nvme_root_t root;
			nvme_mi_ep_t ep;
			nvme_mi_ctrl_t ctrl;
		} mi;
	};

	const char *name;
};
```

jk-ozlabs, 2025-05-28:
OK, sounds sensible!

igaw, 2025-05-28:
I've pushed my current WIP. The first few patches are fairly okay, obviously needs proper reviewing. 

I've marked the last patches as WIP, which is just trying to figure out what could go away in the MI code base. If I got this right, there is plenty of code which gets replaced by the generic API, which is very cool.  I used the big ax, so I expect I've gone a bit too far. 

The code for redirecting to the MI commands is a bit dirty but good enough for testing. I think we need to make this part nicer, but this is all internal, so we can delay cleanups also for later. The most important task is to get the public API in shape.

So, please if check if this is going into the right direction. I haven't figured out what a good way to structure the series. I'd like to have a series which can be actually reviewed. When the changes are done in steps and not everything in one big step, it's also possible to spot errors.

Thus my question, should we first switch over all the test to use the generic API and then remove stuff? Would you like to do this, as you have better knowledge about this part of the library.

jk-ozlabs, 2025-05-30:
This might warrant moving to a separate issue, but has some overlap here: have you considered a v2 API that allows for asynchronous commands?

It's something we'll likely need in MI; but I assume there are similarities in in-band ioctl too. What happens when you issue a long-running command (say, Sanitize) via ioctl? is there any way to do that non-blocking?

igaw, 2025-06-02:
No I haven't thought about this, thanks for bringing it up. Need to look into it.