linux-nvme/libnvme #452: compile error

3735943886, 2022-08-02:
~/libnvme-1.1 $ ninja -C .build
ninja: Entering directory `.build'
[4/39] Compiling C object src/libnvme.so.1.1.0.p/nvme_util.c.o
FAILED: src/libnvme.so.1.1.0.p/nvme_util.c.o
cc -Isrc/libnvme.so.1.1.0.p -Isrc -I../src -I. -I.. -Iccan -I../ccan -Iinternal -I../internal -I/data/data/com.termux/files/usr/include/uuid -I/data/data/com.termux/files/usr/include -I/data/data/com.termux/files/usr/include/json-c -fcolor-diagnostics -D_FILE_OFFSET_BITS=64 -Wall -Winvalid-pch -std=gnu99 -O3 -fomit-frame-pointer -D_GNU_SOURCE -include internal/config.h -fPIC -MD -MQ src/libnvme.so.1.1.0.p/nvme_util.c.o -MF src/libnvme.so.1.1.0.p/nvme_util.c.o.d -o src/libnvme.so.1.1.0.p/nvme_util.c.o -c ../src/nvme/util.c
In file included from ../src/nvme/util.c:20:
../ccan/ccan/endian/endian.h:107:9: warning: '__LITTLE_ENDIAN' macro redefined [-Wmacro-redefined]
#define __LITTLE_ENDIAN 1234
        ^
/data/data/com.termux/files/usr/include/sys/endian.h:39:9: note: previous definition is here
#define __LITTLE_ENDIAN _LITTLE_ENDIAN
        ^
In file included from ../src/nvme/util.c:20:
../ccan/ccan/endian/endian.h:108:9: warning: '__BIG_ENDIAN' macro redefined [-Wmacro-redefined]
#define __BIG_ENDIAN    4321
        ^
/data/data/com.termux/files/usr/include/sys/endian.h:42:9: note: previous definition is here
#define __BIG_ENDIAN _BIG_ENDIAN
        ^
../src/nvme/util.c:746:16: error: use of undeclared identifier 'LINE_MAX'
                char    line[LINE_MAX];
                             ^
2 warnings and 1 error generated.

Seems like that there are no definition of LINE_MAX in source files.
How should I fix it?
Thanks in advance.

igaw, 2022-08-02:
It looks like your libc is not providing not all  `POSIX.2` defines. `LINE_MAX` is supposed to be defined in `unistd.h` which `util.c `actually includes. It's also a bit worrisome that the `ENDIAN` macros are redefined...

Anyway, the `LINE_MAX` value is 2048 on my system. Does this here solve your problem:

```diff
diff --git a/src/nvme/util.c b/src/nvme/util.c
index ff5e0d8ccd60..e45d564a3d5a 100644
--- a/src/nvme/util.c
+++ b/src/nvme/util.c
@@ -23,6 +23,10 @@
 #include "util.h"
 #include "log.h"
 
+#ifndef LINE_MAX
+#define LINE_MAX 2048
+#endif
+
 /* Source Code Control System, query version of binary with 'what' */
 const char sccsid[] = "@(#)libnvme " GIT_VERSION;
```

Not particular a big fan of it but well it's not too bad either.

3735943886, 2022-08-02:
Thank you for prompt answer. 
I compiled it under android OS and there might be some differences of libraries from pure linux system. 
Anyway your suggestion solved my problem and it works flawlessly now. (Redefinition warnings are still noted but it doesn't make any problem atm)
I hope you could please add this modification to project for maximal compatibilities. (Or letting meson script check LINE_MAX definition is possible?)
Thank you.

closed 2022-08-02 by 3735943886

igaw, 2022-08-03:
Android OS, I didn't think about that this. Indeed Android ships with boinic, which is not providing the complete POSIX APIs. If this was the only problem you run into, I think we just add the `define` with a comment.

PR #453 (2022-08-11): util: Add LINE_MAX define
The bionic libc implementation doesn't implement the complete
POSIX API. Apperantly we just need this define to be able
to compile for Android OS. Thus, keep it as simple as possible
and just define it here.

Fixes: #452