linux-nvme/libnvme #454: Incorrect check for discovery controllers

martin-gpy, 2022-08-10:
With TP 8013 - Unique Discovery Controller ID implemented, one can no longer identify a discovery controller by checking whether its subsystem NQN matches the well-known discovery NQN. But looks like there are still multiple places in the libnvme code where this check is done. For e.g.

```
src/nvme/fabrics.c : build_options():

if (!strcmp(nvme_ctrl_get_subsysnqn(c), NVME_DISC_SUBSYS_NAME)) {
     nvme_ctrl_set_discovery_ctrl(c, true);
     discovery_nqn = true;
}
```
```
src/nvme/json.c : json_update_subsys():

/* Skip discovery subsystems as the nqn is not unique */
if (!strcmp(subsysnqn, NVME_DISC_SUBSYS_NAME))
     return;
```
```
src/nvme/tree.c : nvme_init_subsystem():

if (!s->subsystype) {
        if (!strcmp(s->subsysnqn, NVME_DISC_SUBSYS_NAME))
                s->subsystype = strdup("discovery");
        else
                s->subsystype = strdup("nvm");
}
```

martin-gpy, 2022-08-10:
Ideally, one should be checking the subsystype instead (and not the NQN) to identify whether it is a discovery subsystem or not i.e. a subsystype which is not NVME_NQN_NVME (either NVME_NQN_CURR or NVME_NQN_DISC) should qualify as a discovery subsystem.

martin-belanger, 2022-08-10:
Hi @martin-gpy - I did the same investigation a while back and came to these conclusions:

For the first example (`fabrics.c`), this is simply to maintain backwards compatibility with code that is still using the well-known NQN and not explicitly setting the parameter `discovery_ctrl` using the new API `nvme_ctrl_set_discovery_ctrl()`. As you said, TP8013 introduces unique NQNs for DCs. Libnvme users must now explicitly tell the driver when they want a DC by invoking `nvme_ctrl_set_discovery_ctrl()` after creating the controller object. For the legacy cases where an application is still using the well-known NQN, this check in fabrics.c will allow libnvme to work even if the user does not explicitly call `nvme_ctrl_set_discovery_ctrl()`.

The second example (`json.c`) does look suspicious. It seems that the JSON schema is missing a way to specify a discovery controller. Maybe @igaw or @hreinecke can confirm.

The third example (`tree.c`) is similar to the first one where we check the well-known NQN to support legacy code that uses the well-known NQN instead of calling `nvme_ctrl_set_discovery_ctrl()`. In other words, if the NQN happens to be the well-known NQN, then libnvme uses it to identify discovery controllers. If the NQN is not the well-known NQN, then libnvme users MUST invoke `nvme_ctrl_set_discovery_ctrl()` after creating the controller object.

martin-gpy, 2022-08-10:
Fair enough. My only point was that we perhaps are better off having a generic check that works with both TP-8013/8014 changes, while at the same time being backward compatible with legacy cases too.

For e.g. by checking if the subsystype is not 0x02 (i.e. NVME_NQN_NVME), then we should be able to safely conclude that it is a discovery subsystem itself. And that would work with both TP-8013/8014 and legacy scenarios.

igaw, 2022-08-11:
> The second example (`json.c`) does look suspicious. It seems that the JSON schema is missing a way to specify a discovery controller. Maybe @igaw or @hreinecke can confirm.

The comment indicates that the idea to filter out well known discovery controller is correct. But I am a bit lost here too.

hreinecke, 2022-08-25:
> With TP 8013 - Unique Discovery Controller ID implemented, one can no longer identify a discovery controller by simply checking whether its subsystem NQN matches the well-known discovery NQN. But looks like there are still multiple places in the libnvme code where this check is done. For e.g.
> 
> ```
> src/nvme/fabrics.c : build_options():
> 
> if (!strcmp(nvme_ctrl_get_subsysnqn(c), NVME_DISC_SUBSYS_NAME)) {
>      nvme_ctrl_set_discovery_ctrl(c, true);
>      discovery_nqn = true;
> }
> ```
That is by design; a discovery controller can be indicated by either the discovery subsystem name or by the discovery controller attribute. So here we're ensuring that a discovery controller always has the correct type (as the type is not available on older implementations).
> 
> ```
> src/nvme/json.c : json_update_subsys():
> 
> /* Skip discovery subsystems as the nqn is not unique */
> if (!strcmp(subsysnqn, NVME_DISC_SUBSYS_NAME))
>      return;
> ```
'legacy' discovery subsystems all have the same subsystem name, so we cannot list them in the config output as we are relying on a unique subsystem NQN. Unique discovery subystems per definition have a unique subsystem NQN, and as such they can (and should) be listed here, as this allows for additional settings on the controller level like DHCHAP keys.
> 
> ```
> src/nvme/tree.c : nvme_init_subsystem():
> 
> if (!s->subsystype) {
>         if (!strcmp(s->subsysnqn, NVME_DISC_SUBSYS_NAME))
>                 s->subsystype = strdup("discovery");
>         else
>                 s->subsystype = strdup("nvm");
> }
> ```
This is for compability with legacy systems which do not set the subsystem type.

hreinecke, 2022-08-25:
Can we close this?

martin-gpy, 2022-09-13:
> Can we close this?

Yes, thanks for explaining. This may be closed.

closed 2022-09-13 by igaw