linux-nvme/libnvme #456: Improve error reporting in API for MI transport

jk-ozlabs, 2022-08-13:
We currently have a few different sources of potential errors when performing Admin commands over the MI transport:

 - Syscall API failures when interacting with the transport. eg., `sendmsg()` failure when sending a MCTP message
 - Protocol errors detected by the transport implementation: eg., CRC mismatch
 - MI-layer failures: indicated through a MI-specific status byte in response messages eg., invalid opcode.
 - NVMe-layer failures: indicated through the status values from the Completion Queue Entry doublewords

where each of these has a different set of meaningful status values.

It would be nice to properly represent these to the callers; possibly through an extended error struct, or perhaps we can define an encoding that covers these cases (and ones for potential new transports too?)

jk-ozlabs, 2022-08-13:
I can't edit issue metadata, but please feel free to assign to me, tag as 'enhancement', and a future milestone (2.0 perhaps?)

drakedog2008, 2022-10-20:
nvme_status_field  doesn't take full length of a Word.  Can we make a hack like rc(nvme_mi_resp_status)  =  nvme_mi_resp_status + max(nvme_status_field) as a walk-around? I know it is ugly but return code with double meaning really confuses the client, especially for the cases that some nvme_status_field isn't really an error (e.g. NVME_SC_FW_NEEDS_SUBSYS_RESET).

jk-ozlabs, 2022-10-20:
We could certainly look at packing the various status fields in a better way - which would be a proper fix for this bug, rather than a work-around.

The issue at the moment is the aliasing of the status field of the CQE DW 3, and the Status field in the MI response data.

However, we might be in luck here: the parts that we use from the DW3 are only 15 bits (sections 3.3.3.2 and 3.3.3.2.1 of the NVMe base spec defines this), and the MI Status field is only 8. The return values are all `int`, so we have 32 bits to play with here (or do we need to cater for platforms where `int`s are 16 bit?)

One option would be to could pack this as:

 * bit 31(most significant): sign bit, used to distinguish internal/syscall errors
 * bits 30:25: reserved
 * bits 24:16: MI status
 * bit 15: reserved (corresponds to the P bit in CDW3)
 * bits 14:0 (least significant): NVMe status (ie, CDW3 31:17)

This matches the non-MI convention of having the CDW3 returned as the least-significant bits in the return value, we would just augment this with another 8 bits, aligned to start at the 16th bit.

Then, you would have access to both the NVMe and NVMe-MI status values in the one `int`. You'd need to shift-down to get the MI values, but that's probably OK.

Another option is to use a high bit (say, 30) as a flag to indicate a MI vs. NVMe error, and then have the MI values appearing in bits 0:7. This might be better to retain compatibility with the existing interface.

I think I prefer the latter, but I'm open to discussion here.

jk-ozlabs, 2022-10-20:
The change would be something like this:

```diff
From 6d02189240b7dcd6c23d7359f3f528dc17d804da Mon Sep 17 00:00:00 2001
From: Jeremy Kerr <jk@codeconstruct.com.au>
Date: Thu, 20 Oct 2022 15:02:10 +0800
Subject: [PATCH] mi: Distinguish MI status from NVMe (CDW3) status

This change introduces a new encoding for the return values of MI
functions, where we use bit 30 to encode whether the value is either a
MI status value or a NVMe status value.

WIP; proposal only at this stage.

---
 src/nvme/mi.c    |  9 ++++-----
 src/nvme/types.h | 19 +++++++++++++++++++
 2 files changed, 23 insertions(+), 5 deletions(-)

diff --git a/src/nvme/mi.c b/src/nvme/mi.c
index e5ef0a0..fa0e1df 100644
--- a/src/nvme/mi.c
+++ b/src/nvme/mi.c
@@ -315,13 +315,12 @@ static int nvme_mi_admin_parse_status(struct nvme_mi_resp *resp, __u32 *result)
        resp_hdr = (struct nvme_mi_msg_resp *)resp->hdr;
 
        /* If we have a MI error, we can't be sure there's an admin header
-        * following; return just the MI status
-        *
-        * TODO: this may alias the cdw3 result values, see
-        * https://github.com/linux-nvme/libnvme/issues/456
+        * following; return just the MI status, with the status type
+        * indicator of MI.
         */
        if (resp_hdr->status)
-               return resp_hdr->status;
+               return resp_hdr->status |
+                       (NVME_STATUS_TYPE_MI << NVME_STATUS_TYPE_SHIFT);
 
        /* We shouldn't hit this, as we'd have an error reported earlier.
         * However, for pointer safety, ensure we have a full admin header
diff --git a/src/nvme/types.h b/src/nvme/types.h
index 14f50d6..75b2953 100644
--- a/src/nvme/types.h
+++ b/src/nvme/types.h
@@ -6141,6 +6141,25 @@ static inline __u16 nvme_status_code(__u16 status_field)
        return NVME_GET(status_field, SC);
 }
 
+enum nvme_status_type {
+       NVME_STATUS_TYPE_SHIFT          = 30,
+       NVME_STATUS_TYPE_MASK           = 0x1,
+
+       NVME_STATUS_TYPE_NVME           = 0,
+       NVME_STATUS_TYPE_MI             = 1,
+};
+
+static inline bool nvme_status_is_nvme(__u32 status)
+{
+       return NVME_GET(status, STATUS_TYPE) == NVME_STATUS_TYPE_NVME;
+}
+
+static inline bool nvme_status_is_mi(__u32 status)
+{
+       return NVME_GET(status, STATUS_TYPE) == NVME_STATUS_TYPE_MI;
+}
+
+
 /**
  * enum nvme_admin_opcode - Known NVMe admin opcodes
  * @nvme_admin_delete_sq:              Delete I/O Submission Queue
-- 
2.37.2

```

... plus an addition to `nvme-cli` using `nvme_status_is_mi()` when decoding the status values, to print the MI values properly.

drakedog2008, 2022-10-20:
> However, we might be in luck here: the parts that we use from the DW3 are only 15 bits (sections 3.3.3.2 and 3.3.3.2.1 of the NVMe base spec defines this), and the MI Status field is only 8. The [...]

Yes. The DW3 is only 15 bit so why cannot we use the bit 16 as the switch bit for mi status and nvme status, so we don't need to worry about the sizeof(int). 

I don't really think the client need both status at the same time. The cdw3 is only validate when mi status is Success? That is why I suggested using `rc(nvme_mi_resp_status) = nvme_mi_resp_status + max(nvme_status_field)`, which equals to using bit 16 as the switch when `max(nvme_status_field) = 0x8000`. 

Furthermore, the other two types of errors (system, protocol) should be presented with rc < 0 with errno? If not, it can be further encoded to max(nvme_status_field) + max(nvme_mi_resp_status).

drakedog2008, 2022-10-20:
I did forgot the rc is an integer, so that the bit 16 is taken as sign bit on 16 bit integer machines. 

Maybe we can make an integer rc like: 
rc(nvme_status_field) = nvme_status_field
rc(protocol/systemclt) = -257( two's complement system)
rc(nvme_mi_resp_status) = ~nvme_mi_resp_status

jk-ozlabs, 2022-10-21:
Yep, I think we're suggesting the same thing here, just differing on where to put the 'type' bit. 

As you say, we don't have space if int is 16-bit - but I don't think we need to worry about those platforms here (@igaw : any preferences on 16-bit-int platform support?)

If we only care about 32-bit ints, then we can keep it simpler by using the type bit, rather than having an unusual value (rather than -1) for internal errors and the inverted MI status.

Then, I'd rather use bit 30, as that make it easier to extend to further error types in future, if necessary.

drakedog2008, 2022-10-21:
16bit int should have the space for all NVMe/MI status but I agree with anything else you said.

igaw, 2022-10-21:
Not really grasping the complete discussion (just skimmed over). But it looks like you want to encode the return value with additional information. To question of 16 bit integers, these would be `short integers`. `integers` types are 4 bytes. The function uses `int` thus 4 bytes are a safe bet :)

I was wondering if you could use errno in the case status != 0?

jk-ozlabs, 2022-10-21:
> `integers` types are 4 bytes.

not always - the question is whether we want to support that; my inclination would be no.

I think we can get away without a complex return value, and I think we can encode the multiple status values while preserving
backwards API compatibility. We can't really use errno there, as we'll just be overloading another namespace.

igaw, 2022-10-21:
> > `integers` types are 4 bytes.
> 
> not always - the question is whether we want to support platforms where that is not the case; my inclination would be no.

Ah you are referring those arcane systems :). Paul McKenney always reminds me of those.
But these days any sane/relevant platform has 32 bits. So I don't worry to much about `sizeof(int) != 4`.

> I think we can get away without a complex return value, and I think we can encode the multiple status values while preserving backwards API compatibility. We can't really use errno there, as we'll just be overloading another namespace.

Yes, there is that.

FYI, I think we should start thinking on a 2.x schedule. I would like to hold of until February 
with proper 2.x work as there is an incoming feature I need to work on but the spec has first
to be published. I wont have a lot of time until then.

Also I really like to avoid having the same situation with the nvme-cli 1.x/2.0 branch split for a long
time. When we start with libnvme 2.0 we should get stuff sorted out in a timely fashion and not
drag this task around for one year or longer.

jk-ozlabs, 2022-10-21:
OK, sounds good - 32-bit `int` it is then! I'll get a patch proposed for this.

Your 2.x plan sounds good; I'll get the sketches started for an encapsulated device handle API change, so we will have plenty of time to discuss beforehand.

igaw, 2022-10-21:
2.x: FWIW, @safl wants also to work on this but let's get this discussion into the right issue :)

drakedog2008, 2022-10-21:
> OK, sounds good - 32-bit `int` it is then! I'll get a patch proposed for this.

So this is the solution before 2.x, right?

jk-ozlabs, 2022-10-22:
> So this is the solution before 2.x, right?

Yep! this proposal should be API compatible with what we have at the moment.

We may be able to (later) make this a little more obvious with a non-scalar return type, but that would break the API, so that would need to wait until a 2.x release.

igaw, 2022-10-24:
Okay, merged. Let's see what will explode :)

closed 2022-10-24 by igaw