linux-nvme/libnvme #466: C++ Compiler Error on libnvme-mi

drakedog2008, 2022-09-07:
The C++ compiler doesn't fully compatible with C99 standard. So it gave the errors: 

1.  
```
/usr/include/nvme/mi.h:790:56: error: invalid conversion from 'int' to 'nvme_mi_config_smbus_freq' [-fpermissive]
```
2. 
```
/usr/include/nvme/mi.h:2227:9: error: designator order for field 'nvme_ns_mgmt_args::nsid' does not match declaration order in 'nvme_ns_mgmt_args'
 2227 |         };
```

jk-ozlabs, 2022-09-07:
@drakedog2008 thanks for the report, I'll take a look.

jk-ozlabs, 2022-09-09:
Looks like @drakedog has a PR already: https://github.com/linux-nvme/libnvme/pull/468

igaw, 2022-09-22:
The fixes are in the tree. Closing bug report.
8fd81cdb88bb ("mi: Fix C++ compiler errors")

closed 2022-09-22 by igaw