linux-nvme/libnvme #467: Compile warning in mi-mctp.c

igaw, 2022-09-08:
```shell
$ gcc --version
gcc (SUSE Linux) 12.2.1 20220830 [revision e927d1cf141f221c5a32574bde0913307e140984]
```

```shell
[35/100] Compiling C object subprojects/libnvme/examples/mi-mctp.p/mi-mctp.c.o
../subprojects/libnvme/examples/mi-mctp.c: In function ‘do_action_endpoint’:
../subprojects/libnvme/examples/mi-mctp.c:665:16: warning: ‘rc’ may be used uninitialized [-Wmaybe-uninitialized]
  665 |         return rc;
      |                ^~
../subprojects/libnvme/examples/mi-mctp.c:637:13: note: ‘rc’ was declared here
  637 |         int rc;
      |             ^~
```

jk-ozlabs, 2022-09-09:
Will take a look and get a fix done - feel free to assign to me.

jk-ozlabs, 2022-09-09:
Looks like this requires specific CFLAGS to reproduce:

```sh
CFLAGS=-Wmaybe-uninitialized meson obj --optimization=1
```

However, unlike https://github.com/linux-nvme/nvme-cli/issues/1647 , I can reproduce this on Debian.

closed 2022-09-09 by igaw