linux-nvme/libnvme #471: Splitting the address in the JSON output

igaw, 2022-09-14:
Should we split the 'address' for JSON?

Currently we have:

```
  +- nvme9 fc traddr=nn-0x201700a09890f5bf:pn-0x201900a09890f5bf,host_traddr=nn-0x200000109b579ef5:pn-0x100000109b579ef5 live 
 
          {
             "Name":"nvme9",
             "Transport":"fc",
             "Address":"traddr=nn-0x201700a09890f5bf:pn-0x201900a09890f5bf,host_traddr=nn-0x200000109b579ef5:pn-0x100000109b579ef5",
             "State":"live"
           }
```

I don't think we should change the shell output (again)? The nvme-cli 1.x version uses spaces as separator and not comma as by nvme-cli 2.x

But it would make sense to split the values into fields in the JSON output though? Something like

```
Address:  { 
  traddr = "...",
  host_traddr = "... ",
  trsvcid = "..."
}
```

But I assume this is something for 2.x of libnvme. But we might want to keep the current version and just add the spitted out values.

Thoughts?

mwilck, 2022-09-14:
I am for it, but obviously this is sort of an external API which is consumed by other tools.

igaw, 2025-09-25:
Maybe for a later version