linux-nvme/libnvme #475: uint128 type is not always available

igaw, 2022-09-22:
It turns out the uint128 type is not available on all platforms/compiler combinations:

https://build.opensuse.org/package/live_build_log/home:wagi:branches:bsc1201701/nvme-cli/openSUSE_Tumbleweed/i586

igaw, 2022-09-22:
@jk-ozlabs it's not just ARM which hits this bug.

jk-ozlabs, 2022-09-22:
ah! Good to know, thanks.

igaw, 2022-09-22:
Will look into it after processing the github backlog.

jk-ozlabs, 2022-09-23:
Should we just revert those patches introducing `uint128_t` usage? or do we need something more involved instead?

igaw, 2022-09-23:
Not sure, the thing is a `long double` is 12 bytes on 32bit arch and not 16 bytes long either. So I am pondering if we should introduce our own 16 byte type which maps on uint128_t where it is available and on 32bit something home grown. We don't need any math features such as addition things from this type. It's just reading the value and printing it.

jk-ozlabs, 2022-09-23:
yeah, I think that `long double` wouldn't suit either, being a floating-point type. I like the idea of a compat type that we have our own print function for.

igaw, 2022-09-23:
So here is a first dodgy idea how to get it working again but not really fixing it for 32bit.

```c
/* -*- compile-command: "gcc -Wall -std=gnu99 -O0 -g -m32 types.c -o types-32; gcc -Wall -std=gnu99 -O0 -g types.c -o types-64" -*- */

#include <stdint.h>
#include <stdio.h>
#include <string.h>
#include <inttypes.h>

#ifndef __SIZEOF_INT128__
typedef long double __uint128_t;
#endif

typedef unsigned char __u8;

static char __buf[40];

#ifdef __SIZEOF_INT128__

__uint128_t le128_to_cpu(__u8 *data)
{
	__uint128_t result = 0;

	for (int i = 0; i < 16; i++) {
		result *= 256;
		result += data[15 - i];
	}
	return result;
}

char *uint128_t_to_string(__uint128_t val)
{
	uint64_t high = (val >> 64);
	uint64_t low = (uint64_t)val;

	if (high)
		snprintf(__buf, sizeof(__buf), "%" PRIu64 "%" PRIu64, high, low);
	else
		snprintf(__buf, sizeof(__buf), "%" PRIu64, low);

	return __buf;
}

#else

long double le128_to_cpu(__u8 *data)
{
        long double result = 0;

	for (int i = 0; i < 16; i++) {
		result *= 256;
		result += data[15 - i];
	}
	return result;
}

char *uint128_t_to_string(long double val)
{
	snprintf(__buf, sizeof(__buf), "%Lf", val);

	for (int i = strlen(__buf) - 1; i > 0; i--) {
		if (__buf[i] == '.') {
			__buf[i] = '\0';
			break;
		} else if (__buf[i] != '0') {
			__buf[i+1] = '\0';
			break;
		}
	}
	return __buf;
}

#endif

int main(int argc, char *argv[])
{
	uint64_t a = 0;
	long double b = 0;
	__uint128_t c = 0;

	printf("sizeof(uint64_t)    = %zu\n", sizeof(a));
	printf("sizeof(long double) = %zu\n", sizeof(b));
	printf("sizeof(__uint128_t) = %zu\n", sizeof(c));

	__u8 d[8] = { 0, 1, 0, 0, 0, 0, 0, 0 };
	printf("d = %s\n", uint128_t_to_string(le128_to_cpu(d)));

	return a + b + c;
}

```
```

32bit:
sizeof(uint64_t)    = 8
sizeof(long double) = 12
sizeof(__uint128_t) = 12
d = 256

64bit:
sizeof(uint64_t)    = 8
sizeof(long double) = 16
sizeof(__uint128_t) = 16
d = 256

```

igaw, 2022-09-23:
I would like to keep the 64bit code in place, that means things like

```c
__uint128_t v = le128_to_cpu(some_reg);
printf("%s\n", uint128_t_to_string(v));
```

could be left in place.

igaw, 2022-09-23:
So here an idea which could work. In case we don't have the `__uint128_t` type, we just define a pointer type and do the conversion magic with buffers. This might work for our usage of the `uint128_t` type. As we just read and print it. There is no other access to it, so we might just get away. It's not a beauty but I really like to avoid to introduce something like `gmp` as it just makes everything complex.

Also I think I would only print those values as hex values. This avoids the necessity handle the overflow calculation for the 10 based system.

```c
typedef __u8* __uint128_t;                                                                         

union __nvme_uint128 {
        unsigned char v[16];
	uint64_t q[2];
};

typedef union __nvme_uint128 __nvme_uint128_t;

__u8 *le128_to_cpu(__u8 *data)
{
	__nvme_uint128_t u;

	memcpy(u.v, data, 16);

	u.q[0] = le64_to_cpu(u.q[0]);
	u.q[1] = le64_to_cpu(u.q[1]);

	memcpy(data, u.v, 16);

	return data;
}

char *uint128_t_to_string(__u8 *val)
{
	__nvme_uint128_t u;

	memcpy(u.v, val, 16);

	if (u.q[1])
		snprintf(__buf, sizeof(__buf), "%" PRIx64 "%" PRIx64, u.q[1], u.q[0]);
	else
		snprintf(__buf, sizeof(__buf), "%" PRIx64, u.q[0]);

	return __buf;
}
```

```
32bit:
sizeof(uint64_t)    = 8
sizeof(long double) = 12
sizeof(__uint128_t) = 4
d = 100

64bit:
sizeof(uint64_t)    = 8
sizeof(long double) = 16
sizeof(__uint128_t) = 16
d = 100
```

igaw, 2022-09-23:
D'oh the report should have been against nvme-cli not libnvme.

igaw, 2022-09-23:
@jk-ozlabs can you give this a test and see if it actually works?

jk-ozlabs, 2022-09-23:
Yep, I like that second option, as it avoids the floating point confusion there. However, I'm not sure about:

```c
typedef __u8* __uint128_t;
```

\- as it should (ideally) be a fixed-size scalar type, and not pointer-size dependent.

We could just keep everything as `__nvme_uint128_t`?

Regardless, will give it a go on 32-bit platforms.

igaw, 2022-09-23:
As it turns out it was dead simple:

```c
union nvme_uint128 {
        unsigned char v[16];
	uint64_t q[2];
};

typedef union nvme_uint128 nvme_uint128_t;

nvme_uint128_t le128_to_cpu(__u8 *data)
{
	nvme_uint128_t u;

	memcpy(u.v, data, 16);

#if HAVE_BIG_ENDIAN
	u.q[0] = le64_to_cpu(u.q[0]);
	u.q[1] = le64_to_cpu(u.q[1]);
#endif
	return u;
}

char *uint128_to_string(nvme_uint128_t val)
{
	static char buf[40];

	if (val.q[1])
		snprintf(buf, sizeof(buf), "%" PRIx64 "%" PRIx64, val.q[1], val.q[0]);
	else
		snprintf(buf, sizeof(buf), "%" PRIx64, val.q[0]);

	return buf;
}
```

Replaced all `__uint128_t` users with `nvme_uint128_t` all seems to work so far... (famous last words)

jk-ozlabs, 2022-09-24:
nice!

jk-ozlabs, 2022-09-25:
OK, this mostly works, except it looks like we want `_to_string` to print in decimal, given the current implementation.

I have a patch in progress - everything else seems to work fine, but we'll need some kind of decimal conversion to keep consistent with the current version.

jk-ozlabs, 2022-09-25:
PR sent: https://github.com/linux-nvme/nvme-cli/pull/1685

igaw, 2022-09-26:
> OK, this mostly works, except it looks like we want `_to_string` to print in decimal, given the current implementation.
> 
> I have a patch in progress - everything else seems to work fine, but we'll need some kind of decimal conversion to keep consistent with the current version.

Indeed, I was a bit sloppy with my `printf` attempt. Let's keep it backwards compatible and use your version.

igaw, 2022-09-26:
Merged everything. Let's see if there are any fallouts from this.

closed 2022-09-26 by igaw