linux-nvme/libnvme #485: nvme_scan_topology() triggers I/O errors logged in kmsg

tbzatek, 2022-09-26:
The `NVME_IDENTIFY_CNS_NS_DESC_LIST` Namespace Identification Descriptor list (CNS Value 0x03) is a NVMe 1.3 feature yet it's called for all devices as part of the topology scanning, trigerred e.g. by `nvme_scan_topology()` from the `nvme list` command.

https://github.com/linux-nvme/libnvme/blob/212a72bffe2edbf2047defa85682e37ad5c9b262/src/nvme/tree.c#L1785-L1786

This is expected to fail on devices not supporting this and the code copes well with that. The problem are these annoying high-severity messages logged from the kernel side:

```
nvme0: Admin Cmd(0x6), I/O Error (sct 0x0 / sc 0x2) DNR 
nvme1: Admin Cmd(0x6), I/O Error (sct 0x0 / sc 0x2) DNR 
```

https://git.kernel.org/pub/scm/linux/kernel/git/stable/linux.git/tree/drivers/nvme/host/core.c?h=v5.19.7#n298

I haven't found any obvious way to silence them. Ideally status codes from admin commands related to an unsupported functionality should not be logged at all.

For the record, this is my testing system:
```
# nvme list -v
Subsystem        Subsystem-NQN                                                                                    Controllers
---------------- ------------------------------------------------------------------------------------------------ ----------------
nvme-subsys1     nqn.2014.08.org.nvmexpress:80868086PHLE802300CP6P4OGN  INTEL SSDPE2KE064T7                       nvme1
nvme-subsys0     nqn.2014.08.org.nvmexpress:80868086PHLE802300956P4OGN  INTEL SSDPE2KE064T7                       nvme0

Device   SN                   MN                                       FR       TxPort Address        Subsystem    Namespaces      
-------- -------------------- ---------------------------------------- -------- ------ -------------- ------------ ----------------
nvme1    PHLE802300CP6P4OGN   INTEL SSDPE2KE064T7                      QDV101D1 pcie   0005:01:00.0   nvme-subsys1 nvme1n1
nvme0    PHLE802300956P4OGN   INTEL SSDPE2KE064T7                      QDV101D1 pcie   0004:01:00.0   nvme-subsys0 nvme0n1

Device       Generic      NSID     Usage                      Format           Controllers     
------------ ------------ -------- -------------------------- ---------------- ----------------
/dev/nvme1n1 /dev/ng1n1   1          6.40  TB /   6.40  TB    512   B +  0 B   nvme1
/dev/nvme0n1 /dev/ng0n1   1          6.40  TB /   6.40  TB    512   B +  0 B   nvme0
```

igaw, 2022-09-26:
IIRC, @keithbusch suggest to silence this message in the kernel.

igaw, 2022-09-26:
https://github.com/linux-nvme/nvme-cli/issues/1593

tbzatek, 2022-09-26:
Ah, good, thanks for the link!