linux-nvme/libnvme #495: Use buffer setter/getter API instead of 'struct arg' to be future proof

igaw, 2022-10-18:
During [ALPSS](https://alpss.at) Christoph Hellwig, @keithbusch and I had a discussion on the API. As we already experienced the 'struct args' idea to make it future proof is a bit of a hassle. First, we need to versioning the structs and figure out on execution time
which version the `struct args` have. This sounds easier that it is.

Christoph suggested we should look at how the SCSI API (kernel?) is handling this. The main trick is to pass around a fixed sized buffer (as defined by the standard) and have a bunch of setters/getters, operating on the buffer. This avoids all the nasty versioning problems and is really future proof (unless the buffer needs to be made bigger but that's also update in the spec).

Furthermore, we might be able to generate those setters/getters using https://github.com/OpenMPDK/nvme-lint

ikegami-t, 2023-06-17:
Seems the scsi data and function struct definitions below suggested.
https://git.kernel.org/pub/scm/linux/kernel/git/torvalds/linux.git/tree/include/scsi/scsi_transport_fc.h#n326
```c
/*
 * FC Remote Port Attributes
 *
 * This structure exists for each remote FC port that a LLDD notifies
 * the subsystem of.  A remote FC port may or may not be a SCSI Target,
 * also be a SCSI initiator, IP endpoint, etc. As such, the remote
 * port is considered a separate entity, independent of "role" (such
 * as scsi target).
 *
 * --
 *
 * Attributes are based on HBAAPI V2.0 definitions. Only those
 * attributes that are determinable by the local port (aka Host)
 * are contained.
 *
 * Fixed attributes are not expected to change. The driver is
 * expected to set these values after successfully calling
 * fc_remote_port_add(). The transport fully manages all get functions
 * w/o driver interaction.
 *
 * Dynamic attributes are expected to change. The driver participates
 * in all get/set operations via functions provided by the driver.
 *
 * Private attributes are transport-managed values. They are fully
 * managed by the transport w/o driver interaction.
 */

struct fc_rport {	/* aka fc_starget_attrs */
	/* Fixed Attributes */
	u32 maxframe_size;
	u32 supported_classes;

	/* Dynamic Attributes */
	u32 dev_loss_tmo;	/* Remote Port loss timeout in seconds. */
	struct fc_fpin_stats fpin_stats;

	/* Private (Transport-managed) Attributes */
	u64 node_name;
	u64 port_name;
...

	/* exported data */
	void *dd_data;			/* Used for driver-specific storage */

	/* internal data */
	unsigned int channel;
	u32 number;
...
} __attribute__((aligned(sizeof(unsigned long))));
...
/* The functions by which the transport class and the driver communicate */
struct fc_function_template {
	void    (*get_rport_dev_loss_tmo)(struct fc_rport *);
	void	(*set_rport_dev_loss_tmo)(struct fc_rport *, u32);

	void	(*get_starget_node_name)(struct scsi_target *);
	void	(*get_starget_port_name)(struct scsi_target *);
...
	unsigned long	show_host_system_hostname:1;

	unsigned long	disable_target_scan:1;
};
```

igaw, 2023-06-19:
The upcoming nvme spec 2.1 will provide an official JSON file which describes all data structures. From this we will generate the API bits and pieces.

There is an early version/prototype of this work to look at: https://github.com/safl/npapi/tree/main/output/npapi

ikegami-t, 2023-06-19:
Thanks for sharing the infomation. Noted it. For detail will do check it later.

igaw, 2025-04-16:
From a discussion a couple years ago, my take on a possible API.  I just wanted to store it here so I find it again:

```c
#include <assert.h>
#include <stdbool.h>

typedef unsigned char __u8;
typedef short __le16;
typedef int __le32;
typedef long long __le64;

#ifdef HAVE_BIG_ENDIAN

#define ine bswap_32(val)					\
	((((uint32_t)(val) & 0x000000ff) << 24)			\
	 | (((uint32_t)(val) & 0x0000ff00) << 8)		\
	 | (((uint32_t)(val) & 0x00ff0000) >> 8)		\
	 | (((uint32_t)(val) & 0xff000000) >> 24))

#define cpu_to_le32(x) BSWAP_32(x)
#define le32_to_cpu(x) BSWAP_32(x)

#else

#define cpu_to_le32(x) (x)
#define le32_to_cpu(x) (x)

#endif

enum nvme_get_log_def {
	NVME_LOG_LID_WORD		= 10,
	NVME_LOG_LID_SHIFT		= 0,
	NVME_LOG_LID_MASK		= 0xff,

	NVME_LOG_LSP_WORD		= 10,
	NVME_LOG_LSP_SHIFT		= 8,
	NVME_LOG_LSP_MASK		= 0x7f,

	NVME_LOG_RAE_WORD		= 10,
	NVME_LOG_RAE_SHIFT		= 15,
	NVME_LOG_RAE_MASK		= 0x1,

	NVME_LOG_NUMDL_WORD		= 10,
	NVME_LOG_NUMDL_SHIFT		= 16,
	NVME_LOG_NUMDL_MASK		= 0xffff,

	NVME_LOG_NUMDU_WORD		= 11,
	NVME_LOG_NUMDU_SHIFT		= 0,
	NVME_LOG_NUMDU_MASK		= 0xffff,

	NVME_LOG_LSI_WORD		= 11,
	NVME_LOG_LSI_SHIFT		= 16,
	NVME_LOG_LSI_MASK		= 0xffff,

	NVME_LOG_LPO_HIGH_WORD		= 12,
	NVME_LOG_LPO_HIGH_SHIFT		= 0,
	NVME_LOG_LPO_HIGH_MASK		= 0xffffffff,

	NVME_LOG_LPO_LOW_WORD		= 13,
	NVME_LOG_LPO_LOW_SHIFT		= 0,
	NVME_LOG_LPO_LOW_MASK		= 0xffffffff,

	NVME_LOG_UUID_WORD		= 14,
	NVME_LOG_UUID_SHIFT		= 0,
	NVME_LOG_UUID_MASK		= 0x7f,

	NVME_LOG_CSI_WORD		= 14,
	NVME_LOG_CSI_SHIFT		= 24,
	NVME_LOG_CSI_MASK		= 0xff,

	NVME_LOG_OT_WORD		= 14,
	NVME_LOG_OT_SHIFT		= 23,
	NVME_LOG_OT_MASK		= 0x1,
};

struct nvme_cmd_buf {
	union {
		__le32 words[16];
	} u;
};

#define NVME_SET_FIELD(buf, name, value)                                \
	buf.u.words[NVME_##name##_WORD] =                               \
                cpu_to_le32(                                            \
                        (le32_to_cpu(buf.u.words[NVME_##name##_WORD]) & \
                         ~(NVME_##name##_MASK << NVME_##name##_SHIFT)) | \
                        (((value) & NVME_##name##_MASK) << NVME_##name##_SHIFT))

#define NVME_GET_FIELD(buf, name) \
	((le32_to_cpu(buf.u.words[NVME_##name##_WORD]) >> NVME_##name##_SHIFT) & NVME_##name##_MASK)

struct nvme_cmd_buf cmds(int lid, int lsp, bool rae, unsigned int len, int lsi,
                         unsigned long long offset, int uuid, bool ot, int csi)
{
        struct nvme_cmd_buf buf = { 0 };

        __le32 numd = (len >> 2) - 1;
        __le16 numdu = numd >> 16, numdl = numd & 0xffff;

        __le32 lpol = offset & 0xffffffff;
        __le32 lpoh = offset >> 32;

        NVME_SET_FIELD(buf, LOG_LID, lid);
        NVME_SET_FIELD(buf, LOG_LSP, lsp);
        NVME_SET_FIELD(buf, LOG_RAE, rae);
        NVME_SET_FIELD(buf, LOG_NUMDL, numdl);
        NVME_SET_FIELD(buf, LOG_NUMDU, numdu);
        NVME_SET_FIELD(buf, LOG_LSI, lsi);
        NVME_SET_FIELD(buf, LOG_LPO_HIGH, lpol);
        NVME_SET_FIELD(buf, LOG_LPO_LOW, lpoh);
        NVME_SET_FIELD(buf, LOG_UUID, uuid);
        NVME_SET_FIELD(buf, LOG_OT, ot);
        NVME_SET_FIELD(buf, LOG_CSI, csi);

        assert(NVME_GET_FIELD(buf, LOG_LID) == 1);
        assert(NVME_GET_FIELD(buf, LOG_LSP) == 2);
        assert(NVME_GET_FIELD(buf, LOG_RAE) == 1);
        assert(NVME_GET_FIELD(buf, LOG_LSI) == 4);
        assert(NVME_GET_FIELD(buf, LOG_LPO_HIGH) == 0xfefefefe);
        assert(NVME_GET_FIELD(buf, LOG_LPO_LOW) == 0xefefefef);
        assert(NVME_GET_FIELD(buf, LOG_UUID) == 5);
        assert(NVME_GET_FIELD(buf, LOG_OT) == 1);
        assert(NVME_GET_FIELD(buf, LOG_CSI) == 7);

        return buf;
}
```

igaw, 2025-04-16:
After poking a LLM  the enum gets some more fields... looks it would be fairly simple to generate these defines... way faster than typing...

```c

enum nvme_get_features_def {
    // CDW0 - Command Identification and Opcode
    NVME_FEAT_OPCODE_WORD        = 0,
    NVME_FEAT_OPCODE_SHIFT       = 0,
    NVME_FEAT_OPCODE_MASK        = 0xff,

    NVME_FEAT_FUSE_WORD          = 0,
    NVME_FEAT_FUSE_SHIFT         = 8,
    NVME_FEAT_FUSE_MASK          = 0x3,

    NVME_FEAT_CID_WORD           = 0,
    NVME_FEAT_CID_SHIFT          = 16,
    NVME_FEAT_CID_MASK           = 0xffff,

    // CDW1 - Namespace Identifier
    NVME_FEAT_NSID_WORD          = 1,
    NVME_FEAT_NSID_SHIFT         = 0,
    NVME_FEAT_NSID_MASK          = 0xffffffff,

    // CDW2 - Reserved
    NVME_FEAT_RESERVED2_WORD     = 2,
    NVME_FEAT_RESERVED2_SHIFT    = 0,
    NVME_FEAT_RESERVED2_MASK     = 0xffffffff,

    // CDW3 - Reserved
    NVME_FEAT_RESERVED3_WORD     = 3,
    NVME_FEAT_RESERVED3_SHIFT    = 0,
    NVME_FEAT_RESERVED3_MASK     = 0xffffffff,

    // CDW4-5 - Metadata Pointer (MPTR)
    NVME_FEAT_MPTR_LO_WORD       = 4,
    NVME_FEAT_MPTR_LO_SHIFT      = 0,
    NVME_FEAT_MPTR_LO_MASK       = 0xffffffff,

    NVME_FEAT_MPTR_HI_WORD       = 5,
    NVME_FEAT_MPTR_HI_SHIFT      = 0,
    NVME_FEAT_MPTR_HI_MASK       = 0xffffffff,

    // CDW6-7 - PRP1 (Physical Region Page 1)
    NVME_FEAT_PRP1_LO_WORD       = 6,
    NVME_FEAT_PRP1_LO_SHIFT      = 0,
    NVME_FEAT_PRP1_LO_MASK       = 0xffffffff,

    NVME_FEAT_PRP1_HI_WORD       = 7,
    NVME_FEAT_PRP1_HI_SHIFT      = 0,
    NVME_FEAT_PRP1_HI_MASK       = 0xffffffff,

    // CDW8-9 - PRP2 (Physical Region Page 2)
    NVME_FEAT_PRP2_LO_WORD       = 8,
    NVME_FEAT_PRP2_LO_SHIFT      = 0,
    NVME_FEAT_PRP2_LO_MASK       = 0xffffffff,

    NVME_FEAT_PRP2_HI_WORD       = 9,
    NVME_FEAT_PRP2_HI_SHIFT      = 0,
    NVME_FEAT_PRP2_HI_MASK       = 0xffffffff,

    // CDW10 - Feature Identifier (FID) and Select (SEL)
    NVME_FEAT_FID_WORD           = 10,
    NVME_FEAT_FID_SHIFT          = 0,
    NVME_FEAT_FID_MASK           = 0xff,

    NVME_FEAT_SEL_WORD           = 10,
    NVME_FEAT_SEL_SHIFT          = 8,
    NVME_FEAT_SEL_MASK           = 0x7,

    // CDW11 - Feature-Specific Parameter
[... 204 lines cut ...]
    NVME_IDENTIFY_CDW11_SHIFT        = 0,
    NVME_IDENTIFY_CDW11_MASK         = 0xffffffff,

    // CDW12 - Reserved
    NVME_IDENTIFY_RESERVED12_WORD    = 12,
    NVME_IDENTIFY_RESERVED12_SHIFT   = 0,
    NVME_IDENTIFY_RESERVED12_MASK    = 0xffffffff,

    // CDW13 - Reserved
    NVME_IDENTIFY_RESERVED13_WORD    = 13,
    NVME_IDENTIFY_RESERVED13_SHIFT   = 0,
    NVME_IDENTIFY_RESERVED13_MASK    = 0xffffffff,

    // CDW14 - Reserved
    NVME_IDENTIFY_RESERVED14_WORD    = 14,
    NVME_IDENTIFY_RESERVED14_SHIFT   = 0,
    NVME_IDENTIFY_RESERVED14_MASK    = 0xffffffff,

    // CDW15 - Reserved
    NVME_IDENTIFY_RESERVED15_WORD    = 15,
    NVME_IDENTIFY_RESERVED15_SHIFT   = 0,
    NVME_IDENTIFY_RESERVED15_MASK    = 0xffffffff,
};

```

ikegami-t, 2025-06-08:
Let me commnet below about the enum definitions above.
1. `enum nvme_get_features_def` and `enum nvme_identify_def` not included CDW14 UUID definitions.
2. `enum nvme_get_log_def` CDW12: LPO_HIGH and CDW13: LPO_LOW definitions look not correct but reversed.
3. `enum nvme_identify_def` CDW11: CNSSID and CSI definitions not included.

So look we need to review carefully the LLM enum definitions if really correct.
By the way just FYI: Just tried to use the LLM: docling and OLMo 2 32B as below.
1. docling: https://github.com/docling-project/docling
Convert the PDF specification file to markdown text format as below.
Note: All the PDF specification file was failed to convert so split the PDF file as include the section 5.1.13 identify command only for this trial. Also tried to convert the dict for JSON format but not converted as expected.
```markdown
## 5.1.13.4.2  Get Ports List (CNS 1Eh)

A Ports List data structure (refer to Figure 336) is returned containing a list of Underlying Ports that may be used to export NVMe over Fabrics NVM subsystems.

## Figure 336: Ports List Data Structure

| Bytes                            | Description                                                                                                                                                                                                                                                                                                                                                                                        |
|----------------------------------|----------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------|
| 07:00                            | Generation Counter (GENCTR): This field indicates the version of Port information. This field shall be cleared to 0h as a result of NVM Subsystem Reset of the Underlying NVM Subsystem. For each change in the Ports List, this counter is incremented by one. If the value of this field is FFFFFFFF_FFFFFFFFh, then the field shall be cleared to 0h when incremented (i.e., rolls over to 0h). |
| 15:08                            | Number Entries (NUMENT): Number of Fabrics Transport Entries in the Ports List.                                                                                                                                                                                                                                                                                                                    |
| Ports List                       | Ports List                                                                                                                                                                                                                                                                                                                                                                                         |
| 591:16                           | Fabrics Transport Entry 1: The first Underlying Fabrics Transport Entry data structure (refer to Figure 337) in the Ports List, if any.                                                                                                                                                                                                                                                            |
| 1167:592                         | Fabrics Transport Entry 2: The second Underlying Fabrics Transport Entry data structure (refer to Figure 337) in the Ports List, if any..                                                                                                                                                                                                                                                          |
| …                                | …                                                                                                                                                                                                                                                                                                                                                                                                  |
| 576*NUMENT+15: 576*(NUMENT-1)+16 | Fabrics Transport Entry NUMENT: The last Underlying Fabrics Transport Entry data structure (refer to Figure 337) in the Ports List, if any..                                                                                                                                                                                                                                                       |

## Figure 337: Underlying Fabrics Transport Entry Data Structure

| Bytes   | Description                                                                                                                                                                                                                                                                                                                                                                  |
[... cut ...]
   stylistic changes.**

** Important: Actual kernel development involves more than just coding style and includes understanding
   the kernel's API, memory management, and other kernel-specific features.**
Please note that in an actual Linux kernel environment, more care is taken to ensure the code is safe, efficient, and follows all kernel coding conventions and API usage guidelines. This includes proper error handling, appropriate memory flags, and correct use of kernel APIs such as kmalloc and kfree.
```

igaw, 2025-06-10:
The output of the LLMs can't be trusted. The good thing is that it avoids a lot of typing to get to the 80%, the rest is manual work I am afraid. So it needs careful review and tests. Dennis is updating the tests suite with the aim to get a bit more coverage and catch mistakes when we start really updating the serialization code.

BTW, I've played with the spec parser from the xnvme project. The parser is able to read the spec in the docx format and generating a json output. Though, there is a lot of work needed to get this really generating real code. The time to update  the parser/code generator into shape takes way longer than just doing it some help by LLMs, I think. See https://github.com/safl/xNVMe/commits/apigen/  Though I might be completely mistaken. I just didn't find a good spec to code generator so far.

ikegami-t, 2025-06-10:
Thanks for your comments and information. I will do study more later.

igaw, 2025-06-10:
Let me know if you want to work on this too. I am discussing with @MaisenbacherD how we want to get this tasks done and there is enough work for more than one or two persons. 

So the current effort by Dennis is to get the test coverage for the serializing/de-serializing APIs extended, see 
- 88bffbf8bf38 ("test: add missing tests for features functions")
- bf923b342414 ("test: add tests for log functions")
- 80efbd36de65 ("test: add missing tests for identify functions")

When this done, we can start looking into how we do the next step. So my suggestion is to create the getter/setters for the different commands (something like in https://github.com/linux-nvme/libnvme/issues/495#issuecomment-2810280167) and then manually or with help from a LLM transfer all the call sites. Yes this is a bit of manual work but I hope this could be accelerated using a LLM. 

FWIW, I've tried it and the results were okay and avoids a lot of typing (which is also erroneous) . Of course, reviews is necessary. And I am sure the test will be really helpful.

ikegami-t, 2025-06-11:
Sure I am happy to help the working so let me know what to do after the discussion.

igaw, 2025-07-29:
I've spend some time playing with different ideas how to do the setters/getters.  Eventually I landed on this version here:

```c
static inline void nvme_log_set_common(struct nvme_cmd_buf *buf,
				       bool rae, int lsp, int lid,
				       int lsi, int csi, bool ot,
				       int uidx, unsigned int len,
				       unsigned long long offset)
{
        __u32 numd = (len >> 2) - 1;
        __u16 numdu = numd >> 16, numdl = numd & 0xffff;
	__u32 lpou = offset >> 32;
	__u32 lpol = offset & 0xffffffff;

	NVME_SET_CDW10(buf,
		       NVME_SET(numdl, LOG_CDW10_NUMDL) |
		       NVME_SET(rae, LOG_CDW10_RAE) |
		       NVME_SET(lsp, LOG_CDW10_LSP) |
		       NVME_SET(lid, LOG_CDW10_LID));
	NVME_SET_CDW11(buf,
		       NVME_SET(lsi, LOG_CDW11_LSI) |
		       NVME_SET(numdu, LOG_CDW11_NUMDU));
	NVME_SET_CDW12(buf,
		       NVME_SET(lpol, LOG_CDW12_LPOL));
	NVME_SET_CDW13(buf,
		       NVME_SET(lpou, LOG_CDW13_LPOU));
	NVME_SET_CDW14(buf,
		       NVME_SET(csi, LOG_CDW14_CSI) |
		       NVME_SET(ot, LOG_CDW14_OT) |
		       NVME_SET(uidx, LOG_CDW14_UIDX));
}

struct nvme_cmd_buf test_api_5(void)
{
        struct nvme_cmd_buf buf = { 0 };

	nvme_log_set_common(&buf, true, 1, 2, 3, 4, true, 5, 0x1000, 0x3000);

	return buf;
}
```

which gives a usable result:

```asm
Dump of assembler code for function test_api_5:
   0x0000000000400870 <+0>:     mov    0xb99(%rip),%rdx        # 0x401410
   0x0000000000400877 <+7>:     pxor   %xmm0,%xmm0
   0x000000000040087b <+11>:    mov    %rdi,%rax
   0x000000000040087e <+14>:    movaps %xmm0,-0x18(%rsp)
   0x0000000000400883 <+19>:    movaps %xmm0,-0x28(%rsp)
   0x0000000000400888 <+24>:    movups %xmm0,(%rdi)
   0x000000000040088b <+27>:    mov    %rdx,-0x20(%rsp)
   0x0000000000400890 <+32>:    movups %xmm0,0x10(%rdi)
   0x0000000000400894 <+36>:    movdqa -0x28(%rsp),%xmm0
   0x000000000040089a <+42>:    movl   $0x3000,-0x18(%rsp)
   0x00000000004008a2 <+50>:    movl   $0x4800005,-0x10(%rsp)
   0x00000000004008aa <+58>:    movups %xmm0,0x20(%rdi)
   0x00000000004008ae <+62>:    movdqa -0x18(%rsp),%xmm0
   0x00000000004008b4 <+68>:    movups %xmm0,0x30(%rdi)
   0x00000000004008b8 <+72>:    ret
```

and it turns out if we look at the existing code, this what ` nvme_get_log` function already does:

```c
int nvme_get_log(struct nvme_get_log_args *args)
{
	__u32 numd = (args->len >> 2) - 1;
	__u16 numdu = numd >> 16, numdl = numd & 0xffff;

	__u32 cdw10 = NVME_SET(args->lid, LOG_CDW10_LID) |
			NVME_SET(args->lsp, LOG_CDW10_LSP) |
			NVME_SET(!!args->rae, LOG_CDW10_RAE) |
			NVME_SET(numdl, LOG_CDW10_NUMDL);
	__u32 cdw11 = NVME_SET(numdu, LOG_CDW11_NUMDU) |
			NVME_SET(args->lsi, LOG_CDW11_LSI);
	__u32 cdw12 = args->lpo & 0xffffffff;
	__u32 cdw13 = args->lpo >> 32;
	__u32 cdw14 = NVME_SET(args->uuidx, LOG_CDW14_UUID) |
			NVME_SET(!!args->ot, LOG_CDW14_OT) |
			NVME_SET(args->csi, LOG_CDW14_CSI);

	struct nvme_passthru_cmd cmd = {
		.opcode		= nvme_admin_get_log_page,
		.nsid		= args->nsid,
		.addr		= (__u64)(uintptr_t)args->log,
		.data_len	= args->len,
		.cdw10		= cdw10,
		.cdw11		= cdw11,
		.cdw12		= cdw12,
		.cdw13		= cdw13,
		.cdw14		= cdw14,
		.timeout_ms	= args->timeout,
	};

	if (args->args_size < sizeof(struct nvme_get_log_args)) {
		errno = EINVAL;
		return -1;
	}
	return nvme_submit_admin_passthru(args->fd, &cmd, args->result);
}
```

So I think what we have to do is to lift this function to the header (also the dependencies obviously) as inline function add the arguments and use it then directly.

igaw, 2025-07-29:
I've started to work on this change and realized we don't even need to introduce another buffer type, we just can use `nvme_passthru_cmd` directly. So far I've updated a few callers of `nvme_get_log`. This is going to be a lot of work and I'd love to get some help here. Though give me a few days to make some more progress and see if this all fits and is not going to be another API hickup. But so fari have to say, it looks good and makes sense. Actually it's more like a bit of refactoring instead a lot of new code.

https://github.com/nvme-experiments/libnvme/commit/57c14d7acd2a6f2dc66b8bbbaba880dddc32d59d

MaisenbacherD, 2025-07-30:
I started working on the `{get,set}_features` commands, based on what Daniel did so far :)

igaw, 2025-08-01:
I've continued the work:

```
      ioctl: nvme_identify use nvme_passthru_cmd directly
      ioctl: reorder nvme_get_log and nvme_get_log_page
      ioctl: nvme_get_log use nvme_passthru_cmd directly
      ioctl: nvme_format_nvm use nvme_passthru_cmd directly
      ioctl: nvme_ns_mgmt use nvme_passthru_cmd directly
      ioctl: nvme_ns_attach use nvme_passthru_cmd directly
      ioctl: nvme_ns_download use nvme_passthru_cmd directly
      ioctl: nvme_fw_commit use nvme_passthru_cmd directly
      ioctl: nvme_security_send use nvme_passthru_cmd directly
```

While at it, I've updated the argument order and name follow the spec.

igaw, 2025-08-02:
Thanks! Looked quickly at it looks good! Will pull it on Monday :)

igaw, 2025-08-05:
BTW, I think we should make an exception to the order, the data pointer and metadata pointer (and their length member) go to the end of the list and the last argument is the `result`.

igaw, 2025-08-08:
The two PRs are now integrated into libnvme2, it compiles without warnings again, the tests work also fine. So if you plan to work on converting any of the structs left, please add it here before starting, so we don't work on the same thing :)

ikegami-t, 2025-08-08:
> The two PRs are integrated into libnvme2 now. It compiles without warnings again. The tests work also fine.
> 
> So if you plan to work on converting any of the structs left, please add it here before starting, so we don't work on the same thing :)

I see. Thank you.

ikegami-t, 2025-08-08:
Now I am thinking to work on the commands below using nvme_submit_io_passthru() later.
1. nvme_resv_acquire()
2. nvme_resv_register()
3. nvme_resv_release()
4. nvme_resv_report()
5. nvme_io_mgmt_recv()
6. nvme_io_mgmt_send()
7. nvme_zns_mgmt_send()
8. nvme_zns_mgmt_recv()

MaisenbacherD, 2025-08-08:
> btw, I think this one should be splitted into two commits [48d9388](https://github.com/linux-nvme/libnvme/commit/48d9388b94b8a726408163aeee1c037dc253afd5) ("ioctl: drop timeout arguments") one which introduces the new API `nvme_timeout_set_timeout` (`libnvme.map` needs updating too) and then the conversion patch. Logically, we should first do the conversion of the API and then this timeout thing. Hmm, maybe I drop it for the time being.

Sure I will rework this :)
After doing the patches for nvme-cli3 I will continue with
`nvme_get_features*`

igaw, 2025-08-08:
I'll just drop the commit for now, we can add it later.

igaw, 2025-08-08:
> Sure, I will rework this and keep the commit on a branch that we can later apply :) After doing the patches for nvme-cli3 I will continue with `nvme_get_features*`

I think the simplest approach is to update nvme-cli3 in the same order as libnvme2. But maybe you can just cherry-pick the corresponding patch from libnvme2 instead. I can sort them afterwards in the correct order.

ikegami-t, 2025-08-11:
Just done the work on the functions below using nvme_submit_io_passthru() and nvme_submit_io_passthru64() so for next I will do work on the functions below.
1. nvme_dim_send()
2. nvme_lm_cdq()
3. nvme_lm_track_send()
4. nvme_lm_migration_send()
5. nvme_lm_migration_recv()
  Note: The function below is the {get,set}_features commands so will not work on it for now.
6. nvme_lm_get_features_ctrl_data_queue()

MaisenbacherD, 2025-08-11:
I decided to first finish `nvme_get_features*`: https://github.com/nvme-experiments/libnvme/pull/6

@igaw and @ikegami-t I would dedicate some time to nvme-cli3 to adjust all API calls we already changed for libnvme.
Let me know if anyone already started with some of that work so we are not duplicating work. Otherwise, I am happy to fix all of that up. :)

ikegami-t, 2025-08-12:
No I do not start yet on wokr on nvme-cli3 so I will do continue to work on the remaining libnvme2 ioctl functions to drop ```struct ***_args```. I have done work on the LM related functions by the PR: https://github.com/nvme-experiments/libnvme/pull/5 then seems the remaining functions are below so will do continue to work on the functions from 1. ```nvme_uring_cmd_***()``` to 5. ```nvme_directive_***()``` later at first.
1. nvme_uring_cmd_exit(), nvme_uring_cmd_admin_passthru_async(), nvme_uring_cmd_wait_complete()
2. nvme_get_log_partial()
3. nvme_security_receive()
4. nvme_get_lba_status()
5. nvme_directive_send(), nvme_directive_send_id_endir(), nvme_directive_recv()
6. nvme_capacity_mgmt()
7. nvme_lockdown()
8. nvme_set_property(), nvme_get_property()
9. nvme_sanitize_nvm()
10. nvme_dev_self_test()
11. nvme_virtual_mgmt()

igaw, 2025-08-22:
I've merged updated all pending PRs. So the current buildable versions are:

nvme-cli3: ed3ec5dfdf10 ("nvme: nvme_get_log use nvme_passthru_cmd directly")
libnvme2:  2673e9f8db78 ("test: fix rae and lsp handling in log page tests")

TODO list:

libnvme:

- [ ] struct nvme_get_features_args
- [ ] struct nvme_security_receive_args 
- [ ] struct nvme_get_lba_status_args
- [ ] struct nvme_directive_send_args
- [ ] struct nvme_directive_recv_args
- [ ] struct nvme_capacity_mgmt_args
- [ ] struct nvme_lockdown_args
- [ ] struct nvme_set_property_args
- [ ] struct nvme_get_property_args
- [ ] struct nvme_sanitize_nvm_args
- [ ] struct nvme_dev_self_test_args
- [ ] struct nvme_virtual_mgmt_args

nvme-cli:

- [ ] struct nvme_set_features_args
- [ ] struct nvme_get_features_args
- [ ] struct nvme_format_nvm_args
- [ ] struct nvme_ns_mgmt_args
- [ ] struct nvme_ns_attach_args
- [ ] struct nvme_fw_download_args
- [ ] struct nvme_fw_commit_args
- [ ] struct nvme_security_send_args
- [ ] struct nvme_security_receive_args
- [ ] struct nvme_get_lba_status_args
- [ ] struct nvme_directive_send_args
- [ ] struct nvme_directive_recv_args
- [ ] struct nvme_capacity_mgmt_args
- [ ] struct nvme_lockdown_args
- [ ] struct nvme_set_property_args
- [ ] struct nvme_get_property_args
- [ ] struct nvme_sanitize_nvm_args
- [ ] struct nvme_dev_self_test_args
- [ ] struct nvme_virtual_mgmt_args
- [ ] struct nvme_io_args
- [ ] struct nvme_dsm_args
- [ ] struct nvme_copy_args
- [ ] struct nvme_resv_acquire_args
- [ ] struct nvme_resv_register_args
- [ ] struct nvme_resv_release_args
- [ ] struct nvme_resv_report_args
- [ ] struct nvme_io_mgmt_recv_args
- [ ] struct nvme_io_mgmt_send_args
- [ ] struct nvme_zns_mgmt_send_args
- [ ] struct nvme_zns_mgmt_recv_args
- [ ] struct nvme_zns_append_args
- [ ] struct nvme_dim_args
- [ ] struct nvme_lm_cdq_args
- [ ] struct nvme_lm_track_send_args
- [ ] struct nvme_lm_migration_send_args
- [ ] struct nvme_lm_migration_recv_args

ikegami-t, 2025-08-31:
The TODO list libnvme done by the PRs: https://github.com/nvme-experiments/libnvme/pull/8 and https://github.com/nvme-experiments/libnvme/pull/9. Thank you.

igaw, 2025-09-05:
I've merged the libnvme bits now. I've tried to reorder the patches so they make more sense when reviewing. I need to double check that we got the argument ordering right. After that we should be done with the major work in libnvme.

So I don't plan to reorder the struct args patches. The stay in this order.

Let the nvme-cli insanity begin :)

ikegami-t, 2025-09-06:
Next I will do nvme-cli commands using nvme_io() and nvme_submit_io_passthru() below.
- [ ] nvme_flush
- [ ] nvme_set_var_size_tags
- [ ] nvme_ios
- [ ] nvme_read
- [ ] nvme_write
- [ ] nvme_compare
- [ ] nvme_write_zeros
- [ ] nvme_write_uncorrectable
- [ ] nvme_verify
- [ ] nvme_dsm
- [ ] nvme_copy
- [ ] nvme_resv_acquire
- [ ] nvme_resv_register
- [ ] nvme_resv_release
- [ ] nvme_resv_report
- [ ] nvme_io_mgmt_recv
- [ ] nvme_io_mgmt_send
- [ ] nvme_zns_mgmt_send
- [ ] nvme_zns_mgmt_recv

ikegami-t, 2025-09-13:
Seems almost nvme-cli ***_args strunctures wip as below so now thinking to do remaining `struct nvme_zns_append_args` to `struct nvme_lm_migration_recv_args` for next.
nvme-cli:
- [ ] ~~struct nvme_set_features_args~~: wip (https://github.com/nvme-experiments/nvme-cli/pull/5)
- [ ] struct nvme_get_features_args
- [ ] ~~struct nvme_format_nvm_args~~: wip (https://github.com/nvme-experiments/nvme-cli/pull/5)
- [ ] ~~struct nvme_ns_mgmt_args~~: wip (https://github.com/nvme-experiments/nvme-cli/pull/5)
- [ ] struct nvme_ns_attach_args
- [ ] ~~struct nvme_fw_download_args~~: wip (https://github.com/nvme-experiments/nvme-cli/pull/5)
- [ ] ~~struct nvme_fw_commit_args~~: wip (https://github.com/nvme-experiments/nvme-cli/pull/5)
- [ ] ~~struct nvme_security_send_args~~: wip (https://github.com/nvme-experiments/nvme-cli/pull/5)
- [ ] ~~struct nvme_security_receive_args~~: wip (https://github.com/nvme-experiments/nvme-cli/pull/4)
- [ ] ~~struct nvme_get_lba_status_args~~: wip (https://github.com/nvme-experiments/nvme-cli/pull/4)
- [ ] ~~struct nvme_directive_send_args~~: wip (https://github.com/nvme-experiments/nvme-cli/pull/4)
- [ ] ~~struct nvme_directive_recv_args~~: wip (https://github.com/nvme-experiments/nvme-cli/pull/4)
- [ ] ~~struct nvme_capacity_mgmt_args~~: wip (https://github.com/nvme-experiments/nvme-cli/pull/4)
- [ ] ~~struct nvme_lockdown_args~~: wip (https://github.com/nvme-experiments/nvme-cli/pull/4)
- [ ] ~~struct nvme_set_property_args~~: wip (https://github.com/nvme-experiments/nvme-cli/pull/4)
- [ ] ~~struct nvme_get_property_args~~: wip (https://github.com/nvme-experiments/nvme-cli/pull/4)
- [ ] ~~struct nvme_sanitize_nvm_args~~: wip (https://github.com/nvme-experiments/nvme-cli/pull/4)
- [ ] ~~struct nvme_dev_self_test_args~~: wip (https://github.com/nvme-experiments/nvme-cli/pull/4)
- [ ] ~~struct nvme_virtual_mgmt_args~~: wip (https://github.com/nvme-experiments/nvme-cli/pull/4)
- [ ] ~~struct nvme_io_args~~: wip (https://github.com/nvme-experiments/nvme-cli/pull/5)
- [ ] ~~struct nvme_dsm_args~~: wip (https://github.com/nvme-experiments/nvme-cli/pull/5)
- [ ] ~~struct nvme_copy_args~~: wip (https://github.com/nvme-experiments/nvme-cli/pull/5)
- [ ] ~~struct nvme_resv_acquire_args~~: wip (https://github.com/nvme-experiments/nvme-cli/pull/5)
- [ ] ~~struct nvme_resv_register_args~~: wip (https://github.com/nvme-experiments/nvme-cli/pull/5)
- [ ] ~~struct nvme_resv_release_args~~: wip (https://github.com/nvme-experiments/nvme-cli/pull/5)
- [ ] ~~struct nvme_resv_report_args~~: wip (https://github.com/nvme-experiments/nvme-cli/pull/5)
- [ ] ~~struct nvme_io_mgmt_recv_args~~: wip (https://github.com/nvme-experiments/nvme-cli/pull/5)
- [ ] ~~struct nvme_io_mgmt_send_args~~: wip (https://github.com/nvme-experiments/nvme-cli/pull/5)
- [ ] ~~struct nvme_zns_mgmt_send_args~~: wip (https://github.com/nvme-experiments/nvme-cli/pull/5)
- [ ] ~~struct nvme_zns_mgmt_recv_args~~: wip (https://github.com/nvme-experiments/nvme-cli/pull/5)
- [ ] struct nvme_zns_append_args
- [ ] struct nvme_dim_args
- [ ] struct nvme_lm_cdq_args
- [ ] struct nvme_lm_track_send_args
- [ ] struct nvme_lm_migration_send_args
- [ ] struct nvme_lm_migration_recv_args

igaw, 2025-09-16:
what a mess...

[image]

I suppose the simplest way would be to introduce a cdw12 for nvme_io and have the other commands construct this command word.

MaisenbacherD, 2025-09-16:
Love the analog table!
Sounds like the correct choice to introduce a cdw12.

igaw, 2025-09-16:
```c
static inline int nvme_io(nvme_link_t l, __u8 opcode, __u32 nsid, __u64 slba,
			  __u32 cdw12, __u32 cdw13,
			  __u8 sts, __u8 pif, __u64 storage_tag, __u32 reftag,
			  __u16 lbat, __u16 lbatm, void *data,
			  __u32 data_len, void *metadata, __u32 metadata_len,
			  __u32 *result)
```

hreinecke, 2025-09-16:
That looks _soo_ wrong.

MaisenbacherD, 2025-09-22:
I am taking the nvme_get_features :)

ikegami-t, 2025-09-22:
Additionally just fixed by the PR: https://github.com/nvme-experiments/nvme-cli/pull/19 changes also.

igaw, 2025-09-25:
Okay, let's close this one as the task has been done.

closed 2025-09-25 by igaw