linux-nvme/libnvme #516: Cannot read the entire persistent event log

tkenbo, 2022-11-01:
```
nvme version 2.1.2 (git 2.1.2)
libnvme version 1.1 (git 1.1)
4.18.0-372.26.1.el8_6.x86_64
```
I have a drive with TLL (total log length) to be 893320
```
nvme persistent-event-log /dev/nvme2 -a 0 -l 512
Persistent Event Log for device: nvme2
Action for Persistent Event Log: 0
Log Identifier: 13
Total Number of Events: 7511
**Total Log Length : 893320**
```

However when I request the entire log, I get the Invalid argument error.
```
nvme persistent-event-log /dev/nvme2 -a 0 -l 0
persistent event log: Invalid argument
```

It looks like failed in nvme_alloc_request?
```
            nvme-1270309 [033] .... 1122401.152208: nvme_dev_ioctl <-do_vfs_ioctl
            nvme-1270309 [033] .... 1122401.152208: nvme_user_cmd <-nvme_dev_ioctl
            nvme-1270309 [033] .... 1122401.152208: nvme_submit_user_cmd <-nvme_user_cmd
            nvme-1270309 [033] .... 1122401.152208: nvme_alloc_request <-nvme_submit_user_cmd
            nvme-1270309 [033] .... 1122401.152283: nvme_dev_release <-__fput
```

Thanks for your help in advance.

keithbusch, 2022-11-01:
The size exceeds what the kernel driver fit in a request. A log this size will need to be split into smaller requests.

tkenbo, 2022-11-01:
thanks for your response.  yes but persistent_event_log function does not have the offset parameter and it always read from the beginning, thus cannot read the entire log.  In the following function, lpo is always set to 0.
```
libnvme/ioctl.h
static inline int nvme_get_log_persistent_event(int fd,
			enum nvme_pevent_log_action action,
			__u32 size, void *pevent_log)
{
	struct nvme_get_log_args args = {
		**.lpo = 0,**
		.result = NULL,
		.log = pevent_log,
		.args_size = sizeof(args),
		.fd = fd,
		.timeout = NVME_DEFAULT_IOCTL_TIMEOUT,
		.lid = NVME_LOG_LID_PERSISTENT_EVENT,
		.len = size,
		.nsid = NVME_NSID_ALL,
		.csi = NVME_CSI_NVM,
		.lsi = NVME_LOG_LSI_NONE,
		.lsp = (__u8)action,
		.uuidx = NVME_UUID_NONE,
		.rae = false,
		.ot = false,
	};
	return nvme_get_log(&args);
```

keithbusch, 2022-11-01:
Yeah, I'm saying it needs to use a loop, like have it call `nvme_get_log_page()` instead of directly calling `nvme_get_log()`.

igaw, 2022-11-07:
Make sense. The only question I have is: should we move nvme_get_log_page() to the `ioctl.h`. Currently it is declared in `linux.h` which includes `ioctl.h, which gives us then a circular dependency.

closed 2022-11-07 by keithbusch