linux-nvme/libnvme #524: test-mi: ../test/mi.c:333: test_scan_ctrl_list_cb: Assertion `hdr[11] == nvme_mi_dtyp_ctrl_list' failed

bdrung, 2022-11-08:
Running the test suite on s390x (big endian) fails on Debian and Ubuntu:
```
==================================== 2/6 =====================================
test:         mi
start time:   02:35:41
duration:     0.00s
result:       killed by signal 6 SIGABRT
command:      MALLOC_PERTURB_=24 /<<PKGBUILDDIR>>/obj-s390x-linux-gnu/test/test-mi
----------------------------------- stdout -----------------------------------
Running test endpoint_lifetime...  OK
Running test ctrl_lifetime...  OK
Running test read_mi_data...  OK
Running test transport_fail...  OK
Running test transport_describe...  OK
Running test scan_ctrl_list...
----------------------------------- stderr -----------------------------------
test-mi: ../test/mi.c:333: test_scan_ctrl_list_cb: Assertion `hdr[11] == nvme_mi_dtyp_ctrl_list' failed.
==============================================================================
```

Full log: https://buildd.debian.org/status/fetch.php?pkg=libnvme&arch=s390x&ver=1.2-1&stamp=1667874947&raw=0

This looks to an little vs big endian issue to me.

igaw, 2022-11-08:
@jk-ozlabs

jk-ozlabs, 2022-11-09:
Nice to have some BE coverage in the tests, thanks @bdrung . I'll get a BE setup going here and replicate; I'm pretty sure I've found the underlying issue but will also need to verify the fix.

igaw, 2022-11-09:
As previously mentioned we should be able to run the tests also in the cross builds with qemu.

closed 2022-11-10 by igaw

jk-ozlabs, 2022-11-10:
Thanks! I wonder if we can enforce the current `__leXX` type annotations? that would have caught two of these cases.

igaw, 2022-11-10:
Maybe it could be done with `sparse` but I think we would need to figure out how to use it correctly:

```shell
sparse -Isrc/libnvme-mi-test.so.p -Isrc -I../src -I. -I.. -Iccan -I../ccan -Iinternal -I../internal -fdiagnostics-color=always -D_FILE_OFFSET_BITS=64 -Wall -Winvalid-pch -std=gnu99 -O3 -fomit-frame-pointer -D_GNU_SOURCE -include internal/config.h -fPIC -MD -MQ src/libnvme-mi-test.so.p/nvme_mi.c.o -MF src/libnvme-mi-test.so.p/nvme_mi.c.o.d -o src/libnvme-mi-test.so.p/nvme_mi.c.o -c ../src/nvme/mi.c
../src/nvme/mi.c:97:30: warning: incorrect type in argument 1 (different base types)
../src/nvme/mi.c:97:30:    expected restricted leint16_t [usertype] le_val
../src/nvme/mi.c:97:30:    got restricted __le16 [usertype] vid
../src/nvme/mi.c:327:34: warning: incorrect type in argument 1 (different base types)
../src/nvme/mi.c:327:34:    expected restricted leint16_t [usertype] le_val
../src/nvme/mi.c:327:34:    got restricted __le16 [addressable] [usertype] num
../src/nvme/mi.c:337:49: warning: incorrect type in argument 1 (different base types)
../src/nvme/mi.c:337:49:    expected restricted leint32_t [usertype] le_val
../src/nvme/mi.c:337:49:    got restricted __le16
../src/nvme/mi.c:381:5: warning: symbol 'nvme_mi_submit' was not declared. Should it be static?
../src/nvme/mi.c:485:22: warning: incorrect type in assignment (different base types)
../src/nvme/mi.c:485:22:    expected restricted __le16 [usertype] ctrl_id
../src/nvme/mi.c:485:22:    got restricted leint16_t
../src/nvme/mi.c:535:44: warning: incorrect type in argument 1 (different base types)
../src/nvme/mi.c:535:44:    expected restricted leint32_t [usertype] le_val
../src/nvme/mi.c:535:44:    got restricted __le32 [usertype] cdw0
../src/nvme/mi.c:541:44: warning: incorrect type in argument 1 (different base types)
../src/nvme/mi.c:541:44:    expected restricted leint32_t [usertype] le_val
../src/nvme/mi.c:541:44:    got restricted __le32 [usertype] cdw3
../src/nvme/mi.c:598:28: warning: incorrect type in assignment (different base types)
../src/nvme/mi.c:598:28:    expected restricted __le16 [usertype] ctrl_id
../src/nvme/mi.c:598:28:    got restricted leint16_t
../src/nvme/mi.c:615:25: warning: incorrect type in assignment (different base types)
../src/nvme/mi.c:615:25:    expected restricted __le32 [usertype] dlen
../src/nvme/mi.c:615:25:    got restricted leint32_t
../src/nvme/mi.c:616:25: warning: incorrect type in assignment (different base types)
../src/nvme/mi.c:616:25:    expected restricted __le32 [usertype] doff
../src/nvme/mi.c:616:25:    got restricted leint32_t
../src/nvme/mi.c:671:22: warning: incorrect type in assignment (different base types)
../src/nvme/mi.c:671:22:    expected restricted __le32 [addressable] [usertype] cdw1
../src/nvme/mi.c:671:22:    got restricted leint32_t
../src/nvme/mi.c:672:22: warning: incorrect type in assignment (different base types)
../src/nvme/mi.c:672:22:    expected restricted __le32 [addressable] [usertype] cdw2
../src/nvme/mi.c:672:22:    got restricted leint32_t
../src/nvme/mi.c:673:22: warning: incorrect type in assignment (different base types)
../src/nvme/mi.c:673:22:    expected restricted __le32 [addressable] [usertype] cdw3
../src/nvme/mi.c:673:22:    got restricted leint32_t
../src/nvme/mi.c:674:23: warning: incorrect type in assignment (different base types)
../src/nvme/mi.c:674:23:    expected restricted __le32 [addressable] [usertype] cdw10
../src/nvme/mi.c:674:23:    got restricted leint32_t
../src/nvme/mi.c:675:23: warning: incorrect type in assignment (different base types)
../src/nvme/mi.c:675:23:    expected restricted __le32 [addressable] [usertype] cdw11
../src/nvme/mi.c:675:23:    got restricted leint32_t
../src/nvme/mi.c:676:23: warning: incorrect type in assignment (different base types)
../src/nvme/mi.c:676:23:    expected restricted __le32 [addressable] [usertype] cdw12
../src/nvme/mi.c:676:23:    got restricted leint32_t
../src/nvme/mi.c:677:23: warning: incorrect type in assignment (different base types)
../src/nvme/mi.c:677:23:    expected restricted __le32 [addressable] [usertype] cdw13
../src/nvme/mi.c:677:23:    got restricted leint32_t
../src/nvme/mi.c:678:23: warning: incorrect type in assignment (different base types)
../src/nvme/mi.c:678:23:    expected restricted __le32 [addressable] [usertype] cdw14
../src/nvme/mi.c:678:23:    got restricted leint32_t
../src/nvme/mi.c:679:23: warning: incorrect type in assignment (different base types)
../src/nvme/mi.c:679:23:    expected restricted __le32 [addressable] [usertype] cdw15
../src/nvme/mi.c:679:23:    got restricted leint32_t
../src/nvme/mi.c:682:30: warning: incorrect type in assignment (different base types)
../src/nvme/mi.c:682:30:    expected restricted __le32 [addressable] [assigned] [usertype] dlen
../src/nvme/mi.c:682:30:    got restricted leint32_t
../src/nvme/mi.c:738:22: warning: incorrect type in assignment (different base types)
../src/nvme/mi.c:738:22:    expected restricted __le32 [addressable] [usertype] cdw1
../src/nvme/mi.c:738:22:    got restricted leint32_t
../src/nvme/mi.c:739:23: warning: incorrect type in assignment (different base types)
../src/nvme/mi.c:739:23:    expected restricted __le32 [addressable] [usertype] cdw10
../src/nvme/mi.c:739:23:    got restricted leint32_t
../src/nvme/mi.c:740:23: warning: incorrect type in assignment (different base types)
../src/nvme/mi.c:740:23:    expected restricted __le32 [addressable] [usertype] cdw11
[... 125 lines cut ...]
[... cut ...]
../src/nvme/mi.c:1453:22:    got restricted leint32_t
../src/nvme/mi.c:1489:22: warning: incorrect type in assignment (different base types)
../src/nvme/mi.c:1489:22:    expected restricted __le32 [addressable] [assigned] [usertype] cdw0
../src/nvme/mi.c:1489:22:    got restricted leint32_t
../src/nvme/mi.c:1490:22: warning: incorrect type in assignment (different base types)
../src/nvme/mi.c:1490:22:    expected restricted __le32 [addressable] [assigned] [usertype] cdw1
../src/nvme/mi.c:1490:22:    got restricted leint32_t
```

jk-ozlabs, 2022-11-11:
> Thanks. Confirmed it working.

Excellent, thanks for confirming.

From the package build log:

> ```Installing src/libnvme-mi.so.1.2.0 to /<<PKGBUILDDIR>>/debian/tmp/usr/lib/s390x-linux-gnu```

Nice! I am looking forward to a s390-based Baseboard Management Controller! :laughing: 

[We did just do a [powerpc64le based BMC](https://codeconstruct.com.au/docs/dcscm-openbmc/), so maybe that's not too far off?]