linux-nvme/libnvme #541: `nvme_get_log_page` ignored `arg->lpo` input parameter

jinliangw, 2022-11-23:
Inside the in-band [nvme_get_log_page function](https://github.com/linux-nvme/libnvme/blob/master/src/nvme/ioctl.c#L433), 
the input parameter `arg->lpo` is overwrote accidentally. 
The consequent is that the functions will read log page starting from offset 0, disregarding `arg->lpo` input parameter. 

```
$ sudo ./nvme --version
nvme version 2.2.1 (git 2.2.1)
libnvme version 1.2 (git 1.2-22-g70aca88)
```

Read 32 bytes from offset 0
```
$ sudo ./nvme get-log --log-id=0x02 --lpo=0 --log-len=32 /dev/nvme0
Device:nvme0 log-id:2 namespace-id:0xffffffff
       0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
0000: 00 29 01 64 0a 01 00 00 00 00 00 00 00 00 00 00 ".).d............"
0010: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 "................"
```

Read 16 bytes from offset 16, however, the `libnvme` still returns first 16 bytes data.
```
$ sudo ./nvme get-log --log-id=0x02 --lpo=16 --log-len=16 /dev/nvme0
Device:nvme0 log-id:2 namespace-id:0xffffffff
       0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
0000: 00 29 01 64 0a 01 00 00 00 00 00 00 00 00 00 00 ".).d............"
```

jinliangw, 2022-11-29:
The issue still exists, but the example is not applied becase  nvme-cli `get-log` subommand doesn't use [nvme_get_log_page function](https://github.com/linux-nvme/libnvme/blob/master/src/nvme/ioctl.c#L433)

igaw, 2022-11-30:
Okay, let's reopen it for tracking it.

igaw, 2022-12-02:
https://github.com/linux-nvme/libnvme/pull/547 should address this, right?

jinliangw, 2022-12-02:
Yes, we can close this issue now.

closed 2023-03-30 by igaw