linux-nvme/libnvme #563: libnvme-mi test crash

igaw, 2023-01-25:
The test fails on SLE15-SP4 x86_64 installation:

```gdb
(gdb) r
Starting program: /local/dwagner/nvme-cli/subprojects/libnvme/.build/test/test-mi 
Missing separate debuginfos, use: zypper install glibc-debuginfo-2.31-150300.37.1.x86_64
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib64/libthread_db.so.1".
Running test endpoint_lifetime...  OK
Running test ctrl_lifetime...  OK
Running test read_mi_data...  OK
Running test transport_fail...  OK
Running test transport_describe...  OK
Running test scan_ctrl_list...  OK
Running test invalid_crc...  OK
--- begin test output
crc mismatch
--- end test output
Running test admin_id...  OK
Running test admin_err_mi_resp...  OK
Running test admin_err_nvme_resp...  OK
Running test admin_invalid_formats...  OK
Running test resp_req...  OK
Running test resp_hdr_small...  OK
Running test resp_invalid_type...  OK
Running test resp_csi...  OK
--- begin test output
Command slot mismatch: req 0, resp 1
--- end test output
Running test mi_config_get_mtu...  OK
Running test mi_config_set_freq...  OK
Running test mi_config_set_freq_invalid...  OK
Running test get_features_nodata...  OK
Running test get_features_data...  OK
Running test set_features...  OK
Running test admin_id_alloc_ns_list...  OK
Running test admin_id_active_ns_list...  OK
Running test admin_id_alloc_ns...  OK
Running test admin_id_active_ns...  OK
Running test admin_id_nsid_ctrl_list...  OK
Running test admin_id_secondary_ctrl_list...  OK
Running test admin_ns_mgmt_create...  OK
Running test admin_ns_mgmt_delete...  OK
Running test admin_ns_attach...  OK
Running test admin_ns_detach...  OK
Running test admin_fw_download...  OK
Running test admin_fw_commit...  OK
Running test admin_format_nvm...  OK
Running test admin_sanitize_nvm...  OK
Running test admin_get_log_split...
Program received signal SIGSEGV, Segmentation fault.

gdb) bt -full
#0  0x00007ffff7bcacca in nvme_mi_admin_parse_status (resp=0x7fffffffbac0, result=0x7ffff7de43e4 <_dl_name_match_p+86>) at ../src/nvme/mi.c:545
        admin_hdr = 0x7fffffffbb40
        resp_hdr = 0x7fffffffbb40
        nvme_status = 0
        nvme_result = 0
#1  0x00007ffff7bcb752 in __nvme_mi_admin_get_log (ctrl=0x60d170, args=0x7fffffffbbf0, offset=0, lenp=0x7fffffffbb90, final=false) at ../src/nvme/mi.c:837
        log_page_offset = 0
        resp_hdr = {hdr = {type = 132 '\204', nmp = 144 '\220', meb = 0 '\000', rsvd0 = 0 '\000'}, status = 0 '\000', rsvd0 = "\000\000", cdw0 = 0, cdw1 = 0, cdw3 = 0}
        req_hdr = {hdr = {type = 132 '\204', nmp = 16 '\020', meb = 0 '\000', rsvd0 = 0 '\000'}, opcode = 2 '\002', flags = 1 '\001', ctrl_id = 5, cdw1 = 32767, cdw2 = 0, cdw3 = 0, cdw4 = 0, 
          cdw5 = 0, doff = 0, dlen = 4096, rsvd0 = 0, rsvd1 = 0, cdw10 = 67076097, cdw11 = 0, cdw12 = 0, cdw13 = 0, cdw14 = 16777216, cdw15 = 0}
        resp = {hdr = 0x7fffffffbb40, hdr_len = 20, data = 0x7fffffffbc30, data_len = 4096, mic = 178398476}
        req = {hdr = 0x7fffffffbaf0, hdr_len = 68, data = 0x0, data_len = 0, mic = 2824719863}
        len = 4096
        ndw = 1023
        rc = 0
#2  0x00007ffff7bcb882 in nvme_mi_admin_get_log (ctrl=0x60d170, args=0x7fffffffbbf0) at ../src/nvme/mi.c:872
        xfered_size = 4096
        cur_xfer_size = 4096
        final = false
        max_xfer_size = 4096
        xfer_offset = 0
        rc = 0
#3  0x0000000000406e76 in test_admin_get_log_split (ep=0x60bb40) at ../test/mi.c:1849
        buf = '\000' <repeats 4096 times>...
        args = {lpo = 0, result = 0x7ffff7de43e4 <_dl_name_match_p+86>, log = 0x7fffffffbc30, args_size = 64, fd = 32767, timeout = 0, lid = NVME_LOG_LID_ERROR, len = 8196, nsid = 32767, 
          csi = NVME_CSI_KV, lsi = 0, lsp = 0 '\000', uuidx = 0 '\000', rae = 211, ot = false}
        ldata = {n = 1}
        ctrl = 0x60d170
        rc = 0
        __PRETTY_FUNCTION__ = "test_admin_get_log_split"
#4  0x0000000000406f1a in run_test (test=0x60a3f0 <tests+560>, logfd=0x60b2d0, ep=0x60bb40) at ../test/mi.c:1904
No locals.
#5  0x0000000000406fdb in main () at ../test/mi.c:1926
        root = 0x60bb00
        ep = 0x60bb40
        i = 35
        fd = 0x60b2d0
        __PRETTY_FUNCTION__ = "main"
(gdb)
```

@jk-ozlabs any ideas what's going on?

jk-ozlabs, 2023-01-26:
Looks like the `result` pointer is garbage:

```
#0  0x00007ffff7bcacca in nvme_mi_admin_parse_status (resp=0x7fffffffbac0, result=0x7ffff7de43e4 <_dl_name_match_p+86>) at ../src/nvme/mi.c:545
```

Which just comes from a stack var in the test, which we haven't zeroed:

```c
static void test_admin_get_log_split(struct nvme_mi_ep *ep)
{
	unsigned char buf[4096 * 2 + 4];
	struct nvme_get_log_args args;
	struct log_data ldata;
	nvme_mi_ctrl_t ctrl;
	int rc;

	ldata.n = 0;
	test_set_transport_callback(ep, test_admin_get_log_split_cb, &ldata);

	ctrl = nvme_mi_init_ctrl(ep, 5);

	args.args_size = sizeof(args);
	args.lid = 1;
	args.log = buf;
	args.len = sizeof(buf);
	args.lpo = 0;
	args.ot = false;

	rc = nvme_mi_admin_get_log(ctrl, &args);
```

So, just the testcase that needs fixing. PR coming soon.

jk-ozlabs, 2023-01-26:
The fix:

```diff
@@ -1828,8 +1828,8 @@ static int test_admin_get_log_split_cb(struct nvme_mi_ep *ep,
 
 static void test_admin_get_log_split(struct nvme_mi_ep *ep)
 {
+       struct nvme_get_log_args args = { 0 };
        unsigned char buf[4096 * 2 + 4];
-       struct nvme_get_log_args args;
        struct log_data ldata;
        nvme_mi_ctrl_t ctrl;
        int rc;
```

.. but I've included other similar cases in the PR (#565).

The crash doesn't reproduce here, but I do see the uninitalised var reference through valgrind. With that PR applied, the test is now valgrind-clean:

```
[jk@pecola libnvme]$ valgrind obj/test/test-mi > /dev/null
==400368== Memcheck, a memory error detector
==400368== Copyright (C) 2002-2022, and GNU GPL'd, by Julian Seward et al.
==400368== Using Valgrind-3.19.0 and LibVEX; rerun with -h for copyright info
==400368== Command: obj/test/test-mi
==400368== 
==400368== 
==400368== HEAP SUMMARY:
==400368==     in use at exit: 0 bytes in 0 blocks
==400368==   total heap usage: 48 allocs, 48 frees, 10,598 bytes allocated
==400368== 
==400368== All heap blocks were freed -- no leaks are possible
==400368== 
==400368== For lists of detected and suppressed errors, rerun with: -s
==400368== ERROR SUMMARY: 0 errors from 0 contexts (suppressed: 0 from 0)
```

igaw, 2023-01-26:
All test passes on my test machine. Thanks!

closed 2023-01-26 by igaw