linux-nvme/libnvme #588: doc: Wrong path for /etc/nvme

martin-belanger, 2023-03-02:
In the documentation directory ([./doc/rst/*.rst](https://github.com/linux-nvme/libnvme/tree/master/doc/rst)) there are some files that mention `/usr/local/etc/nvme` as the default location for the configuration files instead of `/etc/nvme`.

This is probably due to the fact that the `*.rst` files get auto-generated and the `prefix` in `[meson.build](https://github.com/linux-nvme/libnvme/blob/master/meson.build#L17)` now defaults to `prefix=/usr/local`.

igaw, 2023-03-03:
We can't really solve this for real with the precompiled documentation we ship. Personally, I would just drop it.

igaw, 2023-03-10:
I am trying to figure out how to handle `--prefix` and absolute paths:

https://groups.google.com/g/mesonbuild/c/_7gXr9heNJA

igaw, 2023-03-10:
The problem is that `sysconfdir` needs to be explicitly initialized so that `join_paths()` works consistently with Meson and muon.

igaw, 2023-03-21:
So one thing we could do is rewording the documentation to something like 

> The configuration is read from `$SYSCONFDIR/nvme/discovery.conf` (SYSCONFDIR is usually /etc) or so.

Basically avoid full paths in the documentation.

igaw, 2023-03-29:
> We can't really solve this for real with the precompiled documentation we ship. Personally, I would just drop it.

Let's do this with v2.0.

closed 2023-03-30 by igaw