linux-nvme/libnvme #589: Compile Failures due include files not found

jeffreyalien, 2023-03-03:
I'm trying to build both nvme-cli and libnvme on Ubuntu 18.04 and both are failing with the following errors:
nvme-cli
  ../util/types.h:10:10: fatal error: libnvme.h: No such file or directory
 #include <libnvme.h>

libnvme
../src/nvme/json.c:15:10: fatal error: json.h: No such file or directory
 #include <json.h>

This builds fine on a new install of Ubuntu so I suspect there's some issue with some remnant's leftover from a previous build that are causing this.  Is there a way to clean this up?   

The build messages show that the libnvme and json-c dependencies are found and up to date so they are not being cloned and built in the subprojects dir.  Is there a way to clear the dependency checking so the submodules are rebuilt?  Or force them to be built?   
Run-time dependency libnvme found: YES 1.3
Run-time dependency libnvme-mi found: YES 1.3
Run-time dependency json-c found: YES 0.16

Or is there some other way of cleaning this up (short of reinstalling the system)?

igaw, 2023-03-06:
Did you remove your build directory first? Could you provide logs?

jeffreyalien, 2023-03-06:
Did a "make purge", "meson subprojects update", and "make" and the make command still fails with the message:
"fatal error: libnvme.h: No such file or directory".

Attached the make output along with the meson log file.  Let me know if there are other log files you'd like to me to attach. 

[meson-log.txt](https://github.com/linux-nvme/libnvme/files/10899920/meson-log.txt)
[make-msgs.txt](https://github.com/linux-nvme/libnvme/files/10899922/make-msgs.txt)

igaw, 2023-03-06:
pkg-config reports that libnvme has been installed in `/usr/local/`:

```
Pkg-config binary for 1 is not cached.
Pkg-config binary missing from cross or native file, or env var undefined.
Trying a default Pkg-config fallback at pkg-config
Found pkg-config: /usr/bin/pkg-config (0.29.1)
Determining dependency 'libnvme' with pkg-config executable '/usr/bin/pkg-config'
env[PKG_CONFIG_PATH]: 
Called `/usr/bin/pkg-config --modversion libnvme` -> 0
1.3
env[PKG_CONFIG_PATH]: 
Called `/usr/bin/pkg-config --cflags libnvme` -> 0
-I/usr/local/include -I/usr/local/include/json-c
```

Does the file `libnvme.h` exist in `/usr/local/include`?

jeffreyalien, 2023-03-06:
ddtest@ub18:~/git/nvme-cli$ ls /usr/local/include/*
ls: cannot access '/usr/local/include/*': No such file or directory

igaw, 2023-03-06:
Your libnvme installation is borken. Just cleanup the libnvme related files user `/usr/local`.

closed 2023-03-06 by igaw

jeffreyalien, 2023-03-06:
There are no libnvme related files in /usr/local so I'm not sure what to cleanup.

jeffreyalien, 2023-03-06:
I've uninstalled json-c and removed libnvme nvme  and "meson subprojects update" returns - Not used?  
ddtest@ub18:~/git/nvme-cli$ meson subprojects update 
Updating json-c...
  -> Not used.
Updating libnvme...
  -> Not used.

Also, make is returning  found for libnvme and json-c.  Where is make looking for them?  I've verified there's no libnvme or json-c files in /usr/include and /usr/local/include.  
Run-time dependency libnvme found: YES 1.3
Run-time dependency libnvme-mi found: YES 1.3
Run-time dependency json-c found: YES 0.16

igaw, 2023-03-07:
There is excellent documentation on this topic: https://mesonbuild.com/Dependencies.html#dependency-detection-method

pkg-config finds a libnvme.pc file in $PKG_CONFIG_PATH, most likely it's  `/usr/local/lib64/pkgconfig`

Also you can tell Meson to use the fallbacks (`--wrap-mode=forcefallback`) and ignore the systemd installed libraries.

jeffreyalien, 2023-03-07:
Thanks for the info Dan.