linux-nvme/libnvme #611: Why can't I write 512B in nvme_ns_write?

Rookie-roob, 2023-04-02:
In function nvme_ns_write, I can not write the data when count = 512. And in function nvme_bytes_to_lba, I see 
if (!count || offset & bs || count & bs) {
		errno = EINVAL;
		return -1;
	}

I want to ask why we must judge the result of "count & bs"

igaw, 2023-04-03:
Good question. The function `nvme_bytes_to_lba()` wants enforce that  `count` and `offset` value is a multiple of the device `blocksize`. While I get the offset I do fail the reason why count must also be an multiple of `blocksize` at least for the read and write command.

From the Command Set specification (`3.6.2 write command`)
```
Starting LBA (SLBA): This field indicates the 64-bit address of the first logical block to be
written as part of the operation. Command Dword 10 contains bits 31:00; Command 
Dword 11 contains bits 63:32.

```

```
Number of Logical Blocks (NLB): This field indicates the number of logical blocks to 
be written. This is a 0’s based value.
```

So from my understanding `count % bs != 0`  is valid if `NLB` accounts for it. The len written is limited by `data_len` and `data_len` is `count`.

keithbusch, 2023-04-03:
Oops, that is a mistake. I think it should be `count & (bs - 1)`.

igaw, 2023-04-04:
Okay, that makes sense. Let me spin a fix.

PR #615 (2023-04-05): tree: Fix argument check in nvme_bytes_to_lba
nvme_bytes_to_lba() argument checker is ensuring that all passed in values are valid. That means we have at least one block to write, the offset is aligned to a block starting address and the number of blocks is a multiple of the block size

The last check is wrong, thus fix it.

Fixes #611
closed 2023-04-05 by igaw